cariboulabs/cariboulite #108: Wrong board shipped?

synthetikspace, 2023-04-02:
Did anyone else here have the wrong board shipped when ordering the 6Ghz model but instead received the SMI version?

I have reached out to David Michaeli who responded that the issue would be resolved but I have not heard back in over a month now after following up several times to no avail.

Very disappointing.

ImDroided, 2023-04-02:
There's a thread somewhere buried in the issues page where someone was posting up pictures of their board, not even populated in the right spots or something like that. But yeah it's known that are quite a few people have gotten the wrong stuff. I don't know what they're doing to fix it though.

K7MDL2, 2023-04-03:
Try contacting Crowdsupply support?  I believe they are doing the fulfilment and this has been acknowledged by them for some others and they arranged the swap..

rhammock1, 2023-04-03:
Contacting crowd supply worked for me. I sent pictures of the board I got and packaging and they shipped me a new one within a week or two

ojg78, 2023-04-03:
What are the differences (physically) between the smi and 6ghz version? I feel like it would be smart to check my freshly received board now and try to ignore the possibility that the whole project seems semi abandoned by it's originator.

rhammock1, 2023-04-03:
The ISM version just has empty solder pads on the Mixer section at the bottom of the board and the area surrounded by the gold rectangle on top. When I got mine I thought it had missed a step in the production and was just missing some components.

jadware, 2023-04-24:
happened to me, was an issue at their distributor.  all got sorted out after a couple weeks and have the right boards now.  just be patient, shaking out all the bugs in the process

closed 2023-05-16 by meexmachina