cariboulabs/cariboulite #110: Transmit and receive streams? caribou_smi_read -> Timeout?

nevetts, 2023-04-07:
Has anyone got the python example scripts working? 

Installation etc was not smooth but I have it in a state where SoapySDRUtil --find sees the Cariboulite and the basic transmit  / receive examples (capture.py, soapy_synth.py)  _seem_ to work.

For the other examples I mostly get a lot time out warnings... in particular I wanted to run soapy_iq_calib.py as an example with both tx and rx streams but the script fails and gives the below output. Not sure what else to try here...

partial output of soapy_iq_calib.py :

`clicked run
04-07 15:58:39.663  1519  1519 D CARIBOULITE Radio cariboulite_radio_set_frequency@cariboulite_radio.c:778 Frequency setting CH: 0, Wanted: 868000000.00 Hz, Set: 0.00 Hz (MOD: 867999936.00, MIX: 0.00)
04-07 15:58:39.663  1519  1519 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:803 Activating channel 0, dir = TX, activate = 1
04-07 15:58:39.664  1519  1520 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO09: Transceiver ready
04-07 15:58:39.664  1519  1519 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:835 Setup Modem state tx_prep
[INFO] setFrequency dir: 1, channel: 1, freq: 867999936.00
index... 0
04-07 15:58:39.665  1519  1519 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:803 Activating channel 0, dir = TX, activate = 0
caribou_smi_read -> Timeout
04-07 15:58:39.666  1519  1530 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:967 SMI reading operation returned timeout
04-07 15:58:39.695  1519  1519 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:823 Setup Modem state trx_off
04-07 15:58:39.695  1519  1519 D CARIBOULITE Radio cariboulite_radio_set_frequency@cariboulite_radio.c:778 Frequency setting CH: 0, Wanted: 866750000.00 Hz, Set: 0.00 Hz (MOD: 866749824.00, MIX: 0.00)
04-07 15:58:39.695  1519  1519 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:803 Activating channel 0, dir = TX, activate = 0
04-07 15:58:39.725  1519  1519 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:823 Setup Modem state trx_off
[INFO] setFrequency dir: 0, channel: 0, freq: 866749824.00
04-07 15:58:39.725  1519  1519 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:803 Activating channel 0, dir = TX, activate = 1
04-07 15:58:39.727  1519  1520 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO09: Transceiver ready
04-07 15:58:39.727  1519  1519 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:835 Setup Modem state tx_prep
caribou_smi_read -> Timeout
04-07 15:58:39.989  1519  1530 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:967 SMI reading operation returned timeout
caribou_smi_read -> Timeout
04-07 15:58:40.252  1519  1530 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:967 SMI reading operation returned timeout
caribou_smi_read -> Timeout
04-07 15:58:40.514  1519  1530 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:967 SMI reading operation returned timeout
Error Reading Samples from Device (error code = -5)!
caribou_smi_read -> Timeout
04-07 15:58:40.776  1519  1530 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:967 SMI reading operation returned timeout
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
caribou_smi_read -> Timeout
04-07 15:58:41.039  1519  1530 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:967 SMI reading operation returned timeout
caribou_smi_read -> Timeout
04-07 15:58:41.301  1519  1530 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:967 SMI reading operation returned timeout
caribou_smi_read -> Timeout
`

I also tried [measure_delay.py from the SoapySDR examples](https://github.com/pothosware/SoapySDR/blob/master/swig/python/apps/MeasureDelay.py). It also uses both tx and rx streams...and throws

`    if not sdr.hasHardwareTime():
        raise Exception('this device does not support timed streaming')`

K7MDL2, 2023-04-08:
If you build from main branch you probably won't see the timeouts.

nevetts, 2023-04-08:
This is a build from main branch

dboyan, 2023-04-09:
I don't think transmit and receive can _easily_ work together. According to AT86RF215 (the RF chip) datasheet, it has two transceivers (corresponding to the two antenna ports on the boards), each of which can only work in either TX or RX at a given time.

One could in theory select TX in one transceiver and RX in another. But I'm not sure if the current software and firmware support that well. Also, even if it works, the two transceivers work at different frequency ranges. You do get some meaningful overlap in sub-1G with the full version, though.

nevetts, 2023-04-09:
For my application, I want to use the wide tuning port to transmit and then immediately (after a real time wait of a few microseconds) switch to receive.  

I'm new to this, but.....the datasheet suggests that the RF chip should have this functionality (section 6.15.4 Transmit and Switch to Receive (TX2RX) ).

a-oxide, 2023-05-12:
I have the same issue.

rush0815, 2023-09-04:
I have the same issue, too. 
running dump1090 with soapy support results in the same error.