cariboulabs/cariboulite #113: Alternative Support to Raspberry Pi

the-architech, 2023-04-10:
Since getting a Pi these days has become more challenging. Would there be any interest from the community in getting Cariboulite working on non Pi alternatives?

I am mainly looking at Radxa Zero, which has the same form factor as Pi Zero 2 W, but with more computational power. 

The Radxa Zero has an Amlogic S905Y2.

Since the CaribouLite SDR relies on the pigpio library, would be possible to use an alternative gpio access library?

Possible alternative to pigpio would be wiringpi, which does support Amlogic SOC's. https://github.com/Dangku/WiringPi2-Python-Amlogic

Or ampatio can be an option too.  https://github.com/adir-vrd/ampatio

Thoughts?

ImDroided, 2023-04-11:
I have one of these with a 32gb EMMC https://libre.computer/products/roc-rk3328-cc/ that I tried at first but the script complained about it not being a pi and puked on itself. One of these days I would like to give another go at it because its $50 and has a decent processor decent ram gig ethernet, plus if you buy a $20 32gb emmc then you dont need to worry about killing SD cards.

alphafox02, 2023-04-15:
I would refer to this important section as to what the device will run on, it’s unlikely the other devices being mentioned have the SMI interface. 

https://github.com/cariboulabs/cariboulite#smi-interface