cariboulabs/cariboulite #115: SMI modules are probed

Mudary, 2023-04-14:
RPI 3+, bullseye 64bit, 8gig SD, Cariboulite main branch

To check whether these modules are probed, follow the next command in your Raspberry Pi terminal:

lsmod | grep smi

you should be seeing the following output:

bcm2835_smi_dev     16384   0
bcm2835_smi         20480   1   bcm2835_smi_dev

this is mine

smi_stream_dev         20480  2
bcm2835_smi            20480  1 smi_stream_dev

Thanks Mudary

Mudary, 2023-04-14:
04-14 13:23:45.574  3306  3306 E FPGA caribou_fpga_set_smi_channel@caribou_fpga.c:541 caribou_fpga_set_smi_channel: dev not initialized
04-14 13:23:45.575  3306  3306 E FPGA caribou_fpga_set_io_ctrl_dig@caribou_fpga.c:403 caribou_fpga_set_io_ctrl_dig: dev not initialized
04-14 13:23:45.575  3306  3306 E CARIBOU_SMI caribou_smi_set_driver_streaming_state@caribou_smi.c:35 failed setting smi stream state (0)

Mudary, 2023-04-14:
Sorry still learning github
mudary2@raspberrypi:~ $ sudo sdrpp -s
[14/04/2023 13:47:28.000] [INFO] SDR++ v1.1.0
[14/04/2023 13:47:28.000] [INFO] Loading config
[14/04/2023 13:47:28.000] [INFO] =====| SERVER MODE |=====
[14/04/2023 13:47:28.000] [WARN] ConfigManager locked, waiting...
[14/04/2023 13:47:28.000] [INFO] Loading modules
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[14/04/2023 13:47:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[14/04/2023 13:47:28.000] [INFO] Initializing Airspy Source (airspy_source)
[14/04/2023 13:47:28.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[14/04/2023 13:47:28.000] [INFO] Initializing Audio Source (audio_source)
[14/04/2023 13:47:28.000] [INFO] Initializing File Source (file_source)
[14/04/2023 13:47:28.000] [INFO] Initializing HackRF Source (hackrf_source)
[14/04/2023 13:47:28.000] [INFO] Initializing Hermes Source (hermes_source)
[14/04/2023 13:47:28.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
[14/04/2023 13:47:28.000] [INFO] Initializing RFspace Source (rfspace_source)
[14/04/2023 13:47:28.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[14/04/2023 13:47:28.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[14/04/2023 13:47:28.000] [INFO] Initializing SDR++ Server Source (sdrpp_server_source)
[14/04/2023 13:47:28.000] [INFO] Initializing SoapySDR Source (soapy_source)
[INFO] SoapyCaribouliteSession, sessionCount: 0
04-14 13:47:28.635  3953  3953 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
04-14 13:47:28.635  3953  3953 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
04-14 13:47:28.635  3953  3953 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
04-14 13:47:28.635  3953  3953 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
04-14 13:47:28.635  3953  3953 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
04-14 13:47:28.636  3953  3953 V initInitialise@pigpio.c:8306 Can't lock /var/run/pigpio.pid
04-14 13:47:28.637  3953  3953 E IO_UTILS_Main io_utils_setup@io_utils.c:36 initializing pigpio failed
04-14 13:47:28.637  3953  3953 E CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:205 Error setting up io_utils
[ERROR] cariboulite_init_driver() failed
04-14 13:47:28.637  3953  3953 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
Printing 'findCariboulite' Request:
13:47:28.729250 INFO:   api_drivers_init:109 [ DEF] Probing 'libxtrxll_pcie.so.0' low-level library
13:47:28.738359 INFO:   api_drivers_init:109 [ DEF] Probing 'libxtrxll_libusb3380.so.0' low-level library
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
13:47:28.891385 WARN:   xtrxllpciev0_discovery:243 [PCIE] XTRX PCIe driver isn't loaded
Printing 'findCariboulite' Request:
    {channel: HiF}
    {device_id: 1}
    {driver: Cariboulite}
    {label: CaribouLite HiF[41e2aea5]}
    {name: CaribouLite RPI Hat}
    {serial: 41e2aea5}
    {uuid: 5b3004a3-7b7c-45c3-80bd-a2e781329700}
    {vendor: CaribouLabs LTD}
    {version: 0x0001}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[41e2aea5], ChannelType: HiF
04-14 13:47:29.424  3953  3953 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:803 Activating channel 1, dir = RX, activate = 1
04-14 13:47:29.424  3953  3953 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
04-14 13:47:29.425  3953  3953 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
04-14 13:47:29.425  3953  3953 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:835 Setup Modem state tx_prep
04-14 13:47:29.425  3953  3953 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
04-14 13:47:29.425  3953  3953 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:849 Setup Modem state cmd_rx
04-14 13:47:29.455  3953  3953 E FPGA caribou_fpga_set_smi_channel@caribou_fpga.c:541 caribou_fpga_set_smi_channel: dev not initialized
04-14 13:47:29.455  3953  3953 E FPGA caribou_fpga_set_io_ctrl_dig@caribou_fpga.c:403 caribou_fpga_set_io_ctrl_dig: dev not initialized
[... cut ...]
04-14 13:47:29.547  3953  3953 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
04-14 13:47:29.547  3953  3953 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
04-14 13:47:29.803  3953  3953 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
04-14 13:47:29.804  3953  3953 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
mudary2@raspberrypi:~ $

K7MDL2, 2023-04-17:
I am working in the field for another week and cannot see my system, but I recall seeing the same messages at various times.  Try looking for another process that may running first locking the resources.   When switching around my system between the 5 or so different apps, I seem to hit these kinds of issues.  SoapySDRServer, or not, SDR++ server, or not.
I have OpenWebRX auto starting, and I have to restart or kill processes at times to run others on the same resource (CaribouLite).

Mudary, 2023-04-17:
Thank you, I'll try konimaru fork.