cariboulabs/cariboulite #116: Is this project abandoned?

codyc1515, 2023-04-15:
Seems to be little to no activity in the past few months and many issues are still outstanding during that time. No PRs appear to be accepted. Mixed reports of some people receiving their hardware.

ImDroided, 2023-04-15:
It seems the developer took our money and ran with it. Sure he delivered the board but has been basically a no show for a while now. I mean he hasn't even merged the fixes people have done.

edegraaff, 2023-04-16:
is there a complain procedure rwgarding this? In fact what is delivered is not what was prommissed

great9, 2023-04-16:
Yes, this project is abandoned

meexmachina, 2023-04-16:
Hello,
The project is not abandoned. We have struggled to run it lately due to a severe lack of time. I plan to get into the many problems soon this week. Thank you all for the input.
David

edegraaff, 2023-04-16:
hi David would be great if we get the whole working and share it with everyone, i have at least 2 persons in nl that also struggle to make it in a fully working sdr as we are familiar with.like kiwi, rtl, and others. Please keep us updated also on the main projectpage regarding releases, etc...

great9, 2023-05-04:
> Hello, The project is not abandoned. We have struggled to run it lately due to a severe lack of time. I plan to get into the many problems soon this week. Thank you all for the input. David

3 weeks later, still 3 months old code. 0 PRs merged, 0 lines of code added, modified or deleted.

edegraaff, 2023-05-04:
> > Hello, The project is not abandoned. We have struggled to run it lately due to a severe lack of time. I plan to get into the many problems soon this week. Thank you all for the input. David
> 
> 3 weeks later, still 3 months old code. 0 PRs merged, 0 lines of code added, modified or deleted.

Is  creates some uncomfortable vibes ...

a-oxide, 2023-05-05:
Is there a policy for this sort of thing on Crowdsupply? I feel robbed.

ImDroided, 2023-05-05:
> Is there a policy for this sort of thing on Crowdsupply? I feel robbed.

You know I'm not sure but I'll tell you what I'm going to start contacting them and all of us should and make sure this dude never has another opportunity to make another product again on there because he's not supporting the one that he already did. We're not asking for much. We're literally asking for the couple of software fixes to get this thing working right? If that's too much then he needs the refund money. I'll gladly send this piece of crap back and buy something that actually works.

edegraaff, 2023-05-05:
To be honest, i think most of us have been supporting this project quite some while, and also trusted we get a great SDR. Promisses have been made. I have some respect for the fact chips where hard to get and so on. What realy upsets me is, lack on communication, lack on picking up the issues, the comunity is helping, but some actions should be taken responsibility and/or start a send back program, since it failed. But stop advertising. https://www.crowdsupply.com/cariboulabs/cariboulite-rpi-hat

great9, 2023-05-05:
a simple pull request that fixe the inverted spectrogram consisting of 3 short lines of code weren't merged. what are we talking about here? perhaps the guy has family issues, perhaps he's in jail, maybe he's sick ... but if he has time to comment on an issue on github then he has time to click the merge button or assign write permissions for the repo to another person or change ownership of the repo.

a-oxide, 2023-05-05:
My thoughts exactly.

great9, 2023-05-09:
this project needs a developer who knows what he's doing and doesn't sell stories. I nominate @konimaru and I think we should all chip in and donate some funds to motivate him to bring us a working driver and software support.

ImDroided, 2023-05-26:
> Just chiming in here; I never received my hardware and ordered over a year
> ago (April 2022). I've requested a refund through Crowd Supply.

That's probably a good thing because the developer is proved himself pretty
much useless. Unless we get somebody to take over this project that
actually knows what they're doing and understands it. It's pretty much a
paperweight.. It works mediocre at best when it does. I'm about to contact
crowd supply to find out how we can get a refund because there's plenty
other projects out there that are a little more expensive than this but
they work.. we all just wish the developer would deliver what he sold
because I mean honestly the things pretty cool if it would work.

great9, 2023-05-26:
> That's probably a good thing because the developer is proved himself pretty much useless. Unless we get somebody to take over this project that actually knows what they're doing and understands it. [...]

Agreed. I'd like a refund as well.

edegraaff, 2023-05-26:
i am also figuring out how to refund, i support al lot of projects this is the fist in 30 years where i see a complete faillure in prommiss, in time ,attention, and drive.

ImDroided, 2023-05-26:
If anybody contacts crowd supply and figures out the refund process, please
post it here for us.

a-oxide, 2023-05-26:
I am SOL because I purchased it two months ago. If anyone gets their money
back let me know.

MCKLtech, 2023-05-31:
I received a refund from CrowdSupply. Open a ticket with your Order number, they were very helpful.

codyc1515, 2023-06-08:
This is disappointing. I will take it as given that the project is abandoned.

meexmachina, 2023-06-08:
@codyc1515, the code is updating, and advancing. slowly than expected but it does. We pushed major changes to the 'main' branch lately (kernel module, tx side, bug fixes, install fixes for DragonOS etc.).
Unfortunately, it takes more time than expected as we do it serial rather than parallelly, and we'd love to get your help,  but be assured that each and every issue in GitHub is being checked and we try to make the solution more sustainable than just patching some issue...

MCKLtech, 2023-06-08:
> @codyc1515, the code is updating, and advancing. slowly than expected but it does. We pushed major changes to the 'main' branch lately (kernel module, tx side, bug fixes, install fixes for DragonOS [...]

Completely irrelevant if you haven't shipped hardware to your backers, the people that believed you in the first place.

joyel24, 2023-06-08:
> @codyc1515, the code is updating, and advancing. slowly than expected but it does. We pushed major changes to the 'main' branch lately (kernel module, tx side, bug fixes, install fixes for DragonOS [...]

You can check and link your repository with boutysource, to encourage developers to fix bugs or improve things...

mvmn, 2023-11-13:
The order option is still open and says will ship in 3 business days.
I was foolish enough to order 9 days ago not seeing this issue.
Any chances on getting the hardware? Clearly it's not shipping 3 business days.

ImDroided, 2023-11-13:
> The order option is still open and says will ship in 3 business days. I was foolish enough to order 9 days ago not seeing this issue. Any chances on getting the hardware? Clearly it's not shipping 3 business days.

I received my hardware but I was an early backer I haven't heard of any issues of getting people their hardware. I know there was an issue with the hardware that people were getting and I believe that was corrected but I haven't really heard of anything other than a few posts here and there about people not receiving their stuff. Most of us got the hardware but the software was the issue and they've been working on that slowly but steadily.

MCKLtech, 2023-11-13:
> The order option is still open and says will ship in 3 business days. I was foolish enough to order 9 days ago not seeing this issue. Any chances on getting the hardware? Clearly it's not shipping 3 business days.

I requested a refund from CrowdSupply and they were great, I got it back within a few days. I was an early backer and never got any updates, and never got the hardware. Respectfully to all involved, it's abundantly clear they don't care about this project.

MCKLtech, 2024-01-10:
No, I don't trust this team or company. They never reached out when I asked them about my order or from the cancellation. There are better, more professional products in the space to give money to IMO.

PRufener, 2024-04-12:
Just a shitty project. Never worked. Neither with Raspberry OS, nor with Dragon OS, nor with Kali Linux.
An efficient way to waste your money!