cariboulabs/cariboulite #117: Progress regarding Pi Zero  (N)

edegraaff, 2023-05-07:
Does someone having a normal performing Cariboulite hat on a Pi Zero or Zero N? 
Since i received the cariboulite i have testes all kind of pi's and versions of os, attended and unattended. 
For now i have 1 success working on a pi4 , with 64 bit os unattended with poor results. 

In fact, the whole promisse to get cariboulite running on a Pi Zero is imho a target that is not been realized. This makes the whole project way to expensive. When i take the cost of the cariboulite and a pi4 i could for this money almost buy an airspy, the hackrf is a bit more expensive. But for his money i realy did expected someting that is realy supporting my openwebrx sdr or could do some great work on its own.

All suggestions how to handle this welcome. 

nl13974

great9, 2023-05-09:
nobody has a normal performing cariboulite on anything. software is sucky, support is non existant and github code activity is based on a developer who fixed some bugs in march, some bugs in april and cariboulite founder only integrated them after a large part of the community started raising their voice.

this project needs a developer who knows what he's doing and doesn't sell stories. I nominate @konimaru and I think we should all chip in and donate some funds to motivate him to bring us a working driver and software support.

jadware, 2023-06-21:
We are good on pi zero 2 w fully working using mostly default installation process.

edegraaff, 2023-06-21:
> We are good on pi zero 2 w fully working using mostly default installation process.

Performance also ok?  in that case i will give it a try

Alle1307, 2023-07-03:
My experience is that the installation crashes both on RPi-2Z and RPi-3A both with fresh 64bit OS. Hangs on SOAPY installation 5%.

sethmnone, 2024-10-16:
> We are good on pi zero 2 w fully working using mostly default installation process.

What OS build did you use?

Alle1307, 2024-12-08:
Used RPi-OS-Lite (Bullseye) for that. Tried also Bookworm, Soapy takes a long time but gets through. However SoapySDRUtil does not recognize CaribouLite. Back in the project-fridge.