cariboulabs/cariboulite #125: kfifo alloc error

great9, 2023-06-15:
After running caribou_dump1090 and getting a ModuleABI version 0.8.2 and library version 0.8.3 error I ran it a couple of more times. There's a kfifo alloc error.
did an rmmod and modprobe then caribou_dump1090 again. Here's the dmesg.
```
[129678.706230] smi-stream-dev soc:smi_dev: smi-stream-dev: smi-stream dev removed
[129684.614938] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=6, addr_dir_offset=2, addr_ch_offset=3)
[129684.615102] smi-stream-dev: creating a device and registering it with sysfs
[129684.615395] smi-stream-dev soc:smi_dev: initialised
[129687.145704] caribou_dump109: page allocation failure: order:10, mode:0x40cc0(GFP_KERNEL|__GFP_COMP), nodemask=(null),cpuset=/,mems_allowed=0
[129687.145744] CPU: 3 PID: 148727 Comm: caribou_dump109 Tainted: G         C OE     5.4.0-1086-raspi #97-Ubuntu
[129687.145748] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[129687.145752] Call trace:
[129687.145772]  dump_backtrace+0x0/0x1e0
[129687.145777]  show_stack+0x28/0x38
[129687.145787]  dump_stack+0xd8/0x134
[129687.145797]  warn_alloc+0x110/0x180
[129687.145802]  __alloc_pages_slowpath+0xbd4/0xc00
[129687.145807]  __alloc_pages_nodemask+0x2b4/0x330
[129687.145814]  kmalloc_order+0x34/0x80
[129687.145819]  kmalloc_order_trace+0x40/0x120
[129687.145826]  __kmalloc+0x358/0x398
[129687.145832]  __kfifo_alloc+0x5c/0xa8
[129687.145848]  smi_stream_open+0x80/0x1d8 [smi_stream_dev]
[129687.145854]  chrdev_open+0xac/0x1a8
[129687.145860]  do_dentry_open+0x11c/0x390
[129687.145867]  vfs_open+0x3c/0x48
[129687.145873]  do_last+0x17c/0x8c8
[129687.145877]  path_openat+0x88/0x260
[129687.145882]  do_filp_open+0x88/0x110
[129687.145887]  do_sys_open+0x188/0x2f0
[129687.145892]  __arm64_sys_openat+0x30/0x40
[129687.145898]  el0_svc_common.constprop.0+0x84/0x230
[129687.145902]  el0_svc_handler+0x38/0xa0
[129687.145908]  el0_svc+0x10/0x140
[129687.145913] Mem-Info:
[129687.145930] active_anon:9507 inactive_anon:16952 isolated_anon:0
                 active_file:103773 inactive_file:37793 isolated_file:12
                 unevictable:4205 dirty:77 writeback:0 unstable:0
                 slab_reclaimable:11523 slab_unreclaimable:13256
                 mapped:11773 shmem:9366 pagetables:691 bounce:0
                 free:27922 free_pcp:146 free_cma:11190
[129687.145944] Node 0 active_anon:38028kB inactive_anon:67808kB active_file:415092kB inactive_file:151172kB unevictable:16820kB isolated(anon):0kB isolated(file):48kB mapped:47092kB dirty:308kB writeback:0kB shmem:37464kB shmem_thp: 0kB shmem_pmdmapped: 0kB anon_thp: 0kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[129687.145959] DMA free:111688kB min:22528kB low:28160kB high:33792kB active_anon:37928kB inactive_anon:67808kB active_file:415416kB inactive_file:151072kB unevictable:16820kB writepending:308kB present:970752kB managed:927888kB mlocked:16820kB kernel_stack:3648kB pagetables:2764kB bounce:0kB free_pcp:584kB local_pcp:64kB free_cma:44760kB
[129687.145964] lowmem_reserve[]: 0 0 0 0
[129687.145979] DMA: 510*4kB (UMEC) 267*8kB (UMEC) 439*16kB (UMEC) 705*32kB (UMEC) 226*64kB (UMEC) 48*128kB (UMEC) 18*256kB (M) 22*512kB (MC) 19*1024kB (MC) 9*2048kB (MC) 1*4096kB (C) = 112224kB
[129687.146049] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=1048576kB
[129687.146056] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=32768kB
[129687.146063] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=2048kB
[129687.146070] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=64kB
[129687.146075] 153119 total pagecache pages
[129687.146087] 1895 pages in swap cache
[129687.146093] Swap cache stats: add 54384, delete 52489, find 45114/53100
[129687.146099] Free swap  = 2032380kB
[129687.146104] Total swap = 2097148kB
[129687.146109] 242688 pages RAM
[129687.146115] 0 pages HighMem/MovableOnly
[129687.146120] 10716 pages reserved
[129687.146125] 16384 pages cma reserved
[129687.146133] smi-stream-dev: error tx kfifo_alloc
```

meexmachina, 2023-06-17:
Can you modify please the 
fifo_mtu_multiplier=6
Parameter in /etc/modprobe.d

There is a ko loading directive in there for caribouLite.
This parameter defines the size of the kfifos allocated by the ko module. The actual size is the fifo_mtu_multiplier multiplied by the native mtu (half a megabyte*6=3 mb). In the rpizero model this number is automatically downscaled to 2. We may downscaled it for your model as well...

David

closed 2023-07-18 by meexmachina

great9, 2023-09-26:
```
######################################################
##       CaribouLite DUMP1090 - ADS-B Receiver      ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
09-26 21:08:18.902 19405 19405 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:209 FPGA already operational - not programming (use 'force_prog=true' to force update)
09-26 21:08:18.902 19405 19405 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:534 couldn't open smi driver file '/dev/smi' (No such file or directory)
09-26 21:08:18.902 19405 19405 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule
09-26 21:08:18.904 19405 19405 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
09-26 21:08:18.904 19405 19405 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
[ERROR] cariboulite_init_driver() failed
Printing 'findCariboulite' Request:
    {channel: HiF}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[9e02dae1], ChannelType: HiF
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
09-26 21:08:18.916 19405 19405 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
09-26 21:08:18.916 19405 19405 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:459 FPGA communication release failed (-1)
09-26 21:08:18.916 19405 19405 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
CaribouLite: Signal [11] received from pid=[968]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = 968
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation
```