cariboulabs/cariboulite #126: smi_stream_dev installing in wrong directory

kng, 2023-06-16:
There seems to be some issue with figuring out where to install the .ko
In /lib/modules I have `6.1.21+  6.1.21-v7+  6.1.21-v7l+  6.1.21-v8+`
Running kernel (arm_64bit=0): `Linux raspi-sat2 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
`
Running the install.sh (current main branch) a few times results in this:
```
$ find /lib/modules|grep smi_stream
/lib/modules/6.1.21-v7+/kernel/drivers/char/broadcom/smi_stream_dev.ko.xz
/lib/modules/6.1.21-v7+/kernel/drivers/char/broadcom/broadcom/smi_stream_dev.ko.xz
/lib/modules/6.1.21-v7+/kernel/drivers/char/broadcom/broadcom/broadcom/smi_stream_dev.ko.xz
/lib/modules/6.1.21-v7+/kernel/drivers/char/broadcom/broadcom/broadcom/broadcom/smi_stream_dev.ko.xz
/lib/modules/6.1.21-v7+/kernel/drivers/char/broadcom/broadcom/broadcom/broadcom/broadcom/smi_stream_dev.ko.xz
/lib/modules/6.1.21-v7+/kernel/drivers/char/broadcom/broadcom/broadcom/broadcom/broadcom/broadcom/smi_stream_dev.ko.xz
/lib/modules/6.1.21-v7+/kernel/drivers/char/broadcom/broadcom/broadcom/broadcom/broadcom/broadcom/broadcom/smi_stream_dev.ko.xz
```
While it should be in `/lib/modules/6.1.21-v7l+/kernel/drivers/char/broadcom/` ?
Inserting the module with full path works so the .ko itself is fine.

meexmachina, 2023-06-17:
@kng thanks for the issue. please try the latest commit for the fixed outdir.
couldn't reproduce the repetitive broadcom subdirs though... can you share the install log?

hudouseko, 2023-08-03:
Hi, I'm experiencing the same issue with module being installed into wrong /lib/modules subdirectory. I have identical kernel versions installed as @kng listed above on my RPi4. I'm running kernel (output of `uname -a`):
`Linux pi4ant2 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux`
while having the same kernel versions and appropriate modules installed as @kng has:
`root@pi4ant2:/lib/modules/6.1.21-v7+/kernel/drivers/char/broadcom# ls /lib/modules
6.1.21+  6.1.21-v7+  6.1.21-v7l+  6.1.21-v8+`
I think the issue is here in the driver/install.sh script: https://github.com/cariboulabs/cariboulite/blob/000ecd7aaa17dae7644171ef5b83c92c894d89bc/driver/install.sh#L48C25-L48C25

That find (`output_dir=$(find "/lib/modules" -type f -name "bcm2835_smi_dev*" -exec dirname {} \;)` will output multiple modules subdirectories in case you've got multiple kernels installed, but the following copy command
` sudo cp -r ${ROOT_DIR}/$BUILD_DIR/smi_stream_dev.ko.xz ${output_dir}/`
will only copy module to the first modules subdirectory from the list in `output_dir`. This happens to be in my case `/lib/modules/6.1.21-v7+`, while I'm running 6.1.21-v71+.

Moving the module manually to the correct modules subdirectory resolves the issue.