cariboulabs/cariboulite #129: cariboulite_radio.c:1048 SMI reading operation failed

saarbastler, 2023-07-17:
Hi all,

I have issues to get my Cariboue lite running. If I run the cariboulite_util : `sudo ./cariboulite_util -c 1 -f 90600000 -n 100 -`

07-17 18:13:12.442   948   948 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:562 Setup Modem state to 3 (tx prep)
07-17 18:13:12.445   948   948 D RFFC5072 rffc507x_print_stat@rffc507x.c:427 RFFC507X STAT: 0xDA7C PLL_LOCK: 1, CT_CAL: 90, KV_CAL: 31, CT_CAL_FAIL: 0
07-17 18:13:12.445   948   948 D CARIBOULITE Radio cariboulite_radio_set_frequency@cariboulite_radio.c:837 Frequency setting CH: 1, Wanted: 90600000.00 Hz, Set: 90599999.74 Hz (MOD: 2494999808.00, MIX: 2585599807.74)
07-17 18:13:12.445   948   948 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:863 Activating channel 1, dir = RX, activate = 0
07-17 18:13:12.445   948   948 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:561 Setup Modem state to 2 (trx off)
07-17 18:13:12.446   948   948 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:863 Activating channel 1, dir = RX, activate = 1
07-17 18:13:12.446   948   948 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:561 Setup Modem state to 2 (trx off)
07-17 18:13:12.446   948   948 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:562 Setup Modem state to 3 (tx prep)
07-17 18:13:12.447   948   948 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:885 Setup Modem state tx_prep
07-17 18:13:12.447   948   948 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:564 Setup Modem state to 5 (rx)
07-17 18:13:12.570   948   948 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1048 SMI reading operation failed
07-17 18:13:12.570   948   948 E CARIBOULITE Main main@cariboulite.c:320 Samples read operation failed. Quiting...
07-17 18:13:12.570   948   948 E CARIBOULITE Main main@cariboulite.c:328 Writing into file failed, exiting!

The self test of the test tool works once (after Program the FPGA):

`=====================================
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Digital I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [10]  Synthesizer 85-4200 MHz
 [99]  Quit
    Choice:   3

=====================================
FPGA Programming:
07-17 18:28:22.547   941   941 D CARIBOU_PROG caribou_prog_configure_prepare@caribou_prog.c:148 RESET low, Waiting for CDONE low
07-17 18:28:22.548   941   941 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
07-17 18:28:24.065   941   941 D CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:282 bitstream sent 32220 bytes
07-17 18:28:24.065   941   941 D CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
07-17 18:28:24.066   941   941 D CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:204 0 dummy clocks sent
07-17 18:28:24.066   941   941 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!

        FPGA programming successful, Versions:

=====================================
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Digital I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [10]  Synthesizer 85-4200 MHz
 [99]  Quit
    Choice:   4

=====================================
07-17 18:28:25.483   941   941 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:474 Testing modem communication and versions
07-17 18:28:25.483   941   941 D AT86RF215_Main at86rf215_print_version@at86rf215.c:290 MODEM Version: AT86RF215IQ (without basebands), version: 03
07-17 18:28:25.483   941   941 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:489 Testing mixer communication and versions
07-17 18:28:25.485   941   941 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:503 Self-test process finished successfully!

=====================================
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Digital I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [10]  Synthesizer 85-4200 MHz
 [99]  Quit
    Choice:   4

=====================================
07-17 18:28:26.671   941   941 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:474 Testing modem communication and versions
07-17 18:28:26.672   941   941 W AT86RF215_Main at86rf215_print_version@at86rf215.c:294 MODEM Version: not AT86RF215 IQ capable modem (product number: 0x0d, versions 03)
07-17 18:28:26.672   941   941 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:480 The assembled modem is not AT86RF215 / IQ variant (product number: 0x0d)
07-17 18:28:26.672   941   941 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:489 Testing mixer communication and versions
07-17 18:28:26.673   941   941 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:507 Self-test process finished with errors
`

I tested on a Raspberry PI 3B, 3B+ and 4B using Bullseye 64Bit 2023-05-03 and setup as described (git clone, install.sh)

Same result on all my tests.

Please advise how to get it running.

meexmachina, 2023-07-18:
can you make a photo of the modem on board? its on the back side under the "MODEM" label.
First impression it seems like the board is faulty. but it could be also an SPI problem. can you show the /boot/config.txt file please.

saarbastler, 2023-07-18:
Hi David,

here the photo.

the self-test only work once, then it does not find AT86RF215.

The cariboulite_util has this error:  948 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1048 SMI reading operation failed

SDR++ seems to work, but I could not receive anything. I ordered an antenna, I have to wait until it will be delivered.

Thanks a lot,
Jörg
 

[image]

fransla, 2023-08-25:
I also get SMI reading failed with SDRPP or self test

rush0815, 2023-09-19:
Gettin the same error.
Rpi 4B
Image: 2023-05-03-raspios-bullseye-arm64
Clean install as shown in README

`mkdir ~/projects
cd ~/projects
git clone https://github.com/cariboulabs/cariboulite
cd cariboulite
`
After that, reboot.

First two runs of caribou_dump1090:
`pi@raspberrypi:~/projects/cariboulite/examples/cpp/build $ sudo ./caribou_dump1090 
######################################################
##       CaribouLite DUMP1090 - ADS-B Receiver      ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
09-19 14:25:42.123  1637  1637 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:210 FPGA already operational - not programming (use 'force_prog=true' to force update)
Printing 'findCariboulite' Request:
    {channel: HiF}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[99da225b], ChannelType: HiF
[INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)
[INFO] Entering Reader Thread
[INFO] setFrequency dir: 1, channel: 0, freq: 1090000001.27
[INFO] setupStream: dir= RX, format= CS16

Running Soapy process with CaribouLite Config:
	Stream format: CS16
	Channel: HiF
	Sample size: 4 bytes
Starting stream loop, press Ctrl+C to exit...
09-19 14:25:42.453  1637  1640 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1048 SMI reading operation failed
reader thread failed to read SMI!

^Ccaribou_smi_read -> Timeout
[INFO] Leaving Reader Thread

pi@raspberrypi:~/projects/cariboulite/examples/cpp/build $ sudo ./caribou_dump1090 
######################################################
##       CaribouLite DUMP1090 - ADS-B Receiver      ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
09-19 14:26:01.481  1641  1641 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:210 FPGA already operational - not programming (use 'force_prog=true' to force update)
Printing 'findCariboulite' Request:
    {channel: HiF}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[99da225b], ChannelType: HiF
[INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)
[INFO] Entering Reader Thread
[INFO] setFrequency dir: 1, channel: 0, freq: 1090000001.27
[INFO] setupStream: dir= RX, format= CS16

Running Soapy process with CaribouLite Config:
	Stream format: CS16
	Channel: HiF
	Sample size: 4 bytes
Starting stream loop, press Ctrl+C to exit...
^Ccaribou_smi_read -> Timeout
[INFO] Leaving Reader Thread

^Cpi@raspberrypi:~/projects/cariboulite/examples/cpp/build `

I tried alredy, DragonOS, Ubuntu, Raspbian lite, RPI3 in all possible combinations. The result always ends with a SMI timout.

I tried starting
sudo SoapySDRServer --bind, which started without any problems. Then checking with gqrx or sdr++ the stream. After a few seconds I got some samples, but no signal in it. Just a random noise floor with a fat spike in the middle.  

Running caribout_test_app and receiving some samples seems so work, but only on the second run. The first run brings again some timeouts...

What can I do more at this point ?

ignat-mihai, 2024-01-24:
Same issues here: https://github.com/cariboulabs/cariboulite/issues/165#issuecomment-1908354606