cariboulabs/cariboulite #133: SIGSEGV - invalid way of a valid memory region

great9, 2023-07-29:
######################################################
##       CaribouLite DUMP1090 - ADS-B Receiver      ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
07-29 12:23:45.381 65314 65314 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:210 FPGA already operational - not programming (use 'force_prog=true' to force update)
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.8-3/libremoteSupport.so)
  remote failed ABI check: Library ABI=0.8-3, Module ABI=0.8-2
Printing 'findCariboulite' Request:
    {channel: HiF}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[9e02dae1], ChannelType: HiF
[INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)
[INFO] Entering Reader Thread
[INFO] setFrequency dir: 1, channel: 0, freq: 1090000001.27
[INFO] setupStream: dir= RX, format= CS16

Running Soapy process with CaribouLite Config:
        Stream format: CS16
        Channel: HiF
        Sample size: 4 bytes
Starting stream loop, press Ctrl+C to exit...
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
CaribouLite: Signal [11] received from pid=[-1873653760]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = -1873653760
   signal process uid = 65535
   signal status = 0
   signal errno / SIGSEGV / the process access a valid region of memory in an invalid way - violated memory access permissions
SIGSEGV: memory access violation

rjfrost000, 2023-07-30:
I'm getting the same error using a fresh install, just wanted to say that your not the only one to have this error.

jessupp, 2023-08-09:
I've also done a fresh install.

When I type sudo SoapySDRUtil --find, I am getting in red:

io_utils_spi_add_chip@io_utils_spi.c:434 spi_init function failed with code -1, (SPI_ERR_OPEN)

and 

08-09 16:25:23.605  1569  1569 E FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:204 Programming failed
08-09 16:25:23.606  1569  1569 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:628 FPGA programming failed
08-09 16:25:23.606  1569  1569 E IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:487 the device is already empty
08-09 16:25:23.607  1569  1569 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
[ERROR] cariboulite_init_driver() failed

The when I run the DAB application odr-dabmod I get:
Printing 'findCariboulite' Request:
    {Cariboulite: }
    {device_id: 1}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[d555744f], ChannelType: HiF
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
CaribouLite: Signal [11] received from pid=[968]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = 968
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation

kcman100, 2023-08-27:
Getting the same error here, trying to run the included sample ADS program.

ken@raspberrypi:~/projects2/cariboulite/examples/cpp/build $ sudo ./caribou_dump1090 
######################################################
##       CaribouLite DUMP1090 - ADS-B Receiver      ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
08-27 15:40:21.937  2014  2014 E IO_UTILS_SPI io_utils_spi_add_chip@io_utils_spi.c:434 spi_init function failed with code -1, (SPI_ERR_OPEN)
08-27 15:40:21.939  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.939  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.939  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.939  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.940  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.981  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.981  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.981  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.981  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:21.981  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:22.022  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:22.022  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:22.022  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:22.022  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:22.022  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:22.065  2014  2014 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
08-27 15:40:23.550  2014  2014 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!

08-27 15:40:23.672  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.672  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.672  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.672  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.672  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.713  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.713  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.713  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.713  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.713  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.755  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.755  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.755  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.755  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.755  2014  2014 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
08-27 15:40:23.796  2014  2014 E FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:204 Programming failed
08-27 15:40:23.796  2014  2014 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:628 FPGA programming failed
08-27 15:40:23.796  2014  2014 E IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:487 the device is already empty
08-27 15:40:23.798  2014  2014 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
[ERROR] cariboulite_init_driver() failed
Printing 'findCariboulite' Request:
    {channel: HiF}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[4f1bef61], ChannelType: HiF
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
CaribouLite: Signal [11] received from pid=[968]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = 968
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation

ImDroided, 2024-01-25:
Any fix yet for this? CubicSDR works fine and can tune the Caribou but just trying to start GQRX I get this mem access error.

great9, 2024-01-26:
> If you performed the install.sh script multiple times then your device probably won't work.

why is that? are there leftovers?