cariboulabs/cariboulite #135: Raspberry Pi 34B SMI error

loewal, 2023-08-07:
Hi, When transferring my micro-sd card and the hat from a perfectly functioning Raspberry Pi3B+ to a Pi4B, I get an SMI error:

/dev/smi no such file or directory...

Is the Pi4B not compatible or is there an error in the install-script?

loewal, 2023-08-07:
Recompiling doesn't help:
Installing UDEV rules...
Installation finished
Installation completed.
4. Main software...
[INFO] SoapyCaribouliteSession, sessionCount: 0
08-07 09:41:11.547  5102  5102 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:210 FPGA already operational - not programming (use 'force_prog=true' to force update)
08-07 09:41:11.575  5102  5102 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:534 couldn't open smi driver file '/dev/smi' (No such file or directory)
08-07 09:41:11.575  5102  5102 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule
08-07 09:41:11.585  5102  5102 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
08-07 09:41:11.585  5102  5102 E FPGA caribou_fpga_close@caribou_fpga.c:247 caribou_fpga_close: dev not initialized
[ERROR] cariboulite_init_driver() failed
08-07 09:41:11.593  5102  5102 E FPGA caribou_fpga_close@caribou_fpga.c:247 caribou_fpga_close: dev not initialized
08-07 09:41:11.593  5102  5102 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:459 FPGA communication release failed (-1)
08-07 09:41:11.594  5102  5102 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
[INFO] SoapyCaribouliteSession, sessionCount: 0
08-07 09:41:11.645  5135  5135 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:210 FPGA already operational - not programming (use 'force_prog=true' to force update)
08-07 09:41:11.667  5135  5135 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:534 couldn't open smi driver file '/dev/smi' (No such file or directory)
08-07 09:41:11.667  5135  5135 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule
08-07 09:41:11.669  5135  5135 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
08-07 09:41:11.669  5135  5135 E FPGA caribou_fpga_close@caribou_fpga.c:247 caribou_fpga_close: dev not initialized
[ERROR] cariboulite_init_driver() failed
08-07 09:41:11.679  5135  5135 E FPGA caribou_fpga_close@caribou_fpga.c:247 caribou_fpga_close: dev not initialized
08-07 09:41:11.679  5135  5135 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:459 FPGA communication release failed (-1)
08-07 09:41:11.679  5135  5135 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device

jadware, 2023-08-09:
we had the same issues, and I believe it's caused by the kernel header versions, have not been able to resolve without switching to DragonOS and preventing sudo update/upgrade

great9, 2023-09-25:
@meexmachina stopped development.

great9, 2023-09-27:
`[   54.327156] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=6, addr_dir_offset=2, addr_ch_offset=3)`
and

`[10360.049622] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)`

doesn't work on rpi3b

great9, 2023-09-27:
```
09-27 13:57:57.045 15774 15774 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
09-27 13:58:01.042 15774 15774 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!

09-27 13:58:01.143 15774 15774 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:534 couldn't open smi driver file '/dev/smi' (No such file or directory)
09-27 13:58:01.143 15774 15774 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule
09-27 13:58:01.145 15774 15774 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
09-27 13:58:01.145 15774 15774 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
[ERROR] cariboulite_init_driver() failed
Printing 'findCariboulite' Request:
    {channel: HiF}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[9e02dae1], ChannelType: HiF
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
09-27 13:58:01.181 15774 15774 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
09-27 13:58:01.181 15774 15774 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:459 FPGA communication release failed (-1)
09-27 13:58:01.181 15774 15774 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
CaribouLite: Signal [11] received from pid=[968]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = 968
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation
```

meexmachina, 2023-09-28:
Hi @great9 
it may be due to insufficient memory. the memory size given to the driver is governed by a parameter to the driver installation script "install.sh"
it has the following parameters:
./install.sh [mtu_mult] [dir_offset] [ch_offset]       # all are integers

dir_offs and ch_offs should be 2 and 3 as you mentioned. they are the memory address offsets.
the [mtu_mult] parameter is the number of MTUs to allocate and in the RPI4 this is set to 6 (6xMTU). in RPIzero it is set to 2 (2xMTU).

can you go to the /driver  directory and rerun the "install" script with

./install.sh 2 2 3

then it will allocate a smaller chunk.

The selection of this number is defined in the global installation that checks the RPI model. but the granularity of the selection may be not sufficient...

tell me what you see and it it works, we will define a better granularity for that parameter between different RPI models.
Thank you

great9, 2023-09-28:
> can you go to the /driver directory and rerun the "install" script with
> ./install.sh 2 2 3

Instead of the driver directory i went to the repo root and ran install.sh with those params there.
it resulted in
`
[  7  ] Adding modprobe configuration 6, 2, 3
`
 
> then it will allocate a smaller chunk.
>
> The selection of this number is defined in the global installation that checks the RPI model. but the granularity of the selection may be not sufficient... [...]

Then I went to the driver directory and ran
`
[  7  ] Adding modprobe configuration 6, 2, 3`

which told me I have to specify install or uninstall, so I did:

`./install.sh install 2 2 3`

and this resulted in:
```compiling module smi_stream_dev.ko...
Built target smi_stream_dev
The code filename is path: '../software/libcariboulite/src/caribou_smi/kernel', name: 'smi_stream_dev_gen', ext: 'h'

[  3  ] Installing into '/lib/modules/5.4.0-1094-raspi/kernel/drivers/char/broadcom
/lib/modules/5.4.0-1093-raspi/kernel/drivers/char/broadcom/broadcom
/lib/modules/5.4.0-1093-raspi/kernel/drivers/char/broadcom'
cp: '/lib/modules/5.4.0-1093-raspi/kernel/drivers/char/broadcom/broadcom' and '/lib/modules/5.4.0-1093-raspi/kernel/drivers/char/broadcom/broadcom' are the same file

[  4  ] Updating 'depmod'

[  5  ] Blacklisting original bcm2835_smi_dev module

[  6  ] Adding systemd configuration

[  7  ] Adding modprobe configuration 2, 2, 3

[  8  ] Adding UDEV rules
CaribouLite UDEV Rules (un)installation
=======================================

Installing UDEV rules...
Installation finished
Installation completed.```

great9, 2023-09-28:
and after a reboot let's check dmesg:
```
dmesg | grep smi
[   14.514800] pinctrl-bcm2835 3f200000.gpio: pin gpio7 already requested by 3f204000.spi; cannot claim for 3f600000.smi
[   14.534489] pinctrl-bcm2835 3f200000.gpio: pin-7 (3f600000.smi) status -22
[   14.562504] smi-bcm2835 3f600000.smi: Error applying setting, reverse things back
[   14.582809] smi-bcm2835: probe of 3f600000.smi failed with error -22
[   14.591387] smi_stream_dev: loading out-of-tree module taints kernel.
[   14.592216] smi_stream_dev: module verification failed: signature and/or required key missing - tainting kernel
[   14.594213] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   17.952158] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   18.180420] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   18.267013] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   18.301162] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   18.469937] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   19.600949] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   19.675866] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   53.642049] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
```

let's now try the dump1090 example app from cariboulite repo:
```
######################################################
##       CaribouLite DUMP1090 - ADS-B Receiver      ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
09-28 07:19:48.734  3555  3555 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
09-28 07:19:52.743  3555  3555 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!

09-28 07:19:52.844  3555  3555 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:534 couldn't open smi driver file '/dev/smi' (No such file or directory)
09-28 07:19:52.844  3555  3555 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule
09-28 07:19:52.846  3555  3555 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
09-28 07:19:52.847  3555  3555 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
[ERROR] cariboulite_init_driver() failed
Printing 'findCariboulite' Request:
    {channel: HiF}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[9e02dae1], ChannelType: HiF
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
09-28 07:19:52.883  3555  3555 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
09-28 07:19:52.883  3555  3555 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:459 FPGA communication release failed (-1)
09-28 07:19:52.883  3555  3555 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
CaribouLite: Signal [11] received from pid=[968]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = 968
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation
```

meexmachina, 2023-09-28:
see
[   14.514800] pinctrl-bcm2835 3f200000.gpio: pin gpio7 already requested by 3f204000.spi; cannot claim for 3f600000.smi
[   14.534489] pinctrl-bcm2835 3f200000.gpio: pin-7 (3f600000.smi) status -22

okay in the dmesg i see the problem. the /boot/config.txt file is faulty - there is an spi overlay enabled.
if you are running on raspbian there is a file called config.txt in /boot. if running ubuntu (dragon) it is located in 
/boot/firmware/config.txt

can you send it over? I will check that the config is okay. thanks

great9, 2023-09-28:
commented out #dtoverlay=spi1-3cs

meexmachina, 2023-09-28:
commenting out spi1-3cs is not a good idea, you need this SPI with all three chip selects.
you should comment out the spi0 `#dtparam=spi=on`
See the config.txt in my Raspbian....

The installation script (in the root directory) gives you the directives for that at its end. they are given as warnings.
The install script itself doesn't change the config.txt file itself from the sole reason that that config file is a "system-wide" config file and it should be touched with respect. Lets see if now it is okay.

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

## uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=82

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# Uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
dtparam=i2c_vc=on
dtoverlay=spi1-3cs

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on

# Automatically load overlays for detected cameras
#camera_auto_detect=1

# Automatically load overlays for detected DSI displays
#display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
#max_framebuffers=2

# Run in 64-bit mode
#arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
#arm_boost=1
#over_voltage=5
#arm_freq=1300
#core_freq=525
#gpu_freq=700

[all]
gpu_mem=92
```

meexmachina, 2023-09-28:
@great9 thanks for the last message. did you update the config.txt file?

great9, 2023-09-28:
> @great9 thanks for the last message. did you update the config.txt file?

I run ubuntu on rpi3b, config.txt includes cfg files from different directories. I updated it to have "dtoverlay=spi1-3cs" and we are back to 
```ubuntu@rpi3b:~$ SoapySDRUtil -probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] SoapyCaribouliteSession, sessionCount: 0
09-28 14:23:13.712  5014  5014 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
09-28 14:23:17.686  5014  5014 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!

09-28 14:23:17.787  5014  5014 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:534 couldn't open smi driver file '/dev/smi' (No such file or directory)
09-28 14:23:17.787  5014  5014 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule
09-28 14:23:17.789  5014  5014 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
09-28 14:23:17.789  5014  5014 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
[ERROR] cariboulite_init_driver() failed
Printing 'findCariboulite' Request:
[INFO] Initializing DeviceID: 0, Label: CaribouLite S1G[9e02dae0], ChannelType: S1G
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
09-28 14:23:17.952  5014  5014 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
09-28 14:23:17.952  5014  5014 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:459 FPGA communication release failed (-1)
09-28 14:23:17.952  5014  5014 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
CaribouLite: Signal [11] received from pid=[968]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = 968
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation```

meexmachina, 2023-09-28:
if you are in ubuntu. okay so the config file you are looking for is located in `/boot/firmware/config.txt`
is it a DragonOS? the config.txt that you have in /boot/config.txt is dummy in DragonOS.

check that this config file is okay according to what I sent before.

great9, 2023-09-28:
> if you are in ubuntu. okay so the config file you are looking for is located in `/boot/firmware/config.txt` 
/boot/firmware/usercfg.txt actually

> is it a DragonOS? the config.txt that you have in /boot/config.txt is dummy in DragonOS.
it's not dragonOS

> check that this config file is okay according to what I sent before.
checked. doesn't work.
with dtoverlay=spi1-3cs commented the firmware loads but no /dev/smi with it uncommented (active) it doesn't.

I will remind you that 4 months ago it worked after 3 months of trying.

meexmachina, 2023-09-28:
okay. can you tell me which kernel?
uname -a
do you have an image of that distro so as to try in the lab?
or maybe tell me where I'm getting to build it here.

other things to check:
do you see `/proc/device-tree/hat` - lets check is the default device tree is read from the board's EEPROM device (it contains a device tree inside it that starts smi)

great9, 2023-09-28:
> uname -a
```
Linux rpi3b 5.4.0-1094-raspi #105-Ubuntu SMP PREEMPT Tue Sep 12 08:48:48 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
```
> do you have an image of that distro so as to try in the lab?
```
ubuntu@rpi3b:~$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.6 LTS"
```

It's not available on ubuntu official website

> do you see /proc/device-tree/hat - lets check is the default device tree is read from the board's EEPROM device (it contains a device tree inside it that starts smi)
```
ubuntu@rpi3b:~$ sudo cat /proc/device-tree/hat
cat: /proc/device-tree/hat: Is a directory
ubuntu@rpi3b:~$ sudo cat /proc/device-tree/
#address-cells        axi/                  fixedregulator_5v0/   memreserve            soc/
#size-cells           chosen/               framebuffer@3eaf0000/ model                 system/
__overrides__/        clocks/               hat/                  name                  thermal-zones/
__symbols__/          compatible            interrupt-parent      phy/                  timer/
aliases/              cpus/                 leds/                 reserved-memory/      
arm-pmu/              fixedregulator_3v3/   memory@0/             serial-number     
```

meexmachina, 2023-09-28:
Okay thanks for that. It seems that all is goodb except the fact that you are using spi0 device. 

And spi0 uses gpio7 which contradict the usage of smi. That's why smi is not loading. 

Now you say that spi0 is not requested in the usercng.txt file. And 'spi=on' is not written there...

That means there is another device driver that requests specifically spi0. Does is ring a bell? 

If you can it would be very helpful to help you if you could show the usercfg.txt file... if it is possible on your side.

meexmachina, 2023-09-28:
Thank I'll appreciate it. In addition I need to add your exceptional distro to the install script... thanks

meexmachina, 2023-09-28:
thanks a lot, 
according to the [https://www.raspberrypi.com/documentation/computers/config_txt.html#enable_uart]() the enabling of uart causes GPIOs 14 and 15 to be used by UART. GPIO 14 is used as  SMI_D6 and GPIO15 is used for SMI_D7
see schematics
[image]

So the enabling of UART may cause by itself a problem. I will add this exception to the install shell script.

but here:
```
ubuntu@rpi3b:~$ sudo cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-53, parent: platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-7   (                    |spi0 CS1            ) out hi ACTIVE LOW
 gpio-8   (                    |spi0 CS0            ) out hi ACTIVE LOW
...

ubuntu@rpi3b:~$ sudo rmmod smi_stream_dev
ubuntu@rpi3b:~$ sudo cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-53, parent: platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-7   (                    |spi0 CS1            ) out hi ACTIVE LOW
 gpio-8   (                    |spi0 CS0            ) out hi ACTIVE LOW
```

and here:
```
ubuntu@rpi3b:~$ dmesg | grep spi
[    0.000000] Linux version 5.4.0-1094-raspi (buildd@bos01-arm64-019) (gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)) #105-Ubuntu SMP PREEMPT Tue Sep 12 08:48:48 UTC 2023 (Ubuntu 5.4.0-1094.105-raspi 5.4.246)
[    1.745844] spi-bcm2835 3f204000.spi: could not get clk: -517
[    2.153472] usb usb1: Manufacturer: Linux 5.4.0-1094-raspi dwc_otg_hcd
[   14.708154] pinctrl-bcm2835 3f200000.gpio: pin gpio7 already requested by 3f204000.spi; cannot claim for 3f600000.smi
ubuntu@rpi3b:~$ dmesg | grep smi
[   14.708154] pinctrl-bcm2835 3f200000.gpio: pin gpio7 already requested by 3f204000.spi; cannot claim for 3f600000.smi
[   14.726030] pinctrl-bcm2835 3f200000.gpio: pin-7 (3f600000.smi) status -22
[   14.744869] smi-bcm2835 3f600000.smi: Error applying setting, reverse things back
[   14.752886] smi-bcm2835: probe of 3f600000.smi failed with error -22
[   14.774377] smi_stream_dev: loading out-of-tree module taints kernel.
[   14.775139] smi_stream_dev: module verification failed: signature and/or required key missing - tainting kernel
[   14.777140] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   17.664442] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
```

it seems like there is an spi device loading. But your config files don't load SPI:
```
#dtparam=spi=on
```

I would like to recreate your sdcard at our lab and understand what is happening. in the meantime please check with the following configs:

/boot/config.txt    - this shouldn't even be here right? as its an ubuntu distro... but if necessity called you may comment the uart line
```
ubuntu@rpi3b:~$ cat /boot/config.txt 
#dtparam=audio=on
#dtparam=i2c_arm=off
#dtparam=spi=off
#dtparam=i2c_vc=on
#dtoverlay=spi1-3cs
```

/boot/firmware/config.txt    - this we won't touch as they request
```
# Please DO NOT modify this file; if you need to modify the boot config, the
# "usercfg.txt" file is the place to include user changes. Please refer to
# the README file for a description of the various configuration files on
# the boot partition.

# The unusual ordering below is deliberate; older firmwares (in particular the
# version initially shipped with bionic) don't understand the conditional
# [sections] below and simply ignore them. The Pi4 doesn't boot at all with
# firmwares this old so it's safe to place at the top. Of the Pi2 and Pi3, the
# Pi3 uboot happens to work happily on the Pi2, so it needs to go at the bottom
# to support old firmwares.

[pi4]
kernel=uboot_rpi_4.bin
max_framebuffers=2

[pi2]
kernel=uboot_rpi_2.bin

[pi3]
kernel=uboot_rpi_3.bin

[all]
arm_64bit=1
device_tree_address=0x03000000

# The following settings are "defaults" expected to be overridden by the
# included configuration. The only reason they are included is, again, to
# support old firmwares which don't understand the "include" command.

enable_uart=1  # <===========
cmdline=cmdline.txt

include syscfg.txt
include usercfg.txt
```

/boot/firmware/syscfg.txt
```
# This file is intended to be modified by the pibootctl utility. User
# configuration changes should be placed in "usercfg.txt". Please refer to the
# README file for a description of the various configuration files on the boot
# partition.

#enable_uart=1            # <===========
#dtparam=audio=on
#dtparam=i2c_arm=on
#dtparam=spi=on
cmdline=cmdline.txt
```

/boot/firmware/usercfg.txt    - this is the only legitimate change location. I suggest checking the following.
```
#dtparam=audio=on
dtparam=i2c_arm=off
dtparam=spi=off    # <=========== uncomment
dtparam=i2c_vc=on
dtoverlay=spi1-3cs
enable_uart=0    # <=========== set to zero
arm_freq=550
core_freq=250
gpu_mem=16
```

great9, 2023-09-28:
> /boot/config.txt - this shouldn't even be here right? as its an ubuntu distro...

I created it because I didn't want to remove a bunch of checks from your install.sh 6 months ago.

```
ubuntu@rpi3b:~$ sudo dmesg | grep smi
[   14.720055] smi-bcm2835 3f600000.smi: initialised
[   14.727422] smi_stream_dev: loading out-of-tree module taints kernel.
[   14.728186] smi_stream_dev: module verification failed: signature and/or required key missing - tainting kernel
[   14.730025] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   14.730216] smi-stream-dev: creating a device and registering it with sysfs
[   14.730625] smi-stream-dev soc:smi_dev: initialised
ubuntu@rpi3b:~$ SoapySDRUtil -probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] SoapyCaribouliteSession, sessionCount: 0
09-28 19:48:44.451  3581  3581 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
09-28 19:48:48.699  3581  3581 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!

Printing 'findCariboulite' Request:
[INFO] Initializing DeviceID: 0, Label: CaribouLite S1G[9e02dae0], ChannelType: S1G
[INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)
[INFO] Entering Reader Thread

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=Cariboulite
  hardware=Cariboulite Rev2.8
  device_id=0
  fpga_revision=1
  hardware_revision=0x0001
  product_name=CaribouLite RPI Hat
  serial_number=1325493616
  vendor_name=CaribouLabs LTD

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CS8, CF32, CF64
  Native format: CS16 [full-scale=4095]
  Antennas: TX/RX Sub1GHz
  Full gain range: [0, 69] dB
    Modem AGC gain range: [0, 69] dB
  Full freq range: [389.5, 510], [779, 1020] MHz
    RF freq range: [389.5, 510], [779, 1020] MHz
  Sample rates: 4, 2, 1.333, 1, 0.8, 0.666, 0.5, 0.4 MSps
  Filter bandwidths: 0.02, 0.05, 0.1, 0.2, 0.25, 1, 1.25, 1.5625, 2, 2.5 MHz
  Sensors: RSSI, ENERGY, PLL_LOCK_MODEM
     * RSSI (RX RSSI):[-127, 4] 0.000000
        Modem level RSSI measurment
     * ENERGY (RX ENERGY):[-127, 4] 0.000000
        Modem level ENERGY (EDC) measurment
     * PLL_LOCK_MODEM (PLL Lock Modem): 1.000000
        Modem PLL locking indication

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: NO
  Stream formats: CS16, CS8, CF32, CF64
  Native format: CS16 [full-scale=4095]
  Antennas: TX/RX Sub1GHz
  Full gain range: [0, 31] dB
    Modem PA gain range: [0, 31] dB
  Full freq range: [389.5, 510], [779, 1020] MHz
    RF freq range: [389.5, 510], [779, 1020] MHz
  Sample rates: 4, 2, 1.333, 1, 0.8, 0.666, 0.5, 0.4 MSps
  Filter bandwidths: 0.08, 0.1, 0.125, 0.16, 0.2, 0.4, 0.5, 0.625, 0.8, 1 MHz
  Sensors: PLL_LOCK_MODEM
     * PLL_LOCK_MODEM (PLL Lock Modem): 1.000000
        Modem PLL locking indication

[INFO] Leaving Reader Thread
```

winner, winner... chicken dinner?

great9, 2023-09-28:
ok tomorrow, rock5b installation on 

```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.6 LTS"
```

meexmachina, 2023-09-28:
@great9 happy to see you happy. So just to know: the reason was uart=1?
I will update the install script to contain it as well if you confirm,
David

great9, 2023-09-28:
Confirmed. on rpi3b Ubuntu focal enable_uart should be 0.

great9, 2023-10-01:
found another bug, might be caribou-driver related, might be sdrpp related:
```
[165966.805798] smi-stream-dev soc:smi_dev: Set STREAMING_STATUS = 1, cur_addr = 4
[165966.828426] smi-stream-dev soc:smi_dev: Set STREAMING_STATUS = 0, cur_addr = 4
[165967.848107] smi-stream-dev soc:smi_dev: Reader DMA bounce timed out
[165977.408542] smi-stream-dev soc:smi_dev: Set STREAMING_STATUS = 2, cur_addr = 12
[165981.957200] smi-stream-dev soc:smi_dev: Set STREAMING_STATUS = 0, cur_addr = 12
[165982.075051] smi-bcm2835 3f600000.smi: smi_init_programmed_read returned -2
[165982.079989] smi-stream-dev soc:smi_dev: stream_smi_user_dma returned illegal count = 0, buff_num = 1
[165986.201084] smi-stream-dev soc:smi_dev: Set STREAMING_STATUS = 2, cur_addr = 12
[166025.394792] smi-stream-dev soc:smi_dev: Set STREAMING_STATUS = 0, cur_addr = 12
[166075.531848] smi-stream-dev soc:smi_dev: smi_stream_release: closing device: 0
[166075.531920] smi-stream-dev soc:smi_dev: Left reader thread
[166075.544506] smi-stream-dev soc:smi_dev: Left writer thread
[239690.611625] sdrpp: page allocation failure: order:8, mode:0x40cc0(GFP_KERNEL|__GFP_COMP), nodemask=(null),cpuset=/,mems_allowed=0
[239690.611682] CPU: 0 PID: 271102 Comm: sdrpp Tainted: G         C OE     5.4.0-1094-raspi #105-Ubuntu
[239690.611689] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[239690.611695] Call trace:
[239690.611714]  dump_backtrace+0x0/0x1e0
[239690.611724]  show_stack+0x28/0x38
[239690.611735]  dump_stack+0xd8/0x134
[239690.611749]  warn_alloc+0x110/0x180
[239690.611760]  __alloc_pages_slowpath+0xbd4/0xc00
[239690.611772]  __alloc_pages_nodemask+0x2b4/0x330
[239690.611783]  kmalloc_order+0x34/0x80
[239690.611794]  kmalloc_order_trace+0x40/0x120
[239690.611805]  __kmalloc+0x358/0x398
[239690.611816]  __kfifo_alloc+0x5c/0xa8
[239690.611835]  smi_stream_open+0x5c/0x1e0 [smi_stream_dev]
[239690.611846]  chrdev_open+0xac/0x1a8
[239690.611860]  do_dentry_open+0x11c/0x390
[239690.611871]  vfs_open+0x3c/0x48
[239690.611882]  do_last+0x17c/0x8c8
[239690.611891]  path_openat+0x88/0x260
[239690.611900]  do_filp_open+0x88/0x110
[239690.611910]  do_sys_open+0x188/0x2f0
[239690.611921]  __arm64_sys_openat+0x30/0x40
[239690.611932]  el0_svc_common.constprop.0+0x84/0x230
[239690.611940]  el0_svc_handler+0x38/0xa0
[239690.611950]  el0_svc+0x10/0x140
[239690.611969] Mem-Info:
[239690.612002] active_anon:6573 inactive_anon:19284 isolated_anon:0
                 active_file:105651 inactive_file:40836 isolated_file:44
                 unevictable:4205 dirty:14 writeback:0 unstable:0
                 slab_reclaimable:14218 slab_unreclaimable:10525
                 mapped:10537 shmem:9305 pagetables:688 bounce:0
                 free:23714 free_pcp:131 free_cma:7120
[239690.612029] Node 0 active_anon:26292kB inactive_anon:77136kB active_file:422604kB inactive_file:163444kB unevictable:16820kB isolated(anon):0kB isolated(file):48kB mapped:42148kB dirty:56kB writeback:0kB shmem:37220kB shmem_thp: 0kB shmem_pmdmapped: 0kB anon_thp: 0kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[239690.612059] DMA free:94856kB min:22528kB low:28160kB high:33792kB active_anon:26292kB inactive_anon:77136kB active_file:422604kB inactive_file:163352kB unevictable:16820kB writepending:56kB present:970752kB managed:927892kB mlocked:16820kB kernel_stack:3632kB pagetables:2752kB bounce:0kB free_pcp:524kB local_pcp:12kB free_cma:28480kB
[239690.612069] lowmem_reserve[]: 0 0 0 0
[239690.612101] DMA: 691*4kB (UMEC) 4739*8kB (UMEC) 1340*16kB (UMEC) 490*32kB (UMEC) 149*64kB (UMEC) 22*128kB (C) 20*256kB (C) 1*512kB (C) 0*1024kB 0*2048kB 0*4096kB = 95780kB
[239690.612215] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=1048576kB
[239690.612230] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=32768kB
[239690.612245] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=2048kB
[239690.612259] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=64kB
[239690.612271] 159380 total pagecache pages
[239690.612291] 1976 pages in swap cache
[239690.612305] Swap cache stats: add 4379685, delete 4377802, find 617016/1904705
[239690.612317] Free swap  = 2036220kB
[239690.612328] Total swap = 2097148kB
[239690.612340] 242688 pages RAM
[239690.612351] 0 pages HighMem/MovableOnly
[239690.612363] 10715 pages reserved
[239690.612374] 16384 pages cma reserved
[239690.612389] smi-stream-dev: error rx kfifo_alloc
```

meexmachina, 2023-10-01:
memory allocation error in the kfifo.
I writing a native support for sdrpp & gnuradio right now (c and cpp) without the soapy api

whats that?
> allows sdrpp --server again

if you want please contact me on cariboulabs.co@gmail.com

mvmn, 2023-11-28:
I have a Pi 4 with fresh DragonOS install - and get the same spi issue
```
11-28 08:28:02.989  7920  7920 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:443 couldn't open smi driver file '/dev/smi'                                                                                          
11-28 08:28:02.989  7920  7920 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule 
```
I've installed CaribouLite software from the develop_R1 branch

I've tried adding `enable_uart=0` to the `/boot/firmware/config.txt` - but somehow it broke network connections and everything stopped working, even with the line commented out. I had to remove it to get things working again.

1n4001, 2023-12-02:
> found another bug, might be caribou-driver related, might be sdrpp related:
>
> ```
> [165966.805798] smi-stream-dev soc:smi_dev: Set STREAMING_STATUS = 1, cur_addr = 4
> [165966.828426] smi-stream-dev [...]

What worked for me was to change the multiplier to 2 from 6  in /etc/modprobe.d/smi_stream_mod_cariboulite.conf  -- I'm running Pi4 8Gb module w/ v7l kernel.  I was able to use SoapySDR server to CubicSDR and get data.

K7MDL2, 2023-12-02:
On my working system here, I tried changing the multiplier form 2 ort 6, both work the same for me.  I am on bullseye 64 bit, same hardware with a USB3 SSDR boot drive.   

Spotting that gpio pin 2 error, I am often suspicious of other programs grabbing GPIO pins (like fan control).  For a different program I developed using GPIO pins, I had to set the GPIO direction and state immediately for one pin just before the program startup. Setting it in config.txt was too early.  Never did figure out what program was claiming it on bootup.