cariboulabs/cariboulite #137: MODEM Version: not AT86RF215 IQ capable modem

rush0815, 2023-09-04:
I am trying to get SoapySDRemote working with GQRX but all I ever got was a constant waterfall. Frequency changing did not change anything. After that I ran cariboulite_test_app and got:

`09-04 22:50:26.864  1959  1959 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:474 Testing modem communication and versions
09-04 22:50:26.864  1959  1959 W AT86RF215_Main at86rf215_print_version@at86rf215.c:294 MODEM Version: not AT86RF215 IQ capable modem (product number: 0x0d, versions 03)
09-04 22:50:26.864  1959  1959 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:480 The assembled modem is not AT86RF215 / IQ variant (product number: 0x0d)
09-04 22:50:26.864  1959  1959 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:489 Testing mixer communication and versions
09-04 22:50:26.867  1959  1959 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:507 Self-test process finished with errors`

So the drivers are not for the current assembled modem? Or what else can I do?

Kind regards

synthetikspace, 2023-09-04:
I get the exact same thing. 
I just bought another Pizero2 and another Pi4 board to make sure. This project has driven my crazy trying to get it to work. Hours, days lost. Not to mention I still think I have the wrong board. I backed this project when it launched and ordered the full version only to receive the SMI version. Then I was sent a 'replacement' from the project developer directly and no support other than attitude from Crowd Supply. Does this even work? All of my other SDRs just work. Hackrf, nano3, sdr-rtl etc. I feel like I have been scammed.

synthetikspace, 2023-09-04:
Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Digital I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [10]  Synthesizer 85-4200 MHz
 [99]  Quit
    Choice:   4

=====================================
09-04 23:25:12.401  1493  1493 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:474 Testing modem communication and versions
09-04 23:25:12.401  1493  1493 W AT86RF215_Main at86rf215_print_version@at86rf215.c:294 MODEM Version: not AT86RF215 IQ capable modem (product number: 0x0d, versions 03)
09-04 23:25:12.401  1493  1493 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:480 The assembled modem is not AT86RF215 / IQ variant (product number: 0x0d)
09-04 23:25:12.401  1493  1493 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:489 Testing mixer communication and versions
09-04 23:25:12.404  1493  1493 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:507 Self-test process finished with errors

rush0815, 2023-09-04:
I got mine today. the whole installation process went very well.. only a few warning according to python  and doxygen but I think as long that are warning, stuff should work.

running
sudo ./cariboulite_util -c 1 -f 1090e6 -

bring up some symbols to screen, but the tool exits very quick. Looks like there are no more samples and it simply crashes...

Mr. Caribou...  what do we have to do ?

jh4xsy, 2023-09-08:
I got SoapySDRserver working with GQRX.
I use the AppImage with SoapyRemote is packaged in.

[image]

After that I ran cariboulite_test_app and got the same result:
```
27850 27850 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:474 Testing modem communication and versions
27850 27850 W AT86RF215_Main at86rf215_print_version@at86rf215.c:294 MODEM Version: not AT86RF215 IQ capable modem (product number: 0x0d, versions 03)
27850 27850 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:480 The assembled modem is not AT86RF215 / IQ variant (product number: 0x0d)
27850 27850 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:489 Testing mixer communication and versions
27850 27850 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:507 Self-test process finished with errors
```

rush0815, 2023-09-09:
https://www.youtube.com/watch?v=YVl0995zDS4&ab_channel=cemaxecuter

look at minute 12:27. This could help. I have not tested it yet

OxMarco, 2024-01-24:
I am getting the same error, whatever software linked to CaribouLite I execute, the incompatible modem error arises.
The board is brand new (faulty), running on a RPi4 8GB with Raspian bookworm.

```
E CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1077 PLL didn't lock
E CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1077 PLL didn't lock
W AT86RF215_Main at86rf215_print_version@at86rf215.c:294 MODEM Version: not AT86RF215 IQ capable modem (product number: 0x00, versions 00)
E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:486 The assembled modem is not AT86RF215 / IQ variant (product number: 0x00)
E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:513 Self-test process finished with errors
I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
[ERROR] cariboulite_init_driver() failed
E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:465 FPGA communication release failed (-1)
E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
```

unixpunk, 2024-01-24:
sudo SoapySDRUtil --probe
full output?

OxMarco, 2024-01-24:
```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] SoapyCaribouliteSession, sessionCount: 0
01-24 20:53:28.873  2599  2599 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
01-24 20:53:30.717  2599  2599 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!
01-24 20:53:30.942  2599  2599 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
01-24 20:53:32.782  2599  2599 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!
01-24 20:53:33.008  2599  2599 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
01-24 20:53:34.796  2599  2599 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!
01-24 20:53:35.230  2599  2599 E CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1077 PLL didn't lock
01-24 20:53:35.236  2599  2599 E CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1077 PLL didn't lock
01-24 20:53:35.244  2599  2599 W AT86RF215_Main at86rf215_print_version@at86rf215.c:294 MODEM Version: not AT86RF215 IQ capable modem (product number: 0x00, versions 00)
01-24 20:53:35.244  2599  2599 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:486 The assembled modem is not AT86RF215 / IQ variant (product number: 0x00)
01-24 20:53:35.246  2599  2599 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:500 The assembled mixer is not RFFC5071/2[A]
01-24 20:53:35.246  2599  2599 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:513 Self-test process finished with errors
01-24 20:53:35.248  2599  2599 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
[ERROR] cariboulite_init_driver() failed
Printing 'findCariboulite' Request:
[INFO] Initializing DeviceID: 0, Label: CaribouLite S1G[e387d644], ChannelType: S1G
[INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)
01-24 20:53:35.383  2599  2599 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
01-24 20:53:35.384  2599  2599 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
01-24 20:53:35.384  2599  2599 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
01-24 20:53:35.384  2599  2599 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=Cariboulite
  hardware=Cariboulite Rev2.8
  device_id=0
  fpga_revision=1
  hardware_revision=0x0001
  product_name=CaribouLite RPI Hat
  serial_number=1908665122
  vendor_name=CaribouLabs LTD

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CS8, CF32, CF64
  Native format: CS16 [full-scale=4095]
  Antennas: TX/RX Sub1GHz
  Full gain range: [0, 69] dB
    Modem AGC gain range: [0, 69] dB
  Full freq range: [389.5, 510], [779, 1020] MHz
    RF freq range: [389.5, 510], [779, 1020] MHz
  Sample rates: 4, 2, 1.333, 1, 0.8, 0.666, 0.5, 0.4 MSps
  Filter bandwidths: 0.02, 0.05, 0.1, 0.2, 0.25, 1, 1.25, 1.5625, 2, 2.5 MHz
  Sensors: RSSI, ENERGY, PLL_LOCK_MODEM
     * RSSI (RX RSSI):[-127, 4] -1.000000
        Modem level RSSI measurment
     * ENERGY (RX ENERGY):[-127, 4] 0.000000
        Modem level ENERGY (EDC) measurment
     * PLL_LOCK_MODEM (PLL Lock Modem): 0.000000
        Modem PLL locking indication

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: NO
  Stream formats: CS16, CS8, CF32, CF64
  Native format: CS16 [full-scale=4095]
  Antennas: TX/RX Sub1GHz
  Full gain range: [0, 31] dB
    Modem PA gain range: [0, 31] dB
  Full freq range: [389.5, 510], [779, 1020] MHz
    RF freq range: [389.5, 510], [779, 1020] MHz
  Sample rates: 4, 2, 1.333, 1, 0.8, 0.666, 0.5, 0.4 MSps
  Filter bandwidths: 0.08, 0.1, 0.125, 0.16, 0.2, 0.4, 0.5, 0.625, 0.8, 1 MHz
  Sensors: PLL_LOCK_MODEM
     * PLL_LOCK_MODEM (PLL Lock Modem): 0.000000
        Modem PLL locking indication

01-24 21:53:35.385  2599  2599 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
01-24 21:53:35.385  2599  2599 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:465 FPGA communication release failed (-1)
01-24 21:53:35.385  2599  2599 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
```

unixpunk, 2024-01-25:
config.txt have these?

#dtparam=i2c_arm=on
#dtparam=spi=on
dtparam=i2c_vc=on
dtoverlay=spi1-3cs

or like this (both are valid)

dtparam=i2c_arm=off
dtparam=spi=off
dtparam=i2c_vc=on
dtoverlay=spi1-3cs

OxMarco, 2024-01-25:
Confirm the second. I would opt for a hardware issue but the board is literally brand new

unixpunk, 2024-01-25:
does uname -r show a 6.5 kernel?  if so, you can clone my fork that has a fix for compiling the kernel module and rerun the install.sh.  https://github.com/unixpunk/cariboulite

unixpunk, 2024-01-25:
might also try apt update;apt upgrade;reboot if you haven't updated things in a while and then re-run the install.sh again and reboot before testing again.  all of this stuff needs a reboot after its performed; prior to testing.  hth

unixpunk, 2024-01-25:
i can also say that i haven't tried this on raspbian, i have success using DragonOS.

OxMarco, 2024-01-25:
I am using DragonOS now. I installed _cariboulite_ drivers, rebooted but I still get pretty much the same error:

```
ubuntu@ubuntu:~$ sudo SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] SoapyCaribouliteSession, sessionCount: 0
01-25 07:03:36.783 17076 17076 I HAT hat_detect_board@hat.c:619 This board is not configured yet as a hat.
01-25 07:03:36.783 17076 17076 I EEPROM_UTILS eeprom_init_device@eeprom_utils.c:37 i2c-0 has been found successfully
01-25 07:03:36.783 17076 17076 I EEPROM_UTILS eeprom_init_device@eeprom_utils.c:56 trying to modprobe at24
01-25 07:03:36.935 17076 17076 I EEPROM_UTILS eeprom_init_device@eeprom_utils.c:100 EEPROM on addr 0x50 probing successful
01-25 07:03:36.935 17076 17076 E EEPROM_UTILS eeprom_read@eeprom_utils.c:169 The eeprom driver for bus 0, adde 0x50 is not initialized
01-25 07:03:36.935 17076 17076 E HAT hat_init@hat.c:532 Reading from eeprom failed
01-25 07:03:36.935 17076 17076 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:604 Failed to detect the board in /proc/device-tree/hat - EEPROM needs to be configured.
01-25 07:03:36.935 17076 17076 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:605 Please run the cariboulite_prod application with sudo permissions...
[ERROR] cariboulite_init_driver() failed
Printing 'findCariboulite' Request:
01-25 07:03:38.097 17076 17352 I HAT hat_detect_board@hat.c:619 This board is not configured yet as a hat.
shm_open: No such file or directory
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.1.0.5-0-unknown
[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed
Error probing device: SoapySDR::Device::make() no driver specified and no enumeration results
```

OxMarco, 2024-01-25:
My bad, compilation of one of the drivers failed
```
/home/ubuntu/Desktop/cariboulite/driver/build/smi_stream_dev.c: In function ‘smi_stream_dev_probe’:
./include/linux/export.h:29:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   29 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *                            ^~~~~~~~~~~
/home/ubuntu/Desktop/cariboulite/driver/build/smi_stream_dev.c:1319:28: error: too many arguments to function ‘class_create’
 1319 |         smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
```

unixpunk, 2024-01-26:
if you use my fork that won't happen, otherwise just edit line 1319 in cariboulite/driver/build/smi_stream_dev.c and remove "THIS_MODULE, " so it looks like: smi_stream_class = class_create(DEVICE_NAME);

You will also need to make sure config.txt is good and that you manually add your user to dialout and root groups.  then after running install.sh, reboot and give it another shot.

GastonMazzei, 2024-02-18:
> config.txt have these?
> 
> #dtparam=i2c_arm=on #dtparam=spi=on dtparam=i2c_vc=on dtoverlay=spi1-3cs
> 
> or like this (both are valid)
> 
> dtparam=i2c_arm=off dtparam=spi=off dtparam=i2c_vc=on dtoverlay=spi1-3cs

The spi1-3cs solved this problem in my case, but now `sudo SoapySDRUtil --probe` only shows the sub1Ghz path. 
I'm suspecting the rough handling of the other port has caused some hardware damage, I don't mean to spam this thread but if I keep failing to detect it I will open a new Issue. **Could it be that  there is some overlay missing? Or the driver was installed but only partially?** Thx.

unixpunk, 2024-02-19:
That's expected, open a SDR app and check it.