cariboulabs/cariboulite #141: dump1090 doesn't decodde

jh4xsy, 2023-10-07:
I  run caribou_dump1090 (drivers commits at Oct 2, 2023), but NO decode.
By using the same ANT, I can decode ADS-B by using dump1090-fa, RTLSDR. 

```
$ caribou_dump1090 
######################################################
##       CaribouLite DUMP1090 - ADS-B Receiver      ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
10-07 11:50:45.634  1353  1353 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:209 FPGA already operational - not programming (use 'force_prog=true' to force update)
Printing 'findCariboulite' Request:
    {channel: HiF}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[795f3bbf], ChannelType: HiF
[INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)
[INFO] Entering Reader Thread
[INFO] setFrequency dir: 1, channel: 0, freq: 1090000001.27
[INFO] setupStream: dir= RX, format= CS16

Running Soapy process with CaribouLite Config:
	Stream format: CS16
	Channel: HiF
	Sample size: 4 bytes
Starting stream loop, press Ctrl+C to exit...
```
My system is Raspberry Pi OS Lite (Legacy) & Pi2B.

meexmachina, 2023-10-07:
can you show the setup?

 

 
10-07 11:50:45.634  1353  1353 I FPGA ***@***.***_fpga.c:209 FPGA already operational - not programming (use 'force_prog=true' to force update)

jh4xsy, 2023-10-07:
My setup is shown below.
```
$ cd cariboulite
$ ./install.sh
$ cd examples/cpp
$ mkdir build
$ cmake ..
$ build
```
/boot/config.txt
```
dtparam=i2c_vc=on
dtoverlay=spi1-3cs

camera_auto_detect=1
display_auto_detect=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_overscan=1

[cm4]
otg_mode=1

[all]

[pi4]
arm_boost=1

[all]
enable_uart=0
```

jh4xsy, 2023-10-10:
OK David.

My RF equipment is ６ element Yagi ANT for 70cm band.  NO LNA and SAW filter. 

As I wrote before, this equipment can decode ADS-B with dump1090-fa and RTLSDR.
And I reconfirmed that the ANT cable is connected to HiF ANT port of CaribouLite.

ImDroided, 2024-01-27:
I think the I/Q is swapped and dump1090 has no option to reverse it.

jh4xsy, 2024-02-04:
I updated to the version committed on February 4, 2024, but it still doesn't work.

ImDroided, 2024-02-07:
Where did you see a updated version?