cariboulabs/cariboulite #144: Error _instance not declared in scope (DragonOS install)

alphafox02, 2023-10-15:
A couple things I'd like to share.

I took the latest DragonOS Pi64 image and ran the installer. During the apt upgrade a new kernel was pulled in. Users should consider if this happens pay attention to the cariboulite build process. You'll notice everything is built against the currently installed/running kernel. Will likely be an issue when you reboot.

Additionally, this occurs during the build process.

`In file included from /home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:2:
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLite.hpp:255:17: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
  255 |     static std::shared_ptr<CaribouLite> _instance;
      |                 ^~~~~~~~~~
In file included from /home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:2:
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLite.hpp:25:1: note: ‘std::shared_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
   24 | #include <functional>
  +++ |+#include <memory>
   25 | 
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:4:6: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
    4 | std::shared_ptr<CaribouLite> CaribouLite::_instance = nullptr;
      |      ^~~~~~~~~~
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:3:1: note: ‘std::shared_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
    2 | #include "CaribouLite.hpp"
  +++ |+#include <memory>
    3 | 
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp: In static member function ‘static CaribouLite& CaribouLite::GetInstance(bool, CaribouLite::LogLevel)’:
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:42:9: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   42 |     if (_instance == nullptr)
      |         ^~~~~~~~~
      |         GetInstance
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:46:30: error: ‘shared_ptr’ is not a member of ‘std’
   46 |             _instance = std::shared_ptr<CaribouLite>(new CaribouLite(forceFpgaProg, logLvl));
      |                              ^~~~~~~~~~
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:46:30: note: ‘std::shared_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:46:52: error: expected primary-expression before ‘>’ token
   46 |             _instance = std::shared_ptr<CaribouLite>(new CaribouLite(forceFpgaProg, logLvl));
      |                                                    ^
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:53:13: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   53 |     return *_instance;
      |             ^~~~~~~~~
      |             GetInstance
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp: In member function ‘void CaribouLite::ReleaseResources()’:
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:83:10: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   83 |     if (!_instance) return;
      |          ^~~~~~~~~
      |          GetInstance
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:84:28: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   84 |     for (size_t i = 0; i < _instance->_channels.size(); i++)
      |                            ^~~~~~~~~
      |                            GetInstance
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp: In destructor ‘CaribouLite::~CaribouLite()’:
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:95:9: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   95 |     if (_instance != nullptr)
      |         ^~~~~~~~~
      |         GetInstance
make[2]: *** [CMakeFiles/cariboulite.dir/build.make:132: CMakeFiles/cariboulite.dir/src/CaribouLiteCpp.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:337: CMakeFiles/cariboulite.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
Consolidate compiler generated dependencies of target zf_log
[  2%] Built target zf_log
Consolidate compiler generated dependencies of target caribou_prog
[  5%] Built target caribou_prog
Consolidate compiler generated dependencies of target caribou_fpga
[  8%] Built target caribou_fpga
Consolidate compiler generated dependencies of target datatypes
[ 14%] Built target datatypes
Consolidate compiler generated dependencies of target at86rf215
[ 20%] Built target at86rf215
Consolidate compiler generated dependencies of target caribou_smi
[ 26%] Built target caribou_smi
Consolidate compiler generated dependencies of target io_utils
[ 36%] Built target io_utils
Consolidate compiler generated dependencies of target rffc507x
[... 33 lines cut ...]
   46 |             _instance = std::shared_ptr<CaribouLite>(new CaribouLite(forceFpgaProg, logLvl));
      |                                                    ^
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:53:13: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   53 |     return *_instance;
      |             ^~~~~~~~~
      |             GetInstance
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp: In member function ‘void CaribouLite::ReleaseResources()’:
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:83:10: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   83 |     if (!_instance) return;
      |          ^~~~~~~~~
      |          GetInstance
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:84:28: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   84 |     for (size_t i = 0; i < _instance->_channels.size(); i++)
      |                            ^~~~~~~~~
      |                            GetInstance
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp: In destructor ‘CaribouLite::~CaribouLite()’:
/home/ubuntu/Downloads/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:95:9: error: ‘_instance’ was not declared in this scope; did you mean ‘GetInstance’?
   95 |     if (_instance != nullptr)
      |         ^~~~~~~~~
      |         GetInstance
make[2]: *** [CMakeFiles/cariboulite.dir/build.make:132: CMakeFiles/cariboulite.dir/src/CaribouLiteCpp.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:337: CMakeFiles/cariboulite.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`
Leading to a failure in the build process.

alphafox02, 2023-11-11:
I've tried the same on Raspberry Pi OS. 
include <memory> 
is missing and causing the issue. 

There's still this happening too
https://github.com/cariboulabs/cariboulite/issues/126

There's also a problem where the kernel headers cannot be determined in the driver CMakelists file, so I had to alter that.

bferrell, 2023-11-23:
cat /etc/*rele*
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

`Building CXX object CMakeFiles/cariboulite.dir/src/CaribouLiteCpp.cpp.o
In file included from /usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:2:
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLite.hpp:255:17: error: 'shared_ptr' in namespace 'std' does not name a template type
  255 |     static std::shared_ptr<CaribouLite> _instance;
      |                 ^~~~~~~~~~
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLite.hpp:25:1: note: 'std::shared_ptr' is defined in header '<memory>'; did you forget to '#include <memory>'?
   24 | #include <functional>
  +++ |+#include <memory>
   25 | 
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:5:30: error: 'std::shared_ptr<CaribouLite> CaribouLite::_instance' is not a static data member of 'class CaribouLite'
    5 | std::shared_ptr<CaribouLite> CaribouLite::_instance = nullptr;
      |                              ^~~~~~~~~~~
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp: In static member function 'static CaribouLite& CaribouLite::GetInstance(bool, LogLevel)':
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:43:9: error: '_instance' was not declared in this scope; did you mean 'GetInstance'?
   43 |     if (_instance == nullptr)
      |         ^~~~~~~~~
      |         GetInstance
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:54:13: error: '_instance' was not declared in this scope; did you mean 'GetInstance'?
   54 |     return *_instance;
      |             ^~~~~~~~~
      |             GetInstance
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp: In member function 'void CaribouLite::ReleaseResources()':
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:84:10: error: '_instance' was not declared in this scope; did you mean 'GetInstance'?
   84 |     if (!_instance) return;
      |          ^~~~~~~~~
      |          GetInstance
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:85:28: error: '_instance' was not declared in this scope; did you mean 'GetInstance'?
   85 |     for (size_t i = 0; i < _instance->_channels.size(); i++)
      |                            ^~~~~~~~~
      |                            GetInstance
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp: In destructor 'CaribouLite::~CaribouLite()':
/usr/local/src/CaribouLite/cariboulite/software/libcariboulite/src/CaribouLiteCpp.cpp:96:9: error: '_instance' was not declared in this scope; did you mean 'GetInstance'?
   96 |     if (_instance != nullptr)
      |         ^~~~~~~~~
      |         GetInstance
make[2]: *** [CMakeFiles/cariboulite.dir/build.make:132: CMakeFiles/cariboulite.dir/src/CaribouLiteCpp.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:337: CMakeFiles/cariboulite.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

`