cariboulabs/cariboulite #153: PCB Project?

vgegok, 2023-11-08:
Great project! 

Due to the use of blind holes in the original PCB and the small minimum through-hole diameter, the cost of PCB production is high. I would like to modify the original PCB project and create a board for testing, learning, and improvement.

Can you provide the engineering files for PCB?

great9, 2023-11-10:
All hardware specs, datasheets and PCB file are available in the [cariboulite_hw ](https://github.com/cariboulabs/cariboulite_hw)repository.
I believe [this is the PCB file](https://github.com/cariboulabs/cariboulite_hw/blob/main/hardware/rev2_r2.5/schematics/cariboulite_r2.5_sch.PDF) you're looking for.

vgegok, 2023-11-10:
> All hardware specs, datasheets and PCB file are available in the [cariboulite_hw ](https://github.com/cariboulabs/cariboulite_hw)repository. I believe [this is the PCB [...]

No, I didn't see PCB project files like AltiumDesign, Eagle, and KiCAD. I only saw gerber and drilling files, and I couldn't make any modifications to these files. I need the original project files, such as:
AD: *. PcbDoc
Eagle: *. brd
KiCad: *. Kicad_ PCB

great9, 2023-11-10:
@meexmachina

meexmachina, 2024-01-10:
see Hardware repo, circuit studio project
https://github.com/cariboulabs/cariboulite_hw/tree/main/hardware_src

closed 2024-01-10 by meexmachina