cariboulabs/cariboulite #160: I/Q swapped?

K7MDL2, 2023-12-07:
I do not know if there is a standard for SoapySDR specs regarding I/Q assignment, multiple SDR apps need to invert I/Q to tune properly.

K7MDL2, 2024-01-25:
Some of my SDR apps do not offer I/Q swap.  I have not looked at the code, busy on other projects and getting antennas up in the trees.

ImDroided, 2024-01-31:
Yea this has to be done in the firmware and I have no idea where to even begin with this thing. Ill have to take a look and see if its simple or above my head.

K7MDL2, 2024-02-01:
I/Q order seems correct now.

I just updated the Pi4B OS and this repository which has many changes recently checked in.  I am able to run some SDR apps again though most have chopped audio, SDR++ in server mode works well.  More below

I also updated to tonight's build of SDR++ compiled on the Pi (the server) and the x64 exe running on the WIn11 (the client side).

Listening to a weak signal source at 1296.1MHz in USB mode:

CubicSDR has high speed broken audio, like a motorboating sound, both at 4Msps and 1.33Msps sample rates.  
SDRGlut same as CubicSDR
GQRX has bursts of spectrum and audio with long silence between.
SparkSDR mostly nothing.
Quisk.  I do not have it set up right for SoapySDR yet.

For SDR++ via SoapySDR audio was broken similar to the CubicSDR case but I/Q order is correct now.
Running sdrpp in server mode it works pretty good with a small hint of audio hiccup in place of the audio bursts (maybe 1-2 per second) in USB with a stable carrier.  I/Q order is correct.  Audio and spectrum sounds clean in Broadcast FM mode.  AGC mode and AGC slider working well.

[image]

Here is 1296.1 carrier on in USB mode.  Not sure what that smaller carrier is on 1296.098, that just showed up.
[image]

- Mike K7MDL

ImDroided, 2024-02-07:
Awesome glad to see this might fix stuff. I will try it out later

K7MDL2, 2024-02-07:
Complied the latest changes today and Cubis SDR on Windows is a bit weird with echoes on FM, but the Linux version is working well.  SDR++ via SoapySDR choppy but using sdrpp server it works great.  I find I need to manually run AGC gain down quite a bit to prevent overload.  Listening to AM airport tower traffic I reduce it until l see the spectrum line go near flat and spurs mostly go away and stray signals just start to decrease relative amplitude.  Choosing automatic AGC is too much gain.  The AGC gain control on these apps are working smoothly for me.