cariboulabs/cariboulite #165: Straight forward instructions to getting started

ignat-mihai, 2023-12-11:
Hello.
I bought the Cariboulite almost a year ago and since then, I had it in a drawer. I wanted this weekend to play with it and I gave it a go with a RPI4. Fresh image, installation worked fine, without any errors and I managed to run SoapySDR server on the PI.
But being a noob on the SDR side, I didn't manage to get it running on the client side..
I tried SDR++ client on Windows, but it's not connecting... When selecting SDR server and inputting the IP and port, on the PI side I got "SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word"...
I tried switching to Ubuntu and installed sdrpp there, which manages to see the Caribou device in the drop down, but when trying to start, it crashes the app, don't remember the error... something about device 1...
I then moved to GQRX which also doesn't manage to connect to the device...
Last I tried CubicSDR, which does manage to connect, or so it seems, but it's not doing anything... I have been trying to move the frequency around but with no luck...
Can somebody help with some clear instructions from start to end?
I will also reply with the errors I have been seeing on sdrpp and gqrx.
Thanks.

DeonMarais64, 2023-12-13:
Hi,
Like you I have had two Cariboulite's laying around and finally got round to getting them working in the last few days. One on a Raspberry Pi Zero 2 W and another on a Pi 400 following [this installation guide](https://github.com/cariboulabs/cariboulite/blob/main/installation.md#installation-details) by creating a new image as suggested.

To your question,
I have had mixed success with GQRX, but I have got it to work quite well with SDR++ V1.1 which I compiled on the Pi 400.
On the server side (the pi with the CaibouLite) from a terminal (can be via SSH, you do not need a desktop environment).
```sdrpp --server```
On the client side, you can run sdrpp either from the menu or a terminal which I like to do because then I can see the trace output from sdrpp, if you are running it on Windows type ```sdrpp -c ```.
Then select ```SDR++ Server``` as the Source and you should be good to go.

I hope this helps.

ignat-mihai, 2023-12-27:
@DeonMarais64 Thank you very much. I managed to get things connected together through SDR++, server side on RPI4 with Caribou and a Windows client. I am still struggling to get basic FM radio listen, but I am on a path! Thank you again and Happy Holidays!

ignat-mihai, 2024-01-24:
Hello.
I recently tried again to work with this device... which at this point it just lets me down too much...
I am trying with a RPI Zero 2W because I have a nifty Spotpear display for it and I would really like it to be in this setup.
I managed to install sdrpp on the Pi and I can boot up the SDR and SDR++ sees it, but when trying to run on an actual frequency, I get:
`01-24 17:05:36.251  2224  2399 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1240 SMI reading operation failed
reader thread failed to read SMI!`

Attaching the full log also.
I can see a weird thing when trying the self test app:
`=====================================
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Digital I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [10]  Synthesizer 85-4200 MHz
 [99]  Quit
    Choice:   4

=====================================
01-24 16:59:07.487  1806  1806 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:480 Testing modem communication and versions
01-24 16:59:07.488  1806  1806 W AT86RF215_Main at86rf215_print_version@at86rf215.c:294 MODEM Version: not AT86RF215 IQ capable modem (product number: 0xff, versions ff)
01-24 16:59:07.488  1806  1806 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:486 The assembled modem is not AT86RF215 / IQ variant (product number: 0xff)
01-24 16:59:07.488  1806  1806 D CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:495 Testing mixer communication and versions
01-24 16:59:07.490  1806  1806 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:500 The assembled mixer is not RFFC5071/2[A]
01-24 16:59:07.490  1806  1806 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:513 Self-test process finished with errors`

This is with the latest codebase pulled and installed... Is the board sold in 2022 not good anymore?.... I placed the order on the 8th of January 2022 and received it in February 2023...
@meexmachina 

[sdrp_caribou_fail.txt](https://github.com/cariboulabs/cariboulite/files/14040299/sdrp_caribou_fail.txt)

ignat-mihai, 2024-01-24:
Attaching a picture of the setup
[image]

unixpunk, 2024-01-24:
@ignat-mihai make sure your user is in the dialout and root groups.  are you using sudo for sdr++?  iirc, it isn't needed for sdr++ but is for other software like CubicSDR.  (sudo creates a conflict with pulseaudio, which only runs as the user and so you need to chmod g+rw /dev/gpiomem for apps to be able to use the device without root, and add that command to /etc/rc.local)

banshee1971, 2024-04-10:
> Hi, Like you I have had two Cariboulite's laying around and finally got round to getting them working in the last few days. One on a Raspberry Pi Zero 2 W and another on a Pi 400 following [this installation guide](https://github.com/cariboulabs/cariboulite/blob/main/installation.md#installation-details) by creating a new image as suggested.
> 
> To your question, I have had mixed success with GQRX, but I have got it to work quite well with SDR++ V1.1 which I compiled using [these instructions](https://github.com/AlexandreRouma/SDRPlusPlus#building-on-linux--bsd) on the Pi 400. On the server side (the pi with the CaibouLite) from a terminal (can be via SSH, you do not need a desktop environment). `sdrpp --server` On the client side, you can run sdrpp either from the menu or a terminal which I like to do because then I can see the trace output from sdrpp. If you are running it on Windows type `sdrpp -c` if you want the trace output. Then select `SDR++ Server` as the Source and you should be good to go.
> 
> I hope this helps.

I followed these instructions, and soapy can see the device now (on a zero 2w pi)
But soapyserver always want to start listening on IPv6.. Can't remove it from here. But i can have it listen manually on ipv4 using a different port.
So I'm now one step in the good direction.

Now... Finding a SDR software i can run from my PC or another PI that will use Soapy to connect to the Pi zero 2w... Still searching.
And... Found one for TX... I think only windows version available for TX? (an i notice a post saying it's not working TX with Soapy?)

Any good online reference (like 2 good one) to follow?
Or... If I run Dragon Os on another PI, soapy connect will already been configured? (i will test that)

DeonMarais64, 2024-04-11:
> Now... Finding a SDR software i can run from my PC or another PI that will use Soapy to connect to the Pi zero 2w... Still searching.

After reinstalling from the latest cariboulite repo to my Pi Zero 2 W and Pi 400 it seems that Soapy Remote is eventually working with the cariboulite.

On a recent DragonOS VM I am able to stream from either pi. It works with SDR++ and CubicSDR but for some reason it does not work properly with GQRX. I got identical results on a Raspberry Pi 5 running Raspberry Pi OS Bookworm 64.

On Windows, I have [Radio Conda](https://github.com/ryanvolz/radioconda) installed and from the conda prompt (eg, running `sdrpp -c` from the SDR++ installed directory under the conda prompt) I am able stream from either pi using Soapy. it works properly with SDR++ but not CubicSDR or GQRX.

banshee1971, 2024-04-11:
> After reinstalling from the latest cariboulite repo to my Pi Zero 2 W and Pi 400 it seems that Soapy Remote is eventually working with the cariboulite.
>
> On a recent DragonOS VM I am able to stream [...]

Did you install the OS version describe on Cariboulite instruction (i did my last installation using the 32 bit lite version available in RPI imager... and got less error now)... I can now see interaction of software between my VM, and the PI Zero .. but not yet working.. i will try the "radio Conda trick"

meanwhile.. did you got some issue when start Soapysdr server on the PI Zero ? mine were only start on IPv6.. even if i disable IPv6 following all guide i found... but if i manually start it and specified the IP and another port (like 1234).. it work.. I try to find how to disable the IPv6 "listen" mode

DeonMarais64, 2024-04-11:
[image]

piz2 is running Raspberry Pi OS Bullseye 32 bit, pi400 is running Raspberry Pi OS Bullseye 64 bit

banshee1971, 2024-04-13:
[image]

DeonMarais64, 2024-04-14:
Your version of SoapySDRServer different to mine, and it looks like it is already running as a service (possibly installed that way ??). 
You should be able to check it, `systemctl status SoapySDRServer.service`

In any case I would build and install SoapyRemote from [source](https://github.com/pothosware/SoapyRemote/wiki#building-soapy-remote).

banshee1971, 2024-04-14:
I install it this way :  #sudo apt install soapyremote-server
[image]