cariboulabs/cariboulite #166: Cannot execute the flow graph in gnuradio-companion

gerd0355, 2023-12-17:
I have tried to install the software for cariboulite as described in "installation.md". It works very well. The tests like cariboulite_test_app and SoapySDRUtils were also successful.

Next I have modified the the file 'fm_radio_receiver_soapy.grc' as shown in 'foto1.png'. After generating the flow graph and executing it, I got the following error message.

Generating: '/home/pi/cariboulite/software/gr-cariboulite/examples/grc/fm_radio_receiver_soapy.py'

Executing: /usr/bin/python3 -u /home/pi/cariboulite/software/gr-cariboulite/examples/grc/fm_radio_receiver_soapy.py

Warning: failed to XInitThreads()
Traceback (most recent call last):
  File "/home/pi/cariboulite/software/gr-cariboulite/examples/grc/fm_radio_receiver_soapy.py", line 37, in <module>
    from gnuradio import soapy
ImportError: cannot import name 'soapy' from 'gnuradio' (/usr/lib/python3/dist-packages/gnuradio/__init__.py)

>>> Done (return code 1) 

What can I do?
[image]

righthalfplane, 2024-01-07:
That error means that the python soapy routines have not been installed correctly.  Make sure gnuradio and the soapy routines are using the same version of python. You can try removing and reinstalling the SoapSDR routines or perhaps installing gr-soapy will work. Where is the "installation.md" file ?

righthalfplane, 2024-01-07:
installing gr-soapy got rid of that problem for me.

gerd0355, 2024-01-07:
Thank you for this info. I will tried it.
You will find the "installations.md" at the menupoint "Code" in the second to last line.