cariboulabs/cariboulite #169: PI 4plus SMI conflict

bferrell, 2023-12-24:
This is likely NOT a caribou issue, but I'm stumped

debian bookworm

sudo modprobe bcm2835_smi  
sudo modprobe  bcm2835_smi_dev
sudo modprobe  smi_stream_dev
dmesg
[ 1413.067473] pinctrl-bcm2835 fe200000.gpio: pin gpio2 already requested by fe804000.i2c; cannot claim for fe600000.smi
[ 1413.067490] pinctrl-bcm2835 fe200000.gpio: pin-2 (fe600000.smi) status -22
[ 1413.067498] pinctrl-bcm2835 fe200000.gpio: could not request pin 2 (gpio2) from group gpio2  on device pinctrl-bcm2711
[ 1413.067504] smi-bcm2835 fe600000.smi: Error applying setting, reverse things back
[ 1460.269557] w1_master_driver w1_bus_master1: Attaching one wire slave 00.780000000000 crc 3a
[ 1460.275390] w1_master_driver w1_bus_master1: Family 0 for 00.780000000000.3a is not registered.
[ 1507.882060] w1_master_driver w1_bus_master1: Attaching one wire slave 00.f80000000000 crc b6
[ 1507.887805] w1_master_driver w1_bus_master1: Family 0 for 00.f80000000000.b6 is not registered.
[ 1565.942799] w1_master_driver w1_bus_master1: Attaching one wire slave 00.040000000000 crc 61
[ 1565.948672] w1_master_driver w1_bus_master1: Family 0 for 00.040000000000.61 is not registered.
[ 1569.607506] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)

Any thought on the fix for this?

bferrell, 2023-12-24:
I'm also using an Argon One M.2 case and see this:

using raspi-gpio get
BANK0 (GPIO 0 to 27):
GPIO 2: level=1 alt=0 func=SDA1 pull=UP

K7MDL2, 2023-12-26:
I ran into that GPIO 2 conflict also once.  Something was grabbing it during bootup or perhaps an app you have run. 

Try running:
sudo raspi-gpio set 2 op pu

If that works, then add it to one of the startup scripts.  I find putting it in /boot/config.txt usually works though once I had to find a script that loaded much later.

This is what my working system reports for GPIO 2 now with raspi-gpio get:
GPIO 2: level=1 fsel=5 alt=1 func=SA3 pull=UP

bferrell, 2023-12-26:
after using sudo raspi-gpio set 2 op pu in /etc/rc.local:

usr/bin/raspi-gpio get

GPIO 2: level=0 func=OUTPUT pull=UP

Then unloading
sudo modprope -r smi_stream_dev
sudo modprobe -r bcm2835_smi

sudo modprobe bcm2835_smi
 
[  392.731933] pinctrl-bcm2835 fe200000.gpio: pin gpio2 already requested by fe804000.i2c; cannot claim for fe600000.smi
[  392.731966] pinctrl-bcm2835 fe200000.gpio: pin-2 (fe600000.smi) status -22
[  392.731987] pinctrl-bcm2835 fe200000.gpio: could not request pin 2 (gpio2) from group gpio2  on device pinctrl-bcm2711
[  392.732005] smi-bcm2835 fe600000.smi: Error applying setting, reverse things back

Suggestions?  Any thought on Identifing fe804000.i2c; cannot claim for fe600000.smi?

K7MDL2, 2023-12-26:
I would take a look at your config/boot.txt settings.  Just a guess, seems like some unliked mode of i2c is being enabled (indicated by the fe804000.i2c).  

This is what I have:

In the middle of the file are some default entries, all are commented out except one:

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=i2c_vc=on
#dtparam=spi=on

At the end of my file in section [all]:

#These 2 are disabled for CaribouLite
#dtoverlay=smi
#dtoverlay=smi-dev
#These 5 are enabled for CaribouLite
dtparam=i2c_vc=on
enable_uart=0
dtparam=i2c_arm=off
param=spi=off
dtoverlay=spi1-3cs

bferrell, 2024-01-02:
ns and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
dtparam=spi=off

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1
#
# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

enable_uart=1
#dtoverlay=w1-gpio
#dtparam=i2c_vc=on
#dtoverlay=spi1-3cs
raspi-gpio set 2 op pu

bferrell, 2024-01-02:
This may also help:
https://paste.debian.net/hidden/d1d45f95/

bferrell, 2025-03-11:
after I got the eeprom to flash, /dev/smi did not appear

see:  https://github.com/cariboulabs/cariboulite/issues/208 for my fix

Found this was due to the Argon One v2 case I was using.
The powerboard uses GPIO pin2 blocking the hat

closed 2025-03-11 by bferrell