cariboulabs/cariboulite #170: Raspberry Pi models and DragonOS

ifuchs, 2024-01-02:
I know that the CaribouLite can be used with any 40 pin RPi but which models are recommended? More specifically, I have a Raspberry Pi 0 W, Raspberry Pi 0W2, and a Raspberry Pi 3B. Itwould be nice to be able to run DragonOS so that all of the SDR software is pre-loaded. Which RPis offer acceptable performance and which could run DragonOS?

alphafox02, 2024-01-18:
DragonOS Pi64 is 64bit and I’ve only tested it on a pi3b+ (not a lot of ram), Pi4 and Pi5. However, on Pi5 it requires a 6.+ kernel and I’ve not recently tried to compile CaribouLite on a 6.+ kernel. I don’t think that’s been tested.

ifuchs, 2024-01-18:
So it should work with DragonOS on my Pi3B+? If so, I am doing something wrong cause it does not see the Cariboulite.

alphafox02, 2024-01-18:
Well the catch is you would need to use the r32 release until it’s confirmed a 6.+ kernel would work. 

This detailed the process at the time of recording it

DragonOS Pi64 CaribouLite Setup Update (Pi4, R32)
https://youtu.be/LkVPyfMotgc

unixpunk, 2024-01-21:
fyi, fix for 6+ kernel: https://github.com/cariboulabs/cariboulite/pull/176/commits/73beb2009a26d8dcb13cbd2dfa987e0d4ac7ebf8
see also: https://github.com/cariboulabs/cariboulite/issues/175