cariboulabs/cariboulite #171: Can't get Cariboulite working on DragonOS

ifuchs, 2024-01-03:
I've watched David Michaelis youtube on using cariboulite with DragonOS but I've not gotten very far with it. I am trying to get this working on a Raspberry Pi 3B while I wait for a Raspberry Pi 5. DragonOS is running fine and I cloned the Cariboulite  github and ran install.sh. I have attached the console file. I made the changes to the config.txt file as shown in the video. When I reboot and do an ll/dev/s*, there is no /dev/smi. I would appreciate very much if someone could have a look at the console file and suggest how I can get this working.
[console.txt](https://github.com/cariboulabs/cariboulite/files/13815133/console.txt)

righthalfplane, 2024-01-05:
Your comment inspired me to take another pass at getting the cariboulite stuff working with DragonOS. I rebuild the cariboulite stuff and it still fail. 

I updated and upgraded the system - You must do this or it likely will not work - This took several hours for my system.

sudo apt-get update
sudo apt-get upgrade

 I destroyed the old version -

rm -rf projects

I down loaded it again.

mkdir ~/projects
cd ~/projects
git clone https://github.com/cariboulabs/cariboulite
cd cariboulite

Then you need to add a line of code to "projects/cariboulite/software/libcariboulite/src/CaribouLite.hpp" after the <functional> line

 #include <functional>
  #include <memory>

You can then do the install -

./install

At the end of the install, I got messages about changes need in "/boot/firmware/config.txt" so I did them.

I then rebooted, the system. After the system came up, running gqrx crashed -

sudo gqrx

Running CubicSDR -

sudo CubicSDR

worked but had a rumble in the sound.

Running SdrGlut worked fine, after right clicking "SDR Dialog" and selecting "I/Q Swap" and turning the Agc on and off and on -

cd ~/Desktop
git clone --depth=1 https://github.com/righthalfplane/SdrGlut.git
cd SdrGlut
make -f makefileUbunta -j 8
sudo ./sdrglut.x

ifuchs, 2024-01-05:
Would any of this work any better if I didn't use dragon OS and just use RaspbianOS? Is there in fact a system where the Cariboulite will work with these various SDR tools flawlessly?

righthalfplane, 2024-01-05:
You still have to make the change to the source and then it works Ok on 64 bit wormwood (this is the recommended  version) and it does not work on Bullseye 64 bit.

ifuchs, 2024-01-25:
Just tried for the umpteenth time and now I get sound from the earbuds but it is essentially just noise (worse than Mac).

OxMarco, 2024-01-25:
Can you add a few more details on how you get the software to work? 
I'd like to see if I can get the board to work at all in the next few days or I'll just return it, software support is amateurish at best...

OxMarco, 2024-01-25:
> I'd like to prove to myself that the Cariboulite can at least receive an FM station. Which app would be easiest to demonstrate this?

SDR++ (via server) or CubicSDR

ifuchs, 2024-01-25:
I had no luck actually hearing the FM station with CubicSDR though the waterfall suggested it was working. I will try SDR++ but I have the feeling that the RPi3B is too slow to be usable for this job.

ifuchs, 2024-01-25:
@OxMarco I basically followed: https://www.youtube.com/watch?v=CO4Sg5iWb3k 
Have you tried that? If so, did the install run without error?

ifuchs, 2024-01-25:
SDR++ gives this error:
sdrpp
[25/01/2024 13:43:39.000] [INFO] SDR++ v1.1.0
[25/01/2024 13:43:39.000] [INFO] Loading config
[25/01/2024 13:43:39.000] [WARN] ConfigManager locked, waiting...
[25/01/2024 13:43:42.000] [ERROR] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[25/01/2024 13:43:42.000] [INFO] OpenGL 3.0 was not supported
[25/01/2024 13:43:42.000] [ERROR] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[25/01/2024 13:43:42.000] [INFO] OpenGL 3.1 ES was not supported
[25/01/2024 13:43:42.000] [INFO] Using OpenGL 2.1
Failed to initialize OpenGL loader!
[25/01/2024 13:43:43.000] [WARN] Could not init using OpenGL with normal GLSL version, falling back to GLSL 1.2
Failed to initialize OpenGL loader!
[25/01/2024 13:43:43.000] [ERROR] Failed to initialize OpenGL with GLSL 1.2

ifuchs, 2024-01-25:
Dos anyone know if the Cariboulite with DragonOS works on the RPi 5 or is it best to use the RPi 4?

righthalfplane, 2024-01-25:
It says you are missing OpenGL, freeglut is built on top of OpenGL so a -

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install freeglut3-dev

should get them both.

If you are still not running, you likely need to destroy the Cariboulite stuff and reinstall it - that rechecks for changes in the configuration file that may have been missed.

A RPI 4 works with Cariboulite on DragonOS.

ifuchs, 2024-01-25:
Thanks. My raspberrypi 3B seems to be too slow to do anything useful
although I don’t know that it explains why the audio from CubicSDR is
unintelligible. Will the RPI 5 work as an alternative to the RPI 4 (it
costs about the same these days) Or does that complicate the installation
further?

righthalfplane, 2024-01-25:
Unintelligible audio is symptom of not being able to decode the audio fast enough.

SdrGlut is some what faster than CubicSDR, may be it will work. You can copy the install instructions from-

https://groups.google.com/g/sdrglut-users/c/ZRednl7cf3c

I do not think SOAPY_SDR_ROOT needs to be set on DargonOS.

You can kick in some additional threads - if you have them ?

ifuchs, 2024-01-26:
tried building sdrglut and this:
make -f makefileUbunta -j 8
gave errors:
In file included from MainSource/QuitDialog.cpp:18:
MainSource/audiolib.h:15:10: fatal error: AL/alut.h: No such file or directory
   15 | #include <AL/alut.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: MainSource/QuitDialog.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from ./Radio/RaceFastsdr.h:13,
                 from ./Radio/Radio.h:11,
                 from MainSource/DialogFileOpen.h:12,
                 from MainSource/main.cpp:15:
./MainSource/audiolib.h:15:10: fatal error: AL/alut.h: No such file or directory
   15 | #include <AL/alut.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: MainSource/main.o] Error 1

unixpunk, 2024-01-26:
@ifuchs you need to swap i/q or invert i/q.  sdr++ and cubicsdr work for me as long as you invert i/q.  never tried glut.  then you will hear normal audio.  use a pi 4.  pi 5 not working as it looks like the gpiomem interface was changed so it will require an updated kernel module to support it, afaik.

righthalfplane, 2024-01-26:
In SdrGlut right click and select "SDR Dialog..."->"I/Q Swap" to turn on the swap

ifuchs, 2024-01-26:
I seem to be regressing. Now I can't start CubicSDR. This ran before some updates that were done for SDRGlut. Perhaps I should reimage the OS and start all over. 

$ CubicSDR 
Loading:: configuration file '/home/ubuntu/.CubicSDR/config.xml'
Loaded 262 rig models via hamlib.

Audio Device #0 Monitor of Built-in Audio Stereo
        Default Output? No
        Default Input? Yes
        Input channels: 2
        Output channels: 0
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                24-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz
                192000hz

Audio Device #1 Built-in Audio Stereo
        Default Output? Yes
        Default Input? No
        Input channels: 0
        Output channels: 2
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                24-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz
                192000hz

Audio Device #2 Built-in Audio Digital Stereo (IEC958)
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 2
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                24-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz
                192000hz

Audio Device #3 Monitor of Built-in Audio Digital Stereo (IEC958)
[... 51 lines cut ...]
Make device 0
Printing 'findCariboulite' Request:
    {channel: S1G}
    {device_id: 0}
    {driver: Cariboulite}
    {label: CaribouLite S1G[b81a5782]}
    {name: CaribouLite RPI Hat}
    {serial: b81a5782}
    {uuid: 721cce59-0b70-4f5c-a561-8388057bafbf}
    {vendor: CaribouLabs LTD}
    {version: 0x0001}
[INFO] Initializing DeviceID: 0, Label: CaribouLite S1G[b81a5782], ChannelType: S1G
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
01-26 07:52:09.735  4222  4276 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized                                                                           
01-26 07:52:09.735  4222  4276 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:465 FPGA communication release failed (-1)                                              
01-26 07:52:09.735  4222  4276 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device                                                                              
CaribouLite: Signal [11] received from pid=[968]                                               
Signal [11] caught, with the following information:                                            
   signal errno = 0                                                                            
   signal process pid = 968                                                                    
   signal process uid = 0                                                                      
   signal status = 0                                                                           
   signal errno / SIGSEGV / the process access invalid region of memory                        
SIGSEGV: memory access violation

unixpunk, 2024-01-26:
dragonos desktop, getting started file, try installing the sdrplay drivers.  not sure not doing this will cause any difference, but i've done it on all mine (because i have one) and not had this issue.  ImDroided has same issue w/gqrx but cubic is working for him now, afaik.  gqrx works for me, but again, maybe it doesn't like that there's a soapy error about sdrplay, just a guess.

ifuchs, 2024-01-26:
I had so many new problems that I decided to re-image DragonOS-Pi64_Beta32 and followed the steps to modify /boot/firmware/config.tx as well as the Cariboulite install.sh files. Now I cannot run gqrx or CubicSDr at all. Unless someone has some idea of how to fix this, I think I shall hang it up until I have a Pi4 and then try again. Here is what I see when trying to run gqrx and CubicSDR.
sudo gqrx
[sudo] password for ubuntu: 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'

sudo CubicSDR
"Pulseaudio error: Connection refused"

./src/common/menucmn.cpp(309): assert "wxIsStockID(GetId())" failed in SetItemLabel(): A non-stock menu item with an empty label?

These errors differ from the ones I was getting before. I'm beginning to wonder whether the Cariboulite was a good idea.

unixpunk, 2024-01-26:
you also need to edit this file according to this pr: https://github.com/cariboulabs/cariboulite/pull/176 otherwise the kernel module won't build.

unixpunk, 2024-01-26:
you also need to do this to fix the requirement to use sudo, which then solves your pulseaudio issue: https://sourceforge.net/p/dragonos-focal/discussion/general/thread/a55c055ef9/#20c1

unixpunk, 2024-01-26:
maybe someone finds the udev rule needed to change the default gpiomem permissions to 660 instead of 600.

ifuchs, 2024-01-27:
@righthalfplane Ran the updates and then the make and got:
In file included from MainSource/QuitDialog.cpp:18:
MainSource/audiolib.h:15:10: fatal error: AL/alut.h: No such file or directory
   15 | #include <AL/alut.h>
      |          ^~~~~~~~~~~
compilation terminated.
In file included from MainSource/audiolib.cpp:1:
MainSource/audiolib.h:15:10: fatal error: AL/alut.h: No such file or directory
   15 | #include <AL/alut.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: MainSource/audiolib.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [<builtin>: MainSource/QuitDialog.o] Error 1
In file included from ./Radio/RaceFastsdr.h:13,
                 from ./Radio/Radio.h:11,
                 from MainSource/DialogFileOpen.h:12,
                 from MainSource/main.cpp:15:
./MainSource/audiolib.h:15:10: fatal error: AL/alut.h: No such file or directory
   15 | #include <AL/alut.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: MainSource/main.o] Error 1
In file included from ./Radio/RaceFastsdr.h:13,
                 from ./Radio/Radio.h:11,
                 from Classes/CLines.cpp:15:
./MainSource/audiolib.h:15:10: fatal error: AL/alut.h: No such file or directory
   15 | #include <AL/alut.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: Classes/CLines.o] Error 1

After trying to get DragonOS to work several with cariboulite several times now and failing each time in different ways (probably something I forgot?) it would be great if someone could wriate out the steps to follow and say what will and won't work. I have watched and tried to floow youtubes and the suggestions in this thread (and others) and I'm not much closer to getting this to work. Maybe I'm the only one having this problem but IMHO the videos provided by Cariboulabs are not very good.

righthalfplane, 2024-01-27:
On dragonOS when I do a find -

ubuntu:~> sudo find / -name "alut.h"
[sudo] password for ubuntu: 
/usr/include/AL/alut.h
find: ‘/run/user/1000/doc’: Permission denied
find: ‘/run/user/1000/gvfs’: Permission denied
ubuntu:~> 

That is where it should show up  for you also.

That is where the command -

sudo apt-get install libalut-dev

should put it.

ifuchs, 2024-01-27:
@righthalfplane Ran these commands:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install build-essential
sudo apt-get install libsoapysdr-dev
sudo apt-get install libopenal-dev
sudo apt-get install libliquid-dev
sudo apt-get install freeglut3-dev
sudo apt-get install libalut-dev
sudo apt-get install libsndfile1-dev
sudo apt-get install librtaudio-dev
sudo apt-get install libglew-dev
sudo apt-get install git
cd ~/Desktop
git clone --depth=1 https://github.com/righthalfplane/SdrGlut.git
cd SdrGlut
make -f makefileUbunta -j 8
setenv SOAPY_SDR_ROOT /usr/
./sdrglut.x

Now I can run sdrglut but when tuned to a local FM broadcast station it is receiving but the audio custs in/out ever 1/2 sec or so. I'm guessing that this RPI3 is too slow and I need an RPi4. The problems with CubicSDR and gqrx are different as they won't run at all.(as reported above). gqrx has the pulseaudio error which @unixpunk shared a link to fix but I haven't determined what needs to change and CubicSDR give: ../src/common/menucmn.cpp(309): assert "wxIsStockID(GetId())" failed in SetItemLabel(): A non-stock menu item with an empty label?

unixpunk, 2024-01-27:
pi 4, or you can try reducing the sample rate below 4mhz.  i have a single-core 1ghz, it can send 2msps of CS16 IQ data from an rtlsdr no problem via soapyremote but can't send 768ksps at CF32 from an Airspy HF+ - ymmv with each kind of SDR in terms of min CPU req.

The link I sent around the sudo / pulseaudio is to just run that chmod command using sudo, and then also insert it into /etc/rc.local above exit 0.  Once the root group has rw access to /dev/gpiomem, you don't need sudo anymore and no more pulseaudio conflict.

unixpunk, 2024-01-27:
One day I supposed myself or @ImDroided will write up the steps, but I have a feeling @alphafox02 will beat us to it with the next DragonOS update and some additions to the CaribouLite section of the getting started file on the Desktop. :)

righthalfplane, 2024-01-27:
Running CubeSDR, I get some warning messages, but I continue past them and It plays a little garbled.

ImDroided, 2024-01-28:
I just got caught up with this thread. I can run CubicSDR but gqrx still wont run at all with the mem access error. Give me a bit and I will be redoing my setup because I wiped that sd card by mistake so I will keep a tab of my steps for you.

ifuchs, 2024-01-28:
I have decided to put the RPi3 aside and try again when my RPi4 arrives. There appear to be too many constraints when using the RPi3 if I can even get an app to work.

ImDroided, 2024-01-28:
> I have decided to put the RPi3 aside and try again when my RPi4 arrives. There appear to be too many constraints when using the RPi3 if I can even get an app to work.

I agree wait till the 4 arrives and by then I will have a small writeup for you. I was able to tune both receivers in CubisSDR but gqrx was still complaining about the mem error. I have no idea why this is happening but @unixpunk saw I was missing something in my settings.txt and thats when CubicSDR started working fine. I am currently trying to get this working on a pi5 but I am having no luck so its going back on my 4 until I can figure the 5 out.

ifuchs, 2024-01-28:
@ImDroided Thanks! I will look for your notes after I get the Pi4.

ImDroided, 2024-02-08:
it looks like that commit has been accepted so skip that step