cariboulabs/cariboulite #172: Cariboulite does not transmit using the SoapySDR API

righthalfplane, 2024-01-06:
Cariboulite does not transmit using the SoapySDR API - all that it does is generate a writestream error -5. This error occurs on DargonOS and on BookWorm 64 bit. I have a simple Soapy program that works with all of the other SDR transmitters that I have (HackRF One,  BladerRF-xA5, LimeMini 2.0, ANTSDR e200) on many systems MacOS Monterey, Ubuntu 22.04, Bookworm, Ubuntu 23.10, Windows 7, Windows 10. I have attached the test program - 

[sendManyFFT.cpp.txt](https://github.com/cariboulabs/cariboulite/files/13850810/sendManyFFT.cpp.txt)

Paulo-D2000, 2024-01-10:
After the last PR it should work (Transmits CW only, no errors on stream transmit but also no signal output...)

righthalfplane, 2024-01-11:
On Bullseye, Carboulite HiF and Carboulite S1G both transmit CW. However, on a relative scale the S1G puts out 30 db more power than the HiF. I would not expect that huge difference. The last PR has screwed up the reception - readStream, now returns zero every few reads and it losses the data - so the sound is somewhat garbled. Once, I run a transmit - the receive returns a zero on every read and no longer works at all.

righthalfplane, 2024-01-11:
On DragonOS, everything acted the same as Bullseye, except that you had to run as super user.

great9, 2024-01-12:
> The last PR has screwed up the reception - readStream, now returns zero every few reads and it losses the data - so the sound is somewhat garbled. Once, I run a transmit - the receive returns a zero on every read and no longer works at all.

please open a new issue.

great9, 2024-01-12:
> On DragonOS, everything acted the same as Bullseye, except that you had to run as super user.

can you check if udev rules have been added and that the user you're running as has the permissions to a access the hat?

righthalfplane, 2024-01-12:
"can you check if udev rules have been added and that the user you're running as has the permissions to a access the hat?"

Bullseye and DragonOS have the same dev rules -

# SMI Devices
KERNEL=="smi", SUBSYSTEM=="smi-stream-dev", MODE="0666"

# Other RPI Devices
SUBSYSTEM=="i2c-dev", GROUP="i2c", MODE="0666"
SUBSYSTEM=="spidev", GROUP="spi", MODE="0666"
SUBSYSTEM=="bcm2835-gpiomem", GROUP="gpio", MODE="0666"
SUBSYSTEM=="mem", KERNEL=="mem|kmem|port", GROUP="kmem", MODE="0666"

On DragonOS, it did not have the groups i2c,spi, or gpio. So, I added them and I added them the the user, but I did not do any good. It stops with the error -

Opening rpi in /dev/gpiomem: Permission denied
Module found: /usr/lib/aarch64-linux-gnu/SoapySDR/modules0.8/libSoapyCariboulite.sostart_mmap() error: ubuntu:~> 

 It is as if it a needs a rule for "gpiomem" or something like that. I gave the DragonOS user (ubuntu) all the same groups as the Bullseye user and it still gets the same error.

righthalfplane, 2024-01-12:
The DragonOS has some extra stuff -

raspberrypi:~> sudo find / -name "*gpiomem*"
/dev/gpiomem
find: ‘/run/user/1000/gvfs’: Permission denied
/run/udev/data/+platform:fe200000.gpiomem
/sys/class/bcm2835-gpiomem
/sys/class/bcm2835-gpiomem/gpiomem
/sys/devices/platform/soc/fe200000.gpiomem
/sys/devices/virtual/bcm2835-gpiomem
/sys/devices/virtual/bcm2835-gpiomem/gpiomem
/sys/bus/platform/devices/fe200000.gpiomem
/sys/bus/platform/drivers/gpiomem-bcm2835
/sys/bus/platform/drivers/gpiomem-bcm2835/fe200000.gpiomem
/sys/firmware/devicetree/base/soc/gpiomem

ubuntu:~> sudo find / -name "*gpiomem*"
/sys/class/bcm2835-gpiomem
/sys/class/bcm2835-gpiomem/gpiomem
/sys/devices/platform/soc/fe200000.gpiomem
/sys/devices/virtual/bcm2835-gpiomem
/sys/devices/virtual/bcm2835-gpiomem/gpiomem
/sys/bus/platform/devices/fe200000.gpiomem
/sys/bus/platform/drivers/gpiomem-bcm2835
/sys/bus/platform/drivers/gpiomem-bcm2835/fe200000.gpiomem
/sys/firmware/devicetree/base/soc/gpiomem
/sys/module/bcm2835_gpiomem
/sys/module/bcm2835_gpiomem/drivers/platform:gpiomem-bcm2835
/usr/lib/modules/5.15.0-1037-raspi/kernel/drivers/char/broadcom/bcm2835-gpiomem.ko
/usr/lib/modules/5.15.0-1044-raspi/kernel/drivers/char/broadcom/bcm2835-gpiomem.ko
/dev/gpiomem
find: ‘/run/user/1000/doc’: Permission denied
find: ‘/run/user/1000/gvfs’: Permission denied
/run/udev/data/+platform:fe200000.gpiomem

Perhaps some of that is the problem

righthalfplane, 2024-01-13:
I have attached photos of the power spectrum for the S1G and HiF. The S1G has more than a single CW frequency. The HiF has only one very tiny CW signal - It is the small peak just to the right of the green line at the center of the screen. The HiF signal is smaller than usual this time ?

[image]

[image]

righthalfplane, 2024-01-13:
Filling the transmit buffer with zeros or ones did not change the results.

great9, 2024-01-13:
> Filling the transmit buffer with zeros or ones did not change the results.

@meexmachina

User25514, 2024-03-11:
Im also struggling with this problem, all I can do is CW but I want to actually transmit audio data. If anyone has found a fix recently or some ideas, that would be amazing

wvogel00, 2024-09-07:
Hi, can anyone transmit a signal from Cariboulite with GRC?
As the attached image shows, there was no error, so it seems to be running correctly. However, I can see no signal at 52.525MHz on Spectrum Analyzer. (Of course, I test this by connecting the analyzer and Cariboulite with coaxial cable)
On GRC, it succeeded in finding the ID of the HiF channel.

[image]

K7MDL2, 2024-09-12:
Try CW mode

PeterQFR, 2025-03-22:
@K7MDL2, how do you do that in the Soapy Sink or Soapy Custom Sink? Is it some additional param?

K7MDL2, 2025-03-23:
I do not remember much about this anymore.    What I was able to do is use the utilities to send a carrier.  I modified it into a simplistic RF sweep generator.  I figured I could work on that for CW in lieu of other more standard approaches.  I have not returned to this project for a long time now though.

righthalfplane, 2025-03-24:
@PeterQFR , It been awhile, I think that you just add 

"CW=1"

To the device string.

PeterQFR, 2025-03-24:
I just tried CW=1, but the tx light is not illuminating.

[image]

PeterQFR, 2025-03-24:
I couldn't get the Soapy custom sink to handle that argument without crashing, so I hardcoded it to be active when the stream is TX... Just to confirm, that frequency won't respond to gnuradio.. I only get a CW on the frequency of the sink but I am unable to adjust it.

great9, 2025-03-24:
the driver needs to be refactored, fixed, upgraded. it's held together by a string of spaghetti

User25514, 2025-03-24:
Wow it's been a while, transmit mode is quite destroyed due to the owners basically forgetting about everyone but still taking money. I was making it work on raspberry pi os on python but with soapy there was no inbuilt way to transmit custom data

great9, 2025-03-24:
> Wow it's been a while, transmit mode is quite destroyed due to the owners basically forgetting about everyone but still taking money. I was making it work on raspberry pi os on python but with soapy there was no inbuilt way to transmit custom data

Yup, agreed.
I think the only way forward is to fork the repo, and have the neglected community support the driver updates. I would be willing to donate.

PeterQFR, 2025-03-25:
They should advise crowd supply to update the spec to RX only. I probably wont contribute too much here, as I don't know enough regarding the interfaces (though I saw there was a branch by @matteoserva https://github.com/cariboulabs/cariboulite/pull/197 dealing with this, but seems abandoned) and move on to another SDR hardware like uSDR that has better support.