cariboulabs/cariboulite #178: no Cariboulite blocks in gnu-radio Companion

mjmckenna, 2024-01-22:
Hi.  I've downloaded and installed the Cariboulite source code on the R32 version of DragonOs and also on the 64-bit Bullseye version of RPiOS.  For the DragonOS, I was hoping that I would see the Cariboulite sink/source blocks as part of gnu-radio companion but I don't;  I didn't see them under Bullseye either.   I followed the modifications to install.sh outlined by [cemaxecuter] in his youtube video. 
 (https://www.youtube.com/watch?v=CO4Sg5iWb3k) to stay with the 5.15 kernel and add a sudo before the make on line 144 of install.sh.  The cariboulite-test-app application runs and I can generate the CW tones with the app.  I see the 2 SMI devices with lsmod | grep smi.  

I have gr-soapy installed as well as gr-caribouLite.  Before looking at Issue #166, I had thought that to access cariboulite is was done through a custom Soapy sink or source with a soapyServer also running in the background.  

The only thing that I can think of is that there is a manual step to move files into a directory where grc can find them.  But I have no idea where that would be and what I have to move to see the blocks in grc.  

Does anyone have any suggestions where I might check?

alphafox02, 2024-01-30:
Tested on latest DragonOS Pi64 after installation completed and up and running. 

```
cd cariboulite/software/gr-caribouLite
mkdir build && cd build
cmake ..
make -j4
sudo make install
sudo ldconfig
```

Reopen GNURadio and search for source block. I just tried it, seems to work, but does show some concerning output bottom left area of GNURadio.

mjmckenna, 2024-03-01:
Hi alphafox02. thanks for your comment.   I was searching for a cariboulite sink block.  I can receive a local FM station with SDR++ with the cariboulite lite using a soapy interface.  I get a fatal gilbc error 11 with sysmalloc if I try to use the cariboulite source block in my grc program that works with a plutoSDR source block and a RTLSDR source block (just disabling each source block to test with each.) 

Has anyone gotten the cariboulite source block to work?