cariboulabs/cariboulite #180: Anyone have this working on Pi 5?

ImDroided, 2024-01-27:
Has anyone gotten the caribou to work on a pi5? if so what steps did you take? I am getting /dev/gpiomem not found as it seems they changed the GPIO access in the 5

alphafox02, 2024-01-29:
I think hardware has changed on Pi5. I’m sitting here with one but had to drop back to the Pi4.

ImDroided, 2024-01-29:
> I think hardware has changed on Pi5. I’m sitting here with one but had to drop back to the Pi4.

I've seen they changed the way the gpio is so hopefully this is just a caribou firmware update we need and not the end of the road. I picked up a pi 5 just for this project in hopes of making a all in one sdr system to carry around. 

By the way thanks for all your work on DragonOS it really is a awesome SDR OS build.

steve-m, 2024-02-03:
As of now, it won't work with the RPi 5. In my opinion it should be mentioned everywhere, because that is quite misleading.

The Raspberry Pi 5 doesn't have the SMI (Secondary Memory Interface) the cariobou relies on, because that was a feature of the Broadcom SoC. Now the 40-pin GPIO header is connected to the RP1, which doesn't implement this interface.

As the RP1 has an ARM core with PIO peripherals however, it might be possible to implement an interface to the FPGA there, but that will require quite intensive software changes, patching of the ARM code running in the RP1, and most likely a modified gateware for the FPGA on the caribou.

Currently, there isn't even an official way to run code on the RP1, let alone documentation for the ARM peripherals (like the PIO) of the RP1. However, some people are already hacking/reverse engineering that:

https://github.com/MichaelBell/rp1-hacking
https://github.com/librerpi/rp1-lk

Yet, I doubt there will be RPi5 support for the caribou anytime soon.

great9, 2024-02-04:
They are trying to make it work...
https://github.com/cariboulabs/cariboulite/commit/61ed2057d56563d20c9d43e5d5e249c2c1164209

ImDroided, 2024-02-07:
Awesome I am happy to see at least some progress. The performance of my pi5 8gb's blows my pi4's out the water. sometimes I forget that I am on a pi when in remote desktop. I hope to have a pi5 with my m.2 nvme and the caribou to use with my satellite tracker but I need to run gpredict and satdump at the same time. the pi5 can do live passes while running a GUI desktop and the 4 can barely hang in console mode. This paired with a 10in touchscreen and battery all stuffed in a nice all in one case will be the ultimate mobile sat rig. Sure I could use a laptop but where is the fun in that.

cleverca22, 2024-04-20:
i have control of the PIO block on the pi5, and it can sustain about 500mbit of transfer from gpio to dram
but it drops some samples, so you need some hw flow control to throttle it

the repos:
https://github.com/MichaelBell/rp1-hacking/tree/main/launch_core1
https://github.com/librerpi/rp1-lk
https://github.com/cleverca22/rp1-kernel-example

an example of using it:
```
[clever@amd-nixos:~/apps/rpi/rp1-lk]$ time make PROJECT=rp1-test -j && scp build-rp1-test/lk.bin pi5w:.
root@raspberrypi:~# cd ~/rp1-hacking/launch_core1/ && ./core1_test -f /home/clever/lk.bin
```
`core1_test` will reset the cortex-m3 core1 in the RP1, and then force it to jump into a given binary
rp1-lk is a small kernel, that gives you a repl on uart1 (gpio 0/1), and configures the PIO
rp1-kernel-example sets up dma between /dev/example and the PIO FIFO

after that, its just a matter of getting some PIO code to handshake with the fpga and setup bi-directional comms

great9, 2024-04-21:
> i have control of the PIO block on the pi5

Just tagging these devs in case they didn't see this @AlonSpinner @matteoserva @meexmachina

geerlingguy, 2024-05-10:
Would love to see it :)

Getting into some SDR stuff this year (finally), and I was about to pop it on a Pi 5 but will hold off and stick with a Zero 2W for now. I would be happy to help in any way (even if that's just trying to run a few things up to whoever will listen, for better RP1 support/documentation).

great9, 2024-05-10:
> Would love to see it :)
>
> Getting into some SDR stuff this year (finally), and I was about to pop it on a Pi 5 but will hold off and stick with a Zero 2W for now. I would be happy to help in any way [...]

hi Jeff.
Drivers were really bad before - I couldn't get a stream running without completely blocking my rpi3b+. Even though it was an older SBC it should have worked. Now, the drivers are much better with the last month's overhaul. They're now stable, and transmission is possible with @matteoserva 's [branch](https://github.com/matteoserva/cariboulite/blob/transmission_fix/docs/notes/tutorial.md). There's lots to fix in the drivers and the entire software package but at least it works now.

korneliuszo, 2025-02-10:
Is there any progress on making it work on RPi5?

korneliuszo, 2025-02-11:
There is PIO support now on Rpi5
https://www.raspberrypi.com/news/piolib-a-userspace-library-for-pio-control/