cariboulabs/cariboulite #182: RPi 4B 4GB first time

ajkoren1959, 2024-02-03:
My setup:
- Raspberry Pi 4B, 4GB with a fresh Dragon OS
- Downloaded cariboulite code: 
> git clone...
- Following issue #137, I changed the following: 
I replaced line 1319 in cariboulite/driver/build/smi_stream_dev.c as follows:
        //smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
        smi_stream_class = class_create(DEVICE_NAME);
- Following instructions + issue #135, I edited file /boot/firmware/config.txt. Changed [all] to:
      [all]
      # Enable the audio output, I2C and SPI interfaces on the GPIO header. As these
      # parameters related to the base device-tree they must appear *before* any
      # other dtoverlay= specification
      dtparam=audio=on
      #dtparam=i2c_arm=on
      dtparam=i2c_arm=off
      #dtparam=spi=on
      dtparam=spi=off
      
      # CaribouLite
      dtparam=i2c_vc=on
      dtparam=spi_vc=on
      dtoverlay=spi1-3cs
      #enable_uart=0
      #arm_freq=550
      #core_freq=250
      #gpu_mem=4

My issue:
sdrpp doesn't receive signals. A quick search found that SoapySDRUtil also shows errors. Please see "shm_open: No such file or directory" and following errors below:

ubuntu@ubuntu:~/projects/cariboulite$ sudo SoapySDRUtil --probe
[sudo] password for ubuntu: 
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] SoapyCaribouliteSession, sessionCount: 0
02-02 18:48:22.334  4482  4482 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
02-02 18:48:23.893  4482  4482 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!

Printing 'findCariboulite' Request:
shm_open: No such file or directory
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.1.0.5-0-unknown
[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed
[INFO] Initializing DeviceID: 0, Label: CaribouLite S1G[66b1825c], ChannelType: S1G
[INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)

Also, it might be unrelated, but it's suspicious: I seem to be running out of space in the root partition (96% full):
ubuntu@ubuntu:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           380M  3.5M  376M   1% /run
/dev/mmcblk0p2   15G   14G  630M  96% /
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M  193M   60M  77% /boot/firmware
tmpfs           380M  104K  379M   1% /run/user/1000

Any help with either issue will be greatly appreciated.

ajkoren1959, 2024-02-06:
I ran install.sh again (on a fresh DragonOS). This time I noticed earlier errors:
ubuntu@ubuntu:~/projects/cariboulite$ ./install.sh > install.txt
[sudo] password for ubuntu: 
E: Unable to locate package raspberrypi-kernel-headers
E: Unable to locate package raspberrypi-kernel-headers
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gnu-gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.1.0-8ubuntu1~22.04) 13.1.0
[INFO] SoapyCaribouliteSession, sessionCount: 0
Opening rpi in /dev/gpiomem: Permission denied
[INFO] SoapyCaribouliteSession, sessionCount: 0
Opening rpi in /dev/gpiomem: Permission denied

Is this the easiest thing to do to start from Raspberry Pi OS rather than from DragonOS?

ImDroided, 2024-02-07:
Ok first things first get a bigger sd card lol. 16gb is nothing with updates and such. I'd do 64gb min

I have a writeup at https://radiomenace.com/2023/03/20/getting-cariboulite-working-in-dragonos-on-a-raspberry-pi/  but you still need to edit the THIS_MODULE thing. 

This is what your config.txt should look like. 

[all]
# Enable the audio output, I2C and SPI interfaces on the GPIO header. As these
# parameters related to the base device-tree they must appear *before* any
# other dtoverlay= specification
dtparam=audio=on
dtparam=i2c_arm=off
dtparam=spi=off
dtparam=i2c_vc=on
dtoverlay=spi1-3cs

I would get rid of all that entire section and try this. if this works then add back one by one to see what breaks it. 

next do

sudo usermod -G root,dialout ubuntu

Edit /etc/rc.local and directly above the ‘exit 0’ line, insert a new line with this: chmod g+rw /dev/gpiomem

then reboot with

sudo reboot

You can run chmod g+rw /dev/gpiomem in a terminal and rerun your software and it should work as you are getting 

[INFO] SoapyCaribouliteSession, sessionCount: 0
Opening rpi in /dev/gpiomem: Permission denied

You will need to edit rc.local as per above because it resets the permissions on gpiomem on boot for some reason. 

Hope this helps let me know.

ajkoren1959, 2024-02-08:
Thanks for the advice, ImDroided.

Bigger SD Card: Right on! :-)
config.txt file: That's my file too.
I'm back to using DragonOS. Raspberry Pi OS requires installation of SoapySDR which comes with Dragon, but otherwise, since I got errors the move didn't help.

I implemented your other tips: 
1. > sudo usermod -a -G root,dialout ubuntu
2. Edit /etc/rc.local and directly above the ‘exit 0’ line, insert a new line with this: chmod g+rw /dev/gpiomem

After reboot, I got:
`ubuntu@ubuntu:~/projects/cariboulite$ sudo SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
02-07 19:45:35.995 31274 31274 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:209 FPGA already operational - not programming (use 'force_prog=true' to force update)                                                                                          
02-07 19:45:35.995 31274 31274 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:539 couldn't open smi driver file '/dev/smi' (No such file or directory)                                                                                                              
02-07 19:45:35.995 31274 31274 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule                                                                                                                             
02-07 19:45:35.997 31274 31274 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed                   
02-07 19:45:35.997 31274 31274 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized              
[ERROR] cariboulite_init_driver() failed   

and:
ubuntu@ubuntu:~/projects/cariboulite$ sudo SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] SoapyCaribouliteSession, sessionCount: 0
02-07 20:37:36.856 99227 99227 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:209 FPGA already operational - not programming (use 'force_prog=true' to force update)                                                                                          
02-07 20:37:36.857 99227 99227 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:539 couldn't open smi driver file '/dev/smi' (No such file or directory)                                                                                                              
02-07 20:37:36.857 99227 99227 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule

I see in your write-up that you used RPi 3, not 4. Is there a compatibility issue with RPi 4?

ImDroided, 2024-02-08:
looks like your stuck where I am now. I have a 4 I dont know where I put I had a 3. check @unixpunk responce and see if that helps.

ajkoren1959, 2024-02-08:
[@ImDroided](https://github.com/ImDroided), In [your writeup](https://radiomenace.com/2023/03/20/getting-cariboulite-working-in-dragonos-on-a-raspberry-pi/) you say it's a RPi 3. Your pic of the RPi also matches a 3, not a 4, in my opinion.

ImDroided, 2024-02-09:
> [@ImDroided](https://github.com/ImDroided), In [your writeup](https://radiomenace.com/2023/03/20/getting-cariboulite-working-in-dragonos-on-a-raspberry-pi/) you say it's a RPi 3. Your pic of the RPi also matches a 3, not a 4, in my opinion.

Ah yes I remember that now lol the 3 wasn't fast enough and got switched for a 4 almost instantly. I did a fresh install following my writeup and editing the smi stream file the line of code has moved down a few lines as someone added to the file but didnt fix that. I can run CubicSDR but gqrx still has mem access error

ajkoren1959, 2024-02-12:
I have a partial solution. It turns out that the solution to this:
```
shm_open: No such file or directory
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
```
is to run the corresponding service:
`sudo systemctl start sdrplay`

Who new? ;-)
Now, `sudo SoapySDRUtil --find` doesn't return errors,.
Also, `sudo sdrpp` seems to work. However, it doesn't receive any real stations., only static noise at the selected carrier frequency. I run it (and listen) at 98-102 MHz, where I should see multiple FM radio stations...

ajkoren1959, 2024-02-14:
I'll document a bit more, for the benefit of anyone else who might have the same issues:
- I resolved FM reception, even though the confusing `audio write error, underrun.` shows. I think it's mainly an issue of setting the correct parameters, i.e. enough AGC Gain and, in FM, a bandwidth of 200KHz.
- It's still not clear how to switch the CaribouLite into transmit mode and how to set its frequency and modulation. Do I have to understand its API code? (which I assume is open code...) Did anyone write an interface for the API in order to make it easier to operate?

ajkoren1959, 2024-02-14:
I'll answer my own question again:
- The utility examples/python/soapy_synth.py seems to be design for CW signal transmission. 
- I assume that I'll find ways to better control it by using the SoapySDR API directly.
- The following link offers documentation and whole applications which use SoapySDR:
https://github.com/pothosware/SoapySDR/wiki