cariboulabs/cariboulite #185: SIGSEGV: memory access violation

ImDroided, 2024-02-08:
Has anyone figured this out yet? I cant open any apps. I was able to open CubicSDR the other day but not now QGRX crashes instantly

ubuntu@ubuntu:~$ sudo CubicSDR 
Loaded 262 rig models via hamlib.

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

Audio Device #0 default
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 10000
        Duplex channels: 0
        Native formats:
                8-bit signed integer.
                16-bit signed integer.
                24-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
                64-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                4000hz
                5512hz
                8000hz
                9600hz
                11025hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                88200hz
                96000hz
                176400hz
                192000hz

Audio Device #1 hw:bcm2835 HDMI 1,0
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 8
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
        Supported sample rates:
                8000hz
                9600hz
                11025hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                88200hz
                96000hz
                176400hz
                192000hz

Audio Device #2 hw:bcm2835 HDMI 1,1
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 2
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
        Supported sample rates:
                44100hz
[... 93 lines cut ...]
   signal process pid = -1993413121                                                                                              
   signal process uid = 65536                                                                                                    
   signal status = 0                                                                                                             
   signal errno / SIGSEGV / the process access invalid region of memory                                                          
SIGSEGV: memory access violation                                                                                                 
ubuntu@ubuntu:~$ sudo nano /boot/firmware/config.txt                                                                            
ubuntu@ubuntu:~$ sudo nano /etc/rc.local 
ubuntu@ubuntu:~$ gqrx 
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.3.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 10 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /home/ubuntu/.config/gqrx/bandplan.csv
BookmarksFile is /home/ubuntu/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.1.0.5-0-unknown
[INFO] SoapyCaribouliteSession, sessionCount: 0
CaribouLite: Signal [11] received from pid=[16]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = 16
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation
ubuntu@ubuntu:~$

unixpunk, 2024-02-08:
@ImDroided  can't use sudo, not sure it solves it anyway, sudo breaks pulseaudio unless you run it for the whole system, which isn't default.

this issue is: 
02-07 20:15:17.086 8009 8047 E CARIBOU_SMI [caribou_smi_init@caribou_smi.c](mailto:caribou_smi_init@caribou_smi.c):539 **couldn't open smi driver file '/dev/smi' (No such file or directory)**
02-07 20:15:17.086 8009 8047 E CARIBOULITE Setup [cariboulite_init_submodules@cariboulite_setup.c](mailto:cariboulite_init_submodules@cariboulite_setup.c):288 Error setting up smi submodule

try putting in config.txt these lines after the rest of them and reboot:
dtoverlay=smi
dtoverlay=smi-dev

ImDroided, 2024-02-08:
didnt work 

ubuntu@ubuntu:~$ CubicSDR 
Loading:: configuration file '/home/ubuntu/.CubicSDR/config.xml'
Loaded 262 rig models via hamlib.

Audio Device #0 Built-in Audio Stereo
        Default Output? Yes
        Default Input? No
        Input channels: 0
        Output channels: 2
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                24-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz
                192000hz

Audio Device #1 Monitor of Built-in Audio Stereo
        Default Output? No
        Default Input? Yes
        Input channels: 2
        Output channels: 0
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                24-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz
                192000hz

SDR enumerator starting.
SoapySDR init..
        API Version: v0.8.0
        ABI Version: v0.8
        Install root: /usr
        Loading modules... 
[INFO] SoapyCaribouliteSession, sessionCount: 0
02-07 21:04:59.903  3565  3594 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220                                                                                                                               
02-07 21:05:01.539  3565  3594 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!  
                                                                                                                                 
02-07 21:05:01.640  3565  3594 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:539 couldn't open smi driver file '/dev/smi' (No such file or directory)                                                                                                              
02-07 21:05:01.640  3565  3594 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule                                                                                                                             
02-07 21:05:01.642  3565  3594 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed                   
02-07 21:05:01.642  3565  3594 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized              
[ERROR] cariboulite_init_driver() failed                                                                                         
        Available factories...Cariboulite, airspy, airspyhf, bladerf, hackrf, lime, miri, null, osmosdr, plutosdr, redpitaya, remote, rfspace, rtlsdr, sdrplay, uhd
Printing 'findCariboulite' Request:
shm_open: No such file or directory
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.1.0.5-0-unknown
Available vertical sync SwapInterval functions: 
[... 50 lines cut ...]
Signal [6] caught, with the following information:                                                                               
   signal errno = 0                                                                                                              
   signal process pid = 3565                                                                                                     
   signal process uid = 1000                                                                                                     
   signal status = 0                                                                                                             
   signal errno / tkill / tgkill sent the signal                                                                                 
SIGABRT: sent by abort() command                                                                                                 
ubuntu@ubuntu:~$ gqrx                                                                                                            
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.3.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 10 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /home/ubuntu/.config/gqrx/bandplan.csv
BookmarksFile is /home/ubuntu/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.1.0.5-0-unknown
[INFO] SoapyCaribouliteSession, sessionCount: 0
CaribouLite: Signal [11] received from pid=[16]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = 16
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation
ubuntu@ubuntu:~$

ImDroided, 2024-02-08:
config.txt

[all]
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

[pi4]
max_framebuffers=2
arm_boost=1

[all]
# Enable the audio output, I2C and SPI interfaces on the GPIO header. As these
# parameters related to the base device-tree they must appear *before* any
# other dtoverlay= specification
dtparam=audio=on
dtparam=i2c_arm=off
dtparam=spi=off
dtparam=i2c_vc=on
dtoverlay=spi1-3cs
dtoverlay=smi
dtoverlay=smi-dev

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
#hdmi_drive=2
hdmi_force_hotplug=1

#Next is for headless usage only (no HDMI connected) and 1080P resolution
hdmi_force_mode=1
hdmi_group=2
hdmi_mode=82

[cm4]
# Enable the USB2 outputs on the IO board (assuming your CM4 is plugged into
# such a board)
dtoverlay=dwc2,dr_mode=host

[all]

# Enable the KMS ("full" KMS) graphics overlay, leaving GPU memory as the
# default (the kernel is in control of graphics memory with full KMS)
#Comment the below line for headless usage
#dtoverlay=vc4-kms-v3d

# Autoload overlays for any recognized cameras or displays that are attached
# to the CSI/DSI ports. Please note this is for libcamera support, *not* for
# the legacy camera stack
camera_auto_detect=1
display_auto_detect=1
# Config settings specific to arm64
arm_64bit=1
dtoverlay=dwc2

# Enable the PCIe External connector.
dtparam=pciex1
# This line is an alias for above (you can use either/or to enable the port).
dtparam=nvme
#dtparam=pciex1_gen=3

ImDroided, 2024-02-08:
Dont know why but fresh install with normal config.txt edit and THIS_MODULE delete CubicSDR now works but gqrx is still a mem access error.

ImDroided, 2024-02-09:
So after letting CubicSDR run overnight and all day on a FM radio station I closed CubisSDR and could not open it again until after a reboot. Same mem access error.

great9, 2024-02-13:
on my rpi 3b with 512 mb ram it spins up 2 reader threads an 2 writer threads with immediate 400% Load on SDRPlusPlus. This makes all communication choppy, with or without compression enabled. When I kill the smi threads an rmmod the smi_stream_dev , it crashes with mem error and it's not possible to use it again until reboot.

@ImDroided when yours works, it at least works.

unixpunk, 2024-02-13:
@great9 pi 3b is only good for SoapyRemote / SoapySDRServer scenarios, it can't do all that extra work processing the fft and demodulating audio, etc - BUT if you just run sudo SoapySDRServer --bind and then use sdrpp or cubicsdr, etc on a more capable machine, the audio will work fine.  Side note - use TCP, not UDP when selecting the remote Caribou sdr in the chosen app.

If you want to see the fft and hear non-choppy audio, you'll need a pi 4.

ignat-mihai, 2024-02-17:
Hello, all.
I have the same issue with the /dev/smi missing when trying to use the Caribou and I am at the edge of my patience...
I have this config on DragonOS, fresh install, on a RPI4 8GB...
[config.txt](https://github.com/cariboulabs/cariboulite/files/14318857/config.txt)

I am getting the missing /dev/smi message whenever I want to do something with the caribou and I can see the following output for sudo dmesg:
`ubuntu@ubuntu:~$ sudo dmesg | grep smi
[sudo] password for ubuntu: 
[    6.581459] pinctrl-bcm2835 fe200000.gpio: pin gpio16 already requested by fe215080.spi; cannot claim for fe600000.smi
[    6.581528] pinctrl-bcm2835 fe200000.gpio: pin-16 (fe600000.smi) status -22
[    6.581595] smi-bcm2835 fe600000.smi: Error applying setting, reverse things back
[    6.606318] smi_stream_dev: loading out-of-tree module taints kernel.
[    6.606344] smi_stream_dev: module verification failed: signature and/or required key missing - tainting kernel
[    6.607034] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    7.803572] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    7.896288] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    7.953755] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    7.955999] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    7.982013] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.212845] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.274481] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.292913] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.329299] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.494281] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.516432] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.519442] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.520295] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.520735] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.521160] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.522078] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.657397] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.697743] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    8.786369] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[    9.127903] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   19.167625] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[   19.167755] platform soc:smi_dev: deferred probe pending
[  676.293907] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[  676.298168] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[  676.361852] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[ 1064.092049] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[ 1064.098142] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
[ 1064.164067] smi-stream-dev: smi_stream_dev_probe (fifo_mtu_multiplier=2, addr_dir_offset=2, addr_ch_offset=3)
ubuntu@ubuntu:~$ 
`
Does anyone have any ideas?
Thanks.
BR,
Mihai

alphafox02, 2024-03-17:
Should look like this

dtparam=audio=on
#dtparam=i2c_arm=on
#dtparam=spi_vc=on
dtparam=i2c_vc=on
dtoverlay=spi1-3cs
enable_uart=0
# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

alphafox02, 2024-03-17:
You'll also need a sudo chmod g+rw /dev/gpiomem each time you boot or put it in the /etc/rc.local

bferrell, 2025-03-11:
I finally identified the issue.

My Pi4 is in an ArgonOne M2 case.  The power board used GPIO pin2 and is interfering ARGH!!!