cariboulabs/cariboulite #186: Raspberry Pi 3 issue with SPI

User25514, 2024-02-09:
I want to use the cariboulite with python, but when I started the ./install.sh
This started coming up during the install:
`[INFO] SoapyCaribouliteSession, sessionCount: 0
02-09 02:59:51.999  1617  1617 E IO_UTILS_SPI io_utils_spi_add_chip@io_utils_spi.c:434 spi_init function failed with code -1, (SPI_ERR_OPEN)
02-09 02:59:52.001  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.001  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.001  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.001  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.001  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.042  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.042  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.042  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.042  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.042  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.084  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.084  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.084  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.084  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.084  1617  1617 W IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:606 generic function transfer not implemented
02-09 02:59:52.128  1617  1617 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:260 Sending bitstream of size 32220
02-09 02:59:55.678  1617  1617 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:292 FPGA programming - Success!`
`02-09 03:00:01.924  1617  1617 E CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1077 PLL didn't lock`

and then it continues with this:
`02-09 03:00:01.936  1617  1617 W AT86RF215_Main at86rf215_print_version@at86rf215.c:294 MODEM Version: not AT86RF215 IQ capable modem (product number: 0x00, versions 00)
02-09 03:00:01.936  1617  1617 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:486 The assembled modem is not AT86RF215 / IQ variant (product number: 0x00)
02-09 03:00:01.938  1617  1617 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:513 Self-test process finished with errors
02-09 03:00:01.939  1617  1617 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
[ERROR] cariboulite_init_driver() failed
02-09 03:00:01.940  1617  1617 E FPGA caribou_fpga_close@caribou_fpga.c:246 caribou_fpga_close: dev not initialized
02-09 03:00:01.940  1617  1617 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:465 FPGA communication release failed (-1)
02-09 03:00:01.940  1617  1617 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
[INFO] SoapyCaribouliteSession, sessionCount: 0
02-09 03:00:02.011  1654  1654 E IO_UTILS_SPI io_utils_spi_add_chip@io_utils_spi.c:434 spi_init function failed with code -1, (SPI_ERR_OPEN)`

But it ends and says that all went well.

When I run  `dmesg | grep spi` I get:
`[    9.005525] pinctrl-bcm2835 3f200000.gpio: pin gpio19 already requested by 3f600000.smi; cannot claim for 3f215080.spi
[    9.005560] pinctrl-bcm2835 3f200000.gpio: pin-19 (3f215080.spi) status -22
[    9.005595] spi-bcm2835aux 3f215080.spi: Error applying setting, reverse things back`
And when i run `lsmod | grep spi` I get:
`spi_bcm2835aux         16384  0`

When I use my python file to transmit, i get:
`02-09 03:07:09.251   735   735 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:486 The assembled modem is not AT86RF215 / IQ variant (product number: 0x00)
02-09 03:07:09.253   735   735 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:513 Self-test process finished with errors
02-09 03:07:09.254   735   735 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
[ERROR] cariboulite_init_driver() failed
Printing 'findCariboulite' Request:
    {channel: S1G}
    {driver: Cariboulite}
[INFO] Initializing DeviceID: 0, Label: CaribouLite S1G[91ca1668], ChannelType: S1G
[INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)
02-09 03:07:09.262   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.262   735   735 E FPGA caribou_fpga_set_sys_ctrl_tx_sample_gap@caribou_fpga.c:390 caribou_fpga_set_sys_ctrl_tx_sample_gap: dev not initialized
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.267   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.268   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.268   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.268   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.268   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
02-09 03:07:09.268   735   735 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:520 uninitialized device
[... 36 lines cut ...]
Signal [11] caught, with the following information:
   signal errno = 0
   signal process pid = 1930203137
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation`

These are the only things I have on in my config boot:
`dtparam=audio=on
display_auto_detect=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_overscan=1
[cm4]
otg_mode=1
[all]
[pi4]
arm_boost=1
[all]
dtparam=i2c_vc=on
dtoverlay=spi1-3cs
dtoverlay=smi
dtoverlay=smi-dev`
Please help

User25514, 2024-02-09:
And whenever I try to execute it in python, I get the error:
`soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
CaribouLite: Signal [11] received from pid=[1931251713]
Signal [11] caught, with the following information:
   signal errno = 0
   signal process pid = 1931251713
   signal process uid = 0
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation`

ImDroided, 2024-02-09:
The memory access error is a known issue and I have no idea whats causing it see my issue https://github.com/cariboulabs/cariboulite/issues/185

I dont know what OS you are using but you can head to my site https://radiomenace.com/2023/03/20/getting-cariboulite-working-in-dragonos-on-a-raspberry-pi/ and follow what I did some apps work some are mem access errors. CubicSDR works with both S1G and HiF but gqrx is a instant crash

User25514, 2024-02-10:
> The memory access error is a known issue and I have no idea whats causing it see my issue #185
>
> I dont know what OS you are using but you can head to my site [...]

I've realised the issue, so the cariboulite was finicky but the memory issue is only soapysdr's issue. Which means I will be spending the next couple days trying to find a fix.
I tested cariboulite by transmitting on a licence-exempt freq by just starting up `./cariboulite/build/cariboulite_test_app`.

ImDroided, 2024-02-11:
> I've realised the issue, so the cariboulite was finicky but the memory issue is only soapysdr's issue. Which means I will be spending the next couple days trying to find a fix. I tested cariboulite [...]

God speed my friend... God speed.

I wonder if updating to the latest Soapy would have any effect.

User25514, 2024-02-12:
I got it working!
I found that the tutorial I followed using https://github.com/cariboulabs/cariboulite/blob/main/installation.md#installation-details was partly wrong. Instead of installing soapysdr manually, you can automatically download and build soapy when running ./install in the cariboulite folder. 
But before that, i made sure to put this in my boot/config.txt file.
[all]
dtparam=i2c_vc=on
dtoverlay=spi1-3cs
dtoverlay=smi-dev

ImDroided, 2024-02-12:
> I got it working! I found that the tutorial I followed using https://github.com/cariboulabs/cariboulite/blob/main/installation.md#installation-details was partly wrong. Instead of installing [...]

Ok glad you got it. A lot of us are running DragonOS which has SoapySDR already installed so the install script skips it but looking at the github code hasn't been touched in long time and DrsagonOS is current anyways.

User25514, 2024-02-12:
Personally, if your having issues with soapysdr working with cariboulite, then I would uninstall it on dragon os and reinstall it with cariboulite