cariboulabs/cariboulite #189: AT86RF215 input IQ sync not reached on LVDS interface during TX

matteoserva, 2024-02-16:
I tried sending data via the SMI interface to the cariboulite.
By reading the registers on the AT86RF215 modem I see that the TX interface (from rpi to cariboulite) is correctly configured and ready to receive data.
The register IQIFC2.SYNC never becomes 1 when sending data normally.
If I enable the loopback mode, then the symc is reached.

This is the report from at86rf215_get_iq_if_cfg(&radio->sys->modem,&cfg,1) when sending data
```
Current I/Q interface settings:
   Loopback (RX => TX): disabled
   Drive strength: 4 mA
   Common mode voltage: 1.2 V
   I/Q interface for sub-GHz: enabled
   I/Q interface for 2.4-GHz: enabled
   I/Q Clock <=> Data skew: 4.906 ns
   Status 'Sync Failure': 0
   Status 'Failsafe mode': 0
   Status 'Is synchronized to incoming I/Q': 0
```

matteoserva, 2024-02-17:
update:
The bug was in the fpga code. Now it's syncing. I'll make a pull request once I'm ready.

andrea77F, 2025-11-02:
Hi sirs, i' able to to talk with AT86RF215 through spi (frequency, power settings etc...) and send LVDS I/Q data to it, anyway the only way to see a RF transmission on spectrum analyzer is to activate the loopback bit. Is there someone that can share basic c code setup sequence to activate LVDS? Other than that, i'm sending sync frame (32 empty bytes) then I/Q but i'm not sure is the right sequence). Can you help me?