cariboulabs/cariboulite #193: Sub 1GHz tuning ranges

drom, 2024-03-05:
Great project!

I was wondering if you tested Sub-1GHz tuning ranges for Tx and Rx?
Is there a sharp limits to 389.5-510 MHz / 779-1020 MHz ?
Is it PLL lock issue? or dividers?

matteoserva, 2024-03-05:
The documentation for the modem at86rf215 only states that 
> The frequencies must be within the supported range, otherwise the behavior is undefined. The device is not protected
> against incorrect frequency settings

From the datasheet i don't see anything preventing me from putting arbitrary values in the register containing the center frequency.

The cariboulite library, cariboulite_radio.c, performs an additional check on the requested frequencies and blocks values outside the allowed range.

drom, 2024-03-05:
What is I am explicitly interested in values outside of the range?
Anybody did experiments in the LAB?

matteoserva, 2024-03-09:
I expect the internal PLL to not lock outside its range.
Do you have a specific use case in mind? Why don't you use the wideband channel and avoid that limitation?