cariboulabs/cariboulite #195: transmission confusion

User25514, 2024-03-10:
I need advice as well as information on the kinda non-existing manual.

Currently Im using python and im trying to transmit audio data to a receiver. Im using soapysdr, but Im wondering if theirs a good way to transmit audio data without using CW =1 continous wave. 
Are their other options?

righthalfplane, 2024-03-10:
There are transmit problems. CW =1 works to some extent. The are some pull requests waiting maybe one of those will get things going better.

User25514, 2024-03-11:
> There are transmit problems. CW =1 works to some extent - other options seem to error. There are some pull requests waiting maybe one of those will get things going better.

Do you know what the other ones are so that I can atleast test them out?

matteoserva, 2024-03-11:
The official source code is missing the fpga code for synchronizing the transmission channel with the radio chip.
I fixed the transmission errors in my private branch, tested it by arbitrary waveforms from gnuradio. Right now there are some issues with the soapy API.
My plan is to make the branch public this weekend.

matteoserva, 2024-03-14:
If you feel brave, my branch is online but you have to be comfortable with working from a terminal.
Remember that it's a brach called "transmission_fix", not the main one.
[My branch](https://github.com/matteoserva/cariboulite/blob/transmission_fix/docs/notes/tutorial.md)

User25514, 2024-03-14:
Because this is for my dissertation, I'm trying to transmit a slow data stream atm just to prove the theory, but if I have time after I've proved my theory, then I will definitely try yours out.

martinpetrovaj, 2024-04-23:
@matteoserva Hi Matteo, just went ahead and tried to install your 'transmission_fix' branch by following [your manual](https://github.com/matteoserva/cariboulite/blob/transmission_fix/docs/notes/tutorial.md?plain=1), but I run into errors when compiling kernel drivers:

```
****@raspberrypi:~/repositories/cariboulite_matteo/driver/build $ cmake .. && make
-- Kernel release: 6.6.20+rpt-rpi-v8
-- Kernel headers: /usr/src/linux-headers-6.6.20+rpt-rpi-v8
-- Configuring done
-- Generating done
-- Build files have been written to: /home/martin/repositories/cariboulite_matteo/driver/build
compiling module smi_stream_dev.ko...
In file included from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/linux/linkage.h:7,
                 from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/linux/kernel.h:17,
                 from /home/martin/repositories/cariboulite_matteo/driver/build/smi_stream_dev.c:48:
/home/martin/repositories/cariboulite_matteo/driver/build/smi_stream_dev.c: In function ‘smi_stream_dev_probe’:
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/linux/export.h:29:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   29 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
/home/martin/repositories/cariboulite_matteo/driver/build/smi_stream_dev.c:1077:47: note: in expansion of macro ‘THIS_MODULE’
 1077 |         inst->smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
      |                                               ^~~~~~~~~~~
In file included from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/linux/device.h:31,
                 from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/linux/platform_device.h:13,
                 from /home/martin/repositories/cariboulite_matteo/driver/build/smi_stream_dev.c:51:
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  230 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/home/martin/repositories/cariboulite_matteo/driver/build/smi_stream_dev.c:1077:34: error: too many arguments to function ‘class_create’
 1077 |         inst->smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
      |                                  ^~~~~~~~~~~~
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/linux/device/class.h:230:29: note: declared here
  230 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /home/martin/repositories/cariboulite_matteo/driver/build/smi_stream_dev.o] Error 1
make[4]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /home/martin/repositories/cariboulite_matteo/driver/build] Error 2
make[3]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[2]: *** [CMakeFiles/smi_stream_dev.dir/build.make:79: smi_stream_dev] Error 2
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/smi_stream_dev.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

Is this a current issue you're aware of, or is it more likely an error on my part that I need to investigate?

alphafox02, 2024-08-06:
@matteoserva 

I'm finally getting around to trying this fork. But a few things are not clear.

You say "install original software" but don't show that you should run ./install and only say cd cariboulite_original.

Following that you mention "install my branch in cariboulite" which almost makes it sound like you want your fork git cloned into the above direct.

I'm assuming what you mean is to install the original with ./install and make sure it works? Then git clone your fork into another directory and checkout the tx branch? Or do you mean to install your fork with ./install as well?

You then say to compile the kernel and compile the software, however, iir would not have been pulled in using your steps. Notice in the error above it's saying 

```
 /home/martin/repositories/cariboulite_matteo/software/libcariboulite/src/iir

  does not contain a CMakeLists.txt file.
```
I copied iir from the original cariboulite build that appears after running ./install into your project. I was then able to compile the software which of course made the tx_test.

I created a flow graph as pictured, had socat etc installed, ran the command provided and connected to it with the gnuradio flowgraph. I show nothing transmitting over the air with the gain turned up. 

This leads me to wonder, was I supposed to install install your fork? I compiled the kernel driver, but did nothing with it? 

Can you please clarify your steps and I may be able to recreate the transmission capability on my end?

alphafox02, 2024-08-06:
I also noticed that on the original main that the test app does transmit a cw tone, but on your fork it does not. Maybe unrelated, but at least I know the hat can transmit.

matteoserva, 2024-08-06:
@alphafox02 Yeah sorry for the confusion. I'll update the documentation to clarify the steps.
Basically the steps are these:
1. Install the original software and make sure that it works.
2. clone my branch somewhere else and install it by following the same steps you used in step 1. It will overwrite the original software.
3. perform the step called "compile software". This will create the tx_test application
4. run tx_test

alphafox02, 2024-08-06:
I actually did try this. I used the latest original fully installed. I made sure it worked. I then installed yours, which would’ve installed over top of it. Your tool compiled, but after I built the GNURadio graph and ran your tool I saw no transmission over the air. I also saw no transmission using the default test app in your repo either. If I get a chance, I will refresh and start all over again.

User25514, 2024-08-06:
> If you feel brave, my branch is online but you have to be comfortable with working from a terminal. Remember that it's a brach called "transmission_fix", not the main one. [My branch](https://github.com/matteoserva/cariboulite/blob/transmission_fix/docs/notes/tutorial.md)

hows the branch developing coming along? Did you get transmissions to work properly in Python, or are you still struggling?

User25514, 2024-08-06:
> alphafox02

gnu radio didnt work with mine either, but if you have a receiver, check the receiver. Mine didn't show in gnu but it showed transmissions when I opened up my radio analyser from another pc

alphafox02, 2024-08-06:
I’m using SDR plus plus with an RTLSDR to check the 433 MHz frequency. Using the original and the test app to transmit a tone I can see it transmits so I know that is all working. However, when I try to use the method of sending a transmission in conjunction with gnuradio flow to transmit the wav file, I see no sign of any transmission. I’ll try it all again maybe while I’m at DEFCON.

alphafox02, 2024-08-06:
Actually, now thinking about it. I don’t see the transmit light come on at all when I’m using the method described in the transmit fix branch.

alphafox02, 2024-08-06:
And to clarify, I’m not using gnuradio in order to use the Caribou light as a source. I’m using gnuradio to pump the wav file into the tx test as described in this transmit fix branch mentioned above. I wasn’t sure if we were both talking about the same thing.

SinanAkkoyun, 2025-07-09:
Hi @matteoserva 
I tried your branch but am left with this:
`Frequency setting CH: 0, Wanted: 868300000.00 Hz, Set: 0.00 Hz `

and

`############################   Ora la configurazione mia`

full log of your tx example:

```
sinan@cm4:~/caribou_fix/cariboulite/software/libcariboulite/build$ 
sinan@cm4:~/caribou_fix/cariboulite/software/libcariboulite/build$ ./tx_test -c 0 -f 868300000 -r 4000000 -w 0 - 
DBG: RX Channel = 0
DBG: Frequency = 868300000.0 Hz
07-09 01:42:35.524  7018  7018 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:581 driver initializing
07-09 01:42:35.525  7018  7018 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:591 Initializing signals
07-09 01:42:35.525  7018  7018 D CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:731 setting up signal handler
07-09 01:42:35.526  7018  7018 D CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:206 Setting up board I/Os
07-09 01:42:35.526  7018  7018 D IO_UTILS_Main io_utils_setup@io_utils.c:25 initializing rpi
07-09 01:42:35.526  7018  7018 D IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:323 configuring gpio setups
07-09 01:42:35.526  7018  7018 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:623 Initializing FPGA
07-09 01:42:35.526  7018  7018 D FPGA caribou_fpga_init@caribou_fpga.c:119 configuring reset and irq pins
07-09 01:42:35.527  7018  7018 D FPGA caribou_fpga_init@caribou_fpga.c:127 Initializing io_utils_spi
07-09 01:42:35.529  7018  7018 D CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
07-09 01:42:35.529  7018  7018 D CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
07-09 01:42:35.529  7018  7018 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:631 Programming FPGA
07-09 01:42:35.529  7018  7018 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:210 FPGA already operational - not programming (use 'force_prog=true' to force update)
07-09 01:42:35.529  7018  7018 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:648 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
07-09 01:42:35.529  7018  7018 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:659 Detected Board Information:
07-09 01:42:35.529  7018  7018 D HAT hat_print_board_info@hat.c:705 # Board Info - Category name: hat
07-09 01:42:35.530  7018  7018 D HAT hat_print_board_info@hat.c:706 # Board Info - Product name: CaribouLite RPI Hat
07-09 01:42:35.530  7018  7018 D HAT hat_print_board_info@hat.c:707 # Board Info - Product ID: 0x0001, Numeric: 1
07-09 01:42:35.530  7018  7018 D HAT hat_print_board_info@hat.c:708 # Board Info - Product Version: 0x0001, Numeric: 1
07-09 01:42:35.530  7018  7018 D HAT hat_print_board_info@hat.c:709 # Board Info - Product UUID: 0822cd15-b185-4283-8dfa-799e667d3a61, Numeric serial: 0x345DD38A
07-09 01:42:35.530  7018  7018 D HAT hat_print_board_info@hat.c:710 # Board Info - Vendor: CaribouLabs LTD
07-09 01:42:35.530  7018  7018 D CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:797 # Board Info - Product Type: CaribouLite FULL
07-09 01:42:35.530  7018  7018 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 initializing submodules
07-09 01:42:35.530  7018  7018 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:284 INIT FPGA SMI communication
07-09 01:42:35.530  7018  7018 D CARIBOU_SMI caribou_smi_init@caribou_smi.c:521 initializing caribou_smi
07-09 01:42:35.531  7018  7018 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:294 INIT MODEM - AT86RF215
07-09 01:42:35.531  7018  7018 D AT86RF215_Main at86rf215_init@at86rf215.c:174 configuring reset and irq pins
07-09 01:42:35.532  7018  7018 D AT86RF215_Main at86rf215_init@at86rf215.c:182 Adding chip definition to io_utils_spi
07-09 01:42:35.533  7018  7018 D AT86RF215_Main at86rf215_init@at86rf215.c:211 Modem identity: Version: 03, Product: 35
07-09 01:42:35.533  7018  7018 D AT86RF215_Main at86rf215_calibrate_device@at86rf215.c:129 Calibration of modem channel 0...
07-09 01:42:35.601  7018  7018 D AT86RF215_Main at86rf215_calibrate_device@at86rf215.c:145 Calibration Results of the modem: I=33, Q=33
07-09 01:42:35.601  7018  7018 D AT86RF215_Main at86rf215_calibrate_device@at86rf215.c:129 Calibration of modem channel 1...
07-09 01:42:35.668  7018  7018 D AT86RF215_Main at86rf215_calibrate_device@at86rf215.c:145 Calibration Results of the modem: I=37, Q=35
07-09 01:42:35.668  7018  7018 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:304 Configuring modem initial state
07-09 01:42:35.669  7018  7018 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:347 This board is a Full version CaribouLite - setting ext_ref: modem, 32MHz
07-09 01:42:35.670  7018  7018 D CARIBOULITE Radio cariboulite_radio_ext_ref@cariboulite_radio.c:93 Setting ext_ref = 32MHz
07-09 01:42:35.670  7018  7018 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:373 INIT MIXER - RFFC5072
07-09 01:42:35.670  7018  7018 D RFFC5072 rffc507x_init@rffc507x.c:115 Initializing RFFC507x driver
07-09 01:42:35.670  7018  7018 D RFFC5072 rffc507x_init@rffc507x.c:119 Setting up device GPIOs
07-09 01:42:35.700  7018  7018 D RFFC5072 rffc507x_init@rffc507x.c:132 Received spi handle 3
[... cut ...]
07-09 01:42:36.072  7018  7018 D IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:482 removing an spi device with handle 3
07-09 01:42:36.072  7018  7018 D RFFC5072 rffc507x_release@rffc507x.c:212 Device release completed
07-09 01:42:36.072  7018  7018 D CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:244 Releasing board I/Os - closing SPI
07-09 01:42:36.072  7018  7018 D CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:247 Releasing board I/Os - io_utils_cleanup
07-09 01:42:36.072  7018  7018 D CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:752 driver released
```

matteoserva, 2025-07-10:
> 07-09 01:42:35.524  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:581 driver initializing
> 07-09 01:42:35.525  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:591 Initializing signals
> 07-09 01:42:35.525  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:731 setting up signal handler
> 07-09 01:42:35.526  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:206 Setting up board I/Os
> 07-09 01:42:35.526  7018  7018 D IO_UTILS_Main ***@***.***_utils.c:25 initializing rpi
> 07-09 01:42:35.526  7018  7018 D IO_UTILS_SPI ***@***.***_utils_spi.c:323 configuring gpio setups
> 07-09 01:42:35.526  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:623 Initializing FPGA
> 07-09 01:42:35.526  7018  7018 D FPGA ***@***.***_fpga.c:119 configuring reset and irq pins
> 07-09 01:42:35.527  7018  7018 D FPGA ***@***.***_fpga.c:127 Initializing io_utils_spi
> 07-09 01:42:35.529  7018  7018 D CARIBOU_PROG ***@***.***_prog.c:84 FPGA is already configured and running
> 07-09 01:42:35.529  7018  7018 D CARIBOU_PROG ***@***.***_prog.c:87 device init completed
> 07-09 01:42:35.529  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:631 Programming FPGA
> 07-09 01:42:35.529  7018  7018 I FPGA ***@***.***_fpga.c:210 FPGA already operational - not programming (use 'force_prog=true' to force update)
> 07-09 01:42:35.529  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:648 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
> 07-09 01:42:35.529  7018  7018 D HAT ***@***.***:705 # Board Info - Category name: hat
> 07-09 01:42:35.530  7018  7018 D HAT ***@***.***:706 # Board Info - Product name: CaribouLite RPI Hat
> [... cut ...]
> 07-09 01:42:36.072  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:244 Releasing board I/Os - closing SPI
> 07-09 01:42:36.072  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:247 Releasing board I/Os - io_utils_cleanup
> 07-09 01:42:36.072  7018  7018 D CARIBOULITE Setup ***@***.***_setup.c:752 driver released

Hi

It's been a while since I worked on this so I don't remember anymore how to
work with the code. It looks like you are using the original fpga firmware
instead of my own. With my firmware you should see a updated debug word and

'Is synchronized to incoming I/Q' going to 1.

Matteo

blanj, 2025-09-17:
Thanks.  Yes, I noticed a script in the original branch that was cycling through different seeds and I have been trying the same (so far without luck).