cariboulabs/cariboulite #198: install.sh fails on Raspberry Pi Zero 2 W, Pi OS Legacy 64bit, 2024-03-12 in SoapySDR  v0.8.1-gbb33b2d2 install stage

mrtnkhl, 2024-03-15:
Raspberry Pi Zero 2 W
Raspberry Pi OS legacy, 64-bit
Cloned this git repo at around UTC 2024-03-15 17:30

Followed the installation instructions on the Github page of this project and install.sh had no issues until it got to the SoapySDR parts (v0.8.1-gbb33b2d2).

Installer is now stuck, no progress on the installation for at least 15 minutes.

Relevant Installer output:

> -- SoapySDR version: v0.8.1-gbb33b2d2
-- ABI/so version: v0.8-3
-- Install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mck/projects/cariboulite/installations/SoapySDR/build
Scanning dependencies of target SoapySDR
[  2%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Device.cpp.o
[  5%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Factory.cpp.o
[  7%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Registry.cpp.o
[ 10%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Types.cpp.o
[ 13%] Building CXX object lib/CMakeFiles/SoapySDR.dir/NullDevice.cpp.o
[ 15%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Logger.cpp.o
[ 18%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Errors.cpp.o
[ 21%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Formats.cpp.o
[ 23%] Building CXX object lib/CMakeFiles/SoapySDR.dir/ConverterRegistry.cpp.o
[ 26%] Building CXX object lib/CMakeFiles/SoapySDR.dir/DefaultConverters.cpp.o
[ 28%] Building CXX object lib/CMakeFiles/SoapySDR.dir/TypesC.cpp.o
[ 31%] Building CXX object lib/CMakeFiles/SoapySDR.dir/ModulesC.cpp.o
[ 34%] Building CXX object lib/CMakeFiles/SoapySDR.dir/VersionC.cpp.o
[ 36%] Building CXX object lib/CMakeFiles/SoapySDR.dir/DeviceC.cpp.o
[ 39%] Building CXX object lib/CMakeFiles/SoapySDR.dir/FactoryC.cpp.o
[ 42%] Building CXX object lib/CMakeFiles/SoapySDR.dir/LoggerC.cpp.o
[ 44%] Building CXX object lib/CMakeFiles/SoapySDR.dir/TimeC.cpp.o
[ 47%] Building CXX object lib/CMakeFiles/SoapySDR.dir/ErrorsC.cpp.o
[ 50%] Building CXX object lib/CMakeFiles/SoapySDR.dir/FormatsC.cpp.o
[ 52%] Building CXX object lib/CMakeFiles/SoapySDR.dir/ConvertersC.cpp.o
[ 55%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Modules.cpp.o
[ 57%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Version.cpp.o
[ 60%] Linking CXX shared library libSoapySDR.so
[ 60%] Built target SoapySDR
Scanning dependencies of target SoapySDRUtil
[ 63%] Building CXX object apps/CMakeFiles/SoapySDRUtil.dir/SoapySDRUtil.cpp.o
[ 65%] Building CXX object apps/CMakeFiles/SoapySDRUtil.dir/SoapySDRProbe.cpp.o
[ 68%] Building CXX object apps/CMakeFiles/SoapySDRUtil.dir/SoapyRateTest.cpp.o
[ 71%] Linking CXX executable SoapySDRUtil
[ 71%] Built target SoapySDRUtil
Scanning dependencies of target TestConvertTypes
[ 73%] Building CXX object tests/CMakeFiles/TestConvertTypes.dir/TestConvertTypes.cpp.o
[ 76%] Linking CXX executable TestConvertTypes
[ 76%] Built target TestConvertTypes
Scanning dependencies of target TestKwargsMarkup
[ 78%] Building CXX object tests/CMakeFiles/TestKwargsMarkup.dir/TestKwargsMarkup.cpp.o
[ 81%] Linking CXX executable TestKwargsMarkup
[ 81%] Built target TestKwargsMarkup
Scanning dependencies of target TestFormatParser
[ 84%] Building CXX object tests/CMakeFiles/TestFormatParser.dir/TestFormatParser.cpp.o
[ 86%] Linking CXX executable TestFormatParser
[ 86%] Built target TestFormatParser
Scanning dependencies of target TestTimeConversion
[ 89%] Building CXX object tests/CMakeFiles/TestTimeConversion.dir/TestTimeConversion.cpp.o
[ 92%] Linking CXX executable TestTimeConversion
[ 92%] Built target TestTimeConversion
Scanning dependencies of target SoapySDR3_swig_compilation
[ 94%] Swig compile /home/mck/projects/cariboulite/installations/SoapySDR/build/swig/python/python3/SoapySDR.i for python
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:234: Warning 560: Unknown Doxygen command: parblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:252: Warning 560: Unknown Doxygen command: endparblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:255: Warning 560: Unknown Doxygen command: parblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:259: Warning 560: Unknown Doxygen command: endparblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:261: Warning 560: Unknown Doxygen command: parblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:265: Warning 560: Unknown Doxygen command: endparblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:234: Warning 560: Unknown Doxygen command: parblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:252: Warning 560: Unknown Doxygen command: endparblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:255: Warning 560: Unknown Doxygen command: parblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:259: Warning 560: Unknown Doxygen command: endparblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:261: Warning 560: Unknown Doxygen command: parblock.
/home/mck/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:265: Warning 560: Unknown Doxygen command: endparblock.
[ 94%] Built target SoapySDR3_swig_compilation
Scanning dependencies of target _SoapySDR3
[ 97%] Building CXX object swig/python/python3/CMakeFiles/_SoapySDR3.dir/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx.o
/home/mck/projects/cariboulite/installations/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx: In function ‘PyObject* PyInit__SoapySDR()’:
/home/mck/projects/cariboulite/installations/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:982:65: warning: ‘void PyEval_InitThreads()’ is deprecated [-Wdeprecated-declarations]
  982 | #     define SWIG_PYTHON_INITIALIZE_THREADS  PyEval_InitThreads()
      |                                                                 ^
/home/mck/projects/cariboulite/installations/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:56409:3: note: in expansion of macro ‘SWIG_PYTHON_INITIALIZE_THREADS’
56409 |   SWIG_PYTHON_INITIALIZE_THREADS;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.9/Python.h:145,
                 from /home/mck/projects/cariboulite/installations/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:180:
/usr/include/python3.9/ceval.h:130:37: note: declared here
  130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
      |                                     ^~~~~~~~~~~~~~~~~~

sethmnone, 2024-10-16:
My Zero 2W hangs at "[ 7%] Building CXX object lib/CMakeFiles/SoapySDR.dir/Registry.cpp.o"

Can't get past this...

aiaaaa, 2024-12-29:
I too am having the same issues. Hanging at [5%] building cxx object ... Factory.cpp.o