cariboulabs/cariboulite #200: (Updated Install Guide) DragonOS Pi64 35.1

alphafox02, 2024-03-17:
### Specific to DragonOS Pi64 35.1 +

1. Clone the CaribouLite repository from GitHub:

    ```bash
    git clone https://github.com/cariboulabs/cariboulite.git
    cd cariboulite
    ```

2. Navigate to the SMI stream driver file:

    ```bash
    nano driver/smi_stream_dev.c
    ```

3. Inside the editor, use Ctrl+W to search for "Create sysfs". Look for the line below it:

    ```c
    smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
    ```

   Modify it to:

    ```c
    smi_stream_class = class_create(DEVICE_NAME);
    ```

   Save the changes by pressing Ctrl+O, then Enter, and exit the editor by pressing Ctrl+X.

4. Run the installation script:

    ```bash
    ./install.sh
    ```

5. Modify the `/boot/firmware/config.txt` file:

    ```bash
    sudo nano /boot/firmware/config.txt
    ```

   Ensure the following lines are present:

    ```plaintext
    dtparam=audio=on
    #dtparam=i2c_arm=on
    #dtparam=spi=on
    dtparam=i2c_vc=on
    dtparam=spi1-3cs
    # Comment out the following line if the edges of the desktop appear outside
    # the edges of your display
    ```
6. Edit the `/etc/rc.local` file:

    ```bash
    sudo nano /etc/rc.local
    ```

   Add the following line right below `fi` and above `exit 0`:

    ```bash
    sudo chmod g+rw /dev/gpiomem
    ```

7. Reboot your system and enjoy using CaribouLite.

alphafox02, 2024-03-17:
Is there is a desire to use the Caribou Lite with SDRAngel no the Pi, make sure to start sdrangel from the command line 
sdrangel --soapy

ki6uve, 2024-03-18:
alphafox02:
This version of Cariboulite software doesn't seem to have dump1090 app at all.  Other versions had a /cpp folder under the /examples directory.  Did you install dump1090 manually from another github source?  Thanks!!

alphafox02, 2024-03-18:
It’s still there but under soapy folder now

https://github.com/cariboulabs/cariboulite/tree/main/examples/adsb_soapy

ki6uve, 2024-03-18:
Seemed to compile fine, but still throwing errors:
 IO_UTILS_SPI io_utils_spi_add_chip@io_utils_spi.c:434 spi_init function failed with code -1, (SPI_ERR_OPEN)

CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1099 PLL didn't lock

CaribouLite: Signal [11] received from pid=[-1446313462]
Signal [11] caught, with the following information:
   signal errno = 0
   signal process pid = -1446313462
   signal process uid = 65535
   signal status = 0
   signal errno / SIGSEGV / the process access a valid region of memory in an in                                              valid way - violated memory access permissions
SIGSEGV: memory access violation

alphafox02, 2024-03-18:
I don’t get why it runs for me, but not for you. 
[image]

alphafox02, 2024-03-18:
But I do get a similar failure on Gqrx.

ki6uve, 2024-03-18:
So weird.  Did you change any other permissions for the default ubuntu/dragon login?  My ubuntu login is always asking for sudo password.

ki6uve, 2024-03-18:
I do have a fan hat in between the Pi and the Cariboulite.  I wonder if that if F'ing it up.  I'll try removing that.

alphafox02, 2024-03-19:
You mean the login that comes up at each boot? That’s just the ubutnu user password it’s asking for. You can disable that in gdm3 if you want an auto login. I would try and remove that fan for sure and see what happens then. 

I’m also curious, are you looking for the dump1090 app to do much more then display results? It seems like all of what you’d expect out of the normal 1090 is missing.

ki6uve, 2024-03-19:
Definitely something didn't work during my install.  Even SoapySDRUtil --find is failing:
[image]

alphafox02, 2024-03-19:
Can you maybe share a picture of your Caribou? Perhaps it’s something specific to the model you have vs mine. It seem soapy finds it (ignore sdrplay comments) but I don’t get all that output you get before and after it finds the device.

alphafox02, 2024-03-19:
Add a glance it looks just like mine I think. I don’t know what would be the difference maybe something during the manufacturing or loading of the hat?

ImDroided, 2024-03-21:
This guide worked for me but I was warned about `dtoverlay=spi1-3cs` not being in the config.txt so I added it anyways. I wonder if that line may help @ki6uve. not sure the difference between dtparam and dtoverlay but it seems like we already added this flag above.

ki6uve, 2024-03-21:
Okay, finally got it working.  Damn typo in the rc.local line.  Now seem to just be sitting on that "starting stream loop".  How long did it take before adsb data started being detected?

alphafox02, 2024-03-21:
It just sits and runs but I’ve never seen it get anything. However, I did get adsb results in SDRAngel with it. 

I better go back and look at my guide above, sounds like I might have missed something in the config.txt.

alphafox02, 2024-03-21:
> This guide worked for me but I was warned about `dtoverlay=spi1-3cs` not being in the config.txt so I added it anyways. I wonder if that line may help @ki6uve. not sure the difference between dtparam and dtoverlay but it seems like we already added this flag above.

Thanks for catching that! I went back and fixed it.

hudouseko, 2024-08-02:
> 2. Navigate to the SMI stream driver file:
>    ```shell
>    nano driver/smi_stream_dev.c
>    ```
> 3. Inside the editor, use Ctrl+W to search for "Create sysfs". Look for the line below it:
>    ```c
>    smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
>    ```
>    
>    Modify it to:
>    ```c
>    smi_stream_class = class_create(DEVICE_NAME);
>    ```

This modification actually resolves as well the same issue I have experienced now running Raspbian bookworm, kernel 6.6.31-1+rpt-rpi-v6 on Raspberry Pi Zero WH. Thanks for that

MAVProxyUser, 2024-08-09:
> Definitely something didn't work during my install. Even SoapySDRUtil --find is failing: [...]

same here brother

sethmnone, 2024-10-17:
> If there is a desire to use the Caribou Lite with SDRAngel on the Pi, make sure to start sdrangel from the command line sdrangel --soapy

Can you confirm what hardware was used? 
Thx

Tek465, 2024-11-07:
Something has changed with (https://github.com/cariboulabs/cariboulite.git) and I needed to use an older build.

This method works as of 11.07.2024 and DragonOS build 37.

1) Use an older build ex. ( https://github.com/unixpunk/cariboulite.git )

```
git clone https://github.com/unixpunk/cariboulite.git
cd cariboulite
./install.sh
```

2) Edit files config.txt and rc.local

`sudo nano /boot/firmware/config.txt`

Ensure the following lines are present:
```
dtparam=audio=on
#dtparam=i2c_arm=on
#dtparam=spi=on
dtparam=i2c_vc=on
dtoverlay=spi1-3cs
```
`sudo nano /etc/rc.local`

Add the following line right below fi and above exit 0:
`sudo chmod g+rw /dev/gpiomem`

3) Reboot

alphafox02, 2025-02-08:
FYI - I made the latest DragonOS Pi64 39 image based on 24.04 w/ 6.8 kernel. No matter what I do so far I end up with no /dev/smi

Would be best to stick to a version prior to R39 for now.

ki6uve, 2025-02-08:
Can the Caribou Lite be configured as a drone hunter too?