cariboulabs/cariboulite #201: Not sure about the software...

ruslanbd, 2024-03-18:
Hi, I recently got the cariboulite from a friend. Tried messing around with it for a week, but failed to get tx working.
Does soapy tx even work? And if not, what is missing? The documentation is very much lacking. I would be happy to learn some verilog and try to help out, but digging through the codebase seems like quite a daunting task for an engineering student with a shit ton of homework, so if someone would just give me a short overview of the state of this repo and what needs to be done, it would be helpful for newcomers to know where to start.

matteoserva, 2024-03-19:
The official software doesn't support TX yet.
There is a working implementation for TX on my branch. The codebase is still pretty ugly but I'm working on it.
The branch is called "transmission fix"
https://github.com/matteoserva/cariboulite/blob/transmission_fix/docs/notes/tutorial.md

oldchap7, 2024-06-13:
Hi Matteo, here comes a question re the announcement on your Github Page: Did you follow up the attempt to work out a "Soapy Tx"? Any progress om that? I find that an excellent idea, as it would allow to operate Cariboulite RX and especially TX via SoapyRemote. Thanks.

matteoserva, 2024-06-13:
No. Sadly I hadn't had time to make improvements on the soapy TX side. Right now I don't know when I'll be able to make further progress.

oldchap7, 2024-06-14:
Ok, thanks for info. Good things will take time. Will keep an eye on this thread.