cariboulabs/cariboulite #208: EEPROM needs to be configured.

RSTChile, 2024-07-04:
Hi: 

I try many times to install Cariboulite, but this error appears anytime:

07-04 02:01:02.826  2219  2219 E EEPROM_UTILS eeprom_read@eeprom_utils.c:169 The eeprom driver for bus 0, adde 0x50 is not initialized

07-04 02:01:02.826  2219  2219 E HAT hat_init@hat.c:532 Reading from eeprom failed

07-04 02:01:02.826  2219  2219 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:604 Failed to detect the board in /proc/device-tree/hat - EEPROM needs to be configured.

07-04 02:01:02.826  2219  2219 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:605 Please run the cariboulite_prod application with sudo permissions...
[ERROR] cariboulite_init_driver() failed

Printing 'findCariboulite' Request:
07-04 02:01:02.828  2219  2224 I HAT hat_detect_board@hat.c:619 This board is not configured yet as a hat.
No devices found! 

I install in Ubuntu (Dragon OS), and Raspios. Many times.

My guess is that somehow the Cariboulite firmware got erased.

How can I reinstall Cariboulite Firmware?

Thank you

DumpsterBaby1990, 2025-01-03:
Same. Using these, it doesn't even populate as a hat. What am I missing? https://cariboulabs.github.io/cariboulite/docs/flashing/

bferrell, 2025-02-15:
I have the same issue.

EEPROM Data Structure

The code is located [here](https://github.com/cariboulabs/cariboulite/tree/main/software/libcariboulite/src/cariboulite_eeprom) and is based on [RPi EERPROM Utils Tools](https://github.com/raspberrypi/hats/tree/master/eepromutils) provided by RaspberryPi. The ID EEPROM internal structure is described [here](https://github.com/raspberrypi/hats/blob/master/eeprom-format.md).

cariboulite_eeprom is not found in the source tree. 404 page is displayed

ahodsdon, 2025-02-28:
@bferrell FWIW, it looks like the EEPROM code was moved from cariboulite_eeprom to hat:

https://github.com/cariboulabs/cariboulite/tree/main/software/libcariboulite/src/hat

I'm afraid I can't help further, as I don't have the board, but I'm curious if re-flashing with the tool solves your issue.

bferrell, 2025-02-28:
@ahodsdon It helps but even following the README directions found in libcariboulite and below, nothing called cariboulite_eeprom built and after reboot,  /proc/device-tree/hat/ is not populated.

It's unobvious what the flash utility might be called.  Looking at the code found in hat that checks for the values, it seems the utility would set the eeprom... were it built

ahodsdon, 2025-02-28:
@bferrell From the makefile, it looks like a library called "hat.lib" is placed in /usr/lib.

I *think* that just running the hat test executable ("test_hat") should re-flash the EEPROM, but this executable does not appear to be built by default. To do this, I think you'd need to uncomment lines 20+21 in CMakeLists.txt:

#add_executable(test_hat ${SOURCES})
#target_link_libraries(test_hat rt pthread ${EXTERN_LIBS})

bferrell, 2025-02-28:
@ahodsdon that errors out with 

[ 85%] Building C object CMakeFiles/test_hat.dir/test_hat.c.o
make[2]: *** No rule to make target '../io_utils/build/libio_utils.a', needed by 'test_hat'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/test_hat.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

I physically checked my device per https://cariboulabs.github.io/cariboulite/docs/flashing/

Where the document says it will be marked:   “CFG/1234”

Mine is marked "CFG/3210"

the front is marked R2.8

ahodsdon, 2025-02-28:
@bferrell No idea about the CFG stuff, I'm afraid. The lib it's complaining about is in 

https://github.com/cariboulabs/cariboulite/blob/main/software/libcariboulite/src/io_utils/

You could try (re-)building that directory on its own, but you're probably in dependency hell unless you rebuild the entire tree.

Best of luck!

bferrell, 2025-03-11:
Once the EEPROM configured /dev/smi did not appear

Found s=this was due to my use on an Argon One M2 case.
The power board uses GPIO pin2 and blocks that use
Please close