cariboulabs/cariboulite #219: Hanging at 11%... Pi Zero 2W

sethmnone, 2024-10-16:
As the title says... I loaded my Pi Zero 2 W with bullseye 6.1.21-* from [here](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2023-05-03/). 

Proceeded to complete these commands:

cd SoapySDR
mkdir build && cd build

And eveytime terminal hangs at 11% after I input:
cmake ../

Has anyone experienced this?

Alle1307, 2024-12-22:
I changed to bookworm and to a RPI 3B, when compiling use make -j 2 instead of 4, takes a bit longer but doesn't hang. Haven't got everything working but SoapySDRUtil --probe gives a result now.

bcshort, 2025-01-01:
If you haven't done so already, you may need to increase the swap file size. Out of the box the Pi Zero 2W is configured with 100MB swap.  With only 512MB memory, 100MB swap is going to cause pain for any compiling.

* edit file /etc/dphys-swapfile 
* change "CONF_SWAPSIZE=100" to something bigger - I set mine to "CONF_SWAPSIZE=1000" 
* Reboot Pi

Node if using purely on the SD card, this will speed up the degradation of the card due to writes and you are allowing more swapping to occur.