cariboulabs/cariboulite #221: RPI4 RX- SMI data synchronization failed - DragonOS / Bookworm

ngrabbs, 2024-10-28:
Hi.  I've been trying to get the cariboulite working on a raspberry pi 4 (i have several pi4's and run into the same issue on them all) and i've been having some struggles with the smi interface.  I also want to mention that I have access to three cariboulite hats.  The hats work on a pi zero running bookworm and also on  a pi3 running dragonOS.  I have tested DragonOS_Pi64_Beta36.img and beDragonOS_Pi64_Beta37.imgta36 of dragonOS with the same results as well as 2024-07-04-raspios-bookworm-arm64-lite.img.

The issue im running into is this: 
**E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed**
Very specifically when trying to receieve.  I can transmit a CW on interface 0 or 1 of the cariboulite in all cases just fine, I just cant receive.  

You will see in the git diff below that i've made a few changes to the code after reading through a ton of different issue requests.   I have also tried building without any changes to the code, just fixing the errors, and i've tried building without commenting out the headers in the install.sh scripts, both with  the same results.

I have tried building from the following repo branches all with the same results:
https://github.com/cariboulabs/cariboulite -> main
https://github.com/cariboulabs/cariboulite/tree/meexmachina-patch-1
https://github.com/unixpunk/cariboulite/tree/patch-1

I have tried installing the driver with both of these different methods / parameters with the same results:
```
ubuntu@ubuntu:~/cariboulite/driver$ 
  ./install.sh install 6 2 3
  ./install.sh install 2 2 3
```

I am pretty active with trying to resolve these issues so if you see something I can test I will glady try and report back the results.
Thanks
-Nick

====DRAGONOS DragonOS_Pi64_Beta36.img====
====BUILD DETAILS====
```
ubuntu@ubuntu:~/cariboulite$ git status
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   driver/install.sh
	modified:   driver/smi_stream_dev.c
	modified:   install.sh
	modified:   software/libcariboulite/src/caribou_smi/kernel/smi_stream_dev_gen.h
	modified:   software/libcariboulite/src/iir (untracked content)
	modified:   software/libcariboulite/src/soapy_api/Cariboulite.cpp
	modified:   software/libcariboulite/src/soapy_api/CaribouliteSession.cpp
	modified:   software/libcariboulite/src/soapy_api/CaribouliteStream.hpp
	modified:   software/utils/generate_bin_blob

no changes added to commit (use "git add" and/or "git commit -a")
ubuntu@ubuntu:~/cariboulite$ git diff driver/install.sh driver/smi_stream_dev.c install.sh software/libcariboulite/src/soapy_api/Cariboulite.cpp software/libcariboulite/src/soapy_api/CaribouliteSession.cpp software/libcariboulite/src/soapy_api/CaribouliteStream.hpp
diff --git a/driver/install.sh b/driver/install.sh
index 74ea739..a3fd05f 100755
--- a/driver/install.sh
+++ b/driver/install.sh
@@ -22,7 +22,7 @@ install() {
     printf "${GREEN}Installation started...${NC}\n"
     printf "\n[  1  ] ${GREEN}Updating kernel headers and needed software${NC}\n"
     sudo apt-get update
-    sudo apt-get -y install raspberrypi-kernel-headers module-assistant pkg-config libncurses5-dev cmake git
+    sudo apt-get -y install module-assistant pkg-config libncurses5-dev cmake git
 
     printf "\n[  2  ] ${GREEN}Compiling module${NC}\n"
     if [ -d "$BUILD_DIR" ]; then
diff --git a/driver/smi_stream_dev.c b/driver/smi_stream_dev.c
index e0d6618..c7f3fba 100644
--- a/driver/smi_stream_dev.c
+++ b/driver/smi_stream_dev.c
@@ -1050,7 +1050,7 @@ static int smi_stream_dev_probe(struct platform_device *pdev)
     }
 
     // Create sysfs entries with "smi-stream-dev"
-    smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
+    smi_stream_class = class_create(DEVICE_NAME);
[... 574 lines cut ...]
10-27 21:35:11.337  6337  6337 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1075 Activating channel 1, dir = RX, activate = 0
10-27 21:35:11.337  6337  6337 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:772 Setup Modem state to 2 (trx off)
10-27 21:35:11.337  6337  6337 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:437 CLOSE SMI
CaribouLite: Signal [11] received from pid=[-2057182169]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = -2057182169
   signal process uid = 65535
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation
10-27 21:35:11.338  6337  7107 D CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:743 driver being released
10-27 21:35:11.338  6337  7107 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1075 Activating channel 0, dir = RX, activate = 0
10-27 21:35:11.338  6337  7107 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:772 Setup Modem state to 2 (trx off)
10-27 21:35:11.338  6337  7107 E CARIBOU_SMI caribou_smi_set_driver_streaming_state@caribou_smi.c:35 failed setting smi stream state (0)
10-27 21:35:11.338  6337  7107 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1075 Activating channel 1, dir = RX, activate = 0
10-27 21:35:11.338  6337  7107 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:772 Setup Modem state to 2 (trx off)
10-27 21:35:11.339  6337  7107 E CARIBOU_SMI caribou_smi_set_driver_streaming_state@caribou_smi.c:35 failed setting smi stream state (0)
10-27 21:35:11.339  6337  7107 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:437 CLOSE SMI
Segmentation fault (core dumped)
ubuntu@ubuntu:~/cariboulite/build$ 
```

====END TESTING====
====END DRAGON64 36====

ngrabbs, 2024-10-28:
Hi.  I've been trying to get the cariboulite working on a raspberry pi 4 (i have several pi4's and run into the same issue on them all) and i've been having some struggles with the smi interface.  I also want to mention that I have access to three cariboulite hats.  The hats work on a pi zero running bookworm and also on  a pi3 running dragonOS.  I have tested DragonOS_Pi64_Beta36.img and beDragonOS_Pi64_Beta37.imgta36 of dragonOS with the same results as well as 2024-07-04-raspios-bookworm-arm64-lite.img.

The issue im running into is this: 
**E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed**
Very specifically when trying to receieve.  I can transmit a CW on interface 0 or 1 of the cariboulite in all cases just fine, I just cant receive.  

You will see in the git diff below that i've made a few changes to the code after reading through a ton of different issue requests.   I have also tried building without any changes to the code, just fixing the errors, and i've tried building without commenting out the headers in the install.sh scripts, both with  the same results.

I have tried building from the following repo branches all with the same results:
https://github.com/cariboulabs/cariboulite -> main
https://github.com/cariboulabs/cariboulite/tree/meexmachina-patch-1
https://github.com/unixpunk/cariboulite/tree/patch-1

I have tried installing the driver with both of these different methods / parameters with the same results:
```
ubuntu@ubuntu:~/cariboulite/driver$ 
  ./install.sh install 6 2 3
  ./install.sh install 2 2 3
```

I am pretty active with trying to resolve these issues so if you see something I can test I will glady try and report back the results.
Thanks
-Nick

====DRAGONOS DragonOS_Pi64_Beta36.img====
====BUILD DETAILS====
```
ubuntu@ubuntu:~/cariboulite$ git status
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   driver/install.sh
	modified:   driver/smi_stream_dev.c
	modified:   install.sh
	modified:   software/libcariboulite/src/caribou_smi/kernel/smi_stream_dev_gen.h
	modified:   software/libcariboulite/src/iir (untracked content)
	modified:   software/libcariboulite/src/soapy_api/Cariboulite.cpp
	modified:   software/libcariboulite/src/soapy_api/CaribouliteSession.cpp
	modified:   software/libcariboulite/src/soapy_api/CaribouliteStream.hpp
	modified:   software/utils/generate_bin_blob

no changes added to commit (use "git add" and/or "git commit -a")
ubuntu@ubuntu:~/cariboulite$ git diff driver/install.sh driver/smi_stream_dev.c install.sh software/libcariboulite/src/soapy_api/Cariboulite.cpp software/libcariboulite/src/soapy_api/CaribouliteSession.cpp software/libcariboulite/src/soapy_api/CaribouliteStream.hpp
diff --git a/driver/install.sh b/driver/install.sh
index 74ea739..a3fd05f 100755
--- a/driver/install.sh
+++ b/driver/install.sh
@@ -22,7 +22,7 @@ install() {
     printf "${GREEN}Installation started...${NC}\n"
     printf "\n[  1  ] ${GREEN}Updating kernel headers and needed software${NC}\n"
     sudo apt-get update
-    sudo apt-get -y install raspberrypi-kernel-headers module-assistant pkg-config libncurses5-dev cmake git
+    sudo apt-get -y install module-assistant pkg-config libncurses5-dev cmake git
 
     printf "\n[  2  ] ${GREEN}Compiling module${NC}\n"
     if [ -d "$BUILD_DIR" ]; then
diff --git a/driver/smi_stream_dev.c b/driver/smi_stream_dev.c
index e0d6618..c7f3fba 100644
--- a/driver/smi_stream_dev.c
+++ b/driver/smi_stream_dev.c
@@ -1050,7 +1050,7 @@ static int smi_stream_dev_probe(struct platform_device *pdev)
     }
 
     // Create sysfs entries with "smi-stream-dev"
-    smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
+    smi_stream_class = class_create(DEVICE_NAME);
[... 1358 lines cut ...]
10-28 13:28:10.694  1072  1072 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1075 Activating channel 1, dir = RX, activate = 0
10-28 13:28:10.694  1072  1072 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:772 Setup Modem state to 2 (trx off)
10-28 13:28:10.694  1072  1072 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:437 CLOSE SMI
CaribouLite: Signal [11] received from pid=[-1138031988]
Signal [11] caught, with the following information: 
   signal errno = 0
   signal process pid = -1138031988
   signal process uid = 127
   signal status = 0
   signal errno / SIGSEGV / the process access invalid region of memory
SIGSEGV: memory access violation
10-28 13:28:10.694  1072  1073 D CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:743 driver being released
10-28 13:28:10.694  1072  1073 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1075 Activating channel 0, dir = RX, activate = 0
10-28 13:28:10.694  1072  1073 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:772 Setup Modem state to 2 (trx off)
10-28 13:28:10.694  1072  1073 E CARIBOU_SMI caribou_smi_set_driver_streaming_state@caribou_smi.c:35 failed setting smi stream state (0)
10-28 13:28:10.694  1072  1073 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:1075 Activating channel 1, dir = RX, activate = 0
10-28 13:28:10.694  1072  1073 D CARIBOULITE Radio cariboulite_radio_set_modem_state@cariboulite_radio.c:772 Setup Modem state to 2 (trx off)
10-28 13:28:10.694  1072  1073 E CARIBOU_SMI caribou_smi_set_driver_streaming_state@caribou_smi.c:35 failed setting smi stream state (0)
10-28 13:28:10.694  1072  1073 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:437 CLOSE SMI
Segmentation fault
cubesat@raspberrypi:~/cariboulite/build $ 
```

====END TESTING====
====END BOOKWORM====

sethmnone, 2024-10-28:
We're you able to install SDR++ or any of its dependencies on the zero?

I ran into multiple dependency issues and could never get SoapySDR and SDR++ to play nicely.

ngrabbs, 2024-10-28:
> We're you able to install SDR++ or any of its dependencies on the zero?
> 
> I ran into multiple dependency issues and could never get SoapySDR and SDR++ to play nicely.

I haven't tried to install SDR++ on the zero.  In my case im looking to transmit from the zero via gnu radio python scripts and receive on a pi4.  (my gnu radio build from source has been compiling for almost 24 hours on the pi zero lol)

sethmnone, 2024-10-28:
> I haven't tried to install SDR++ on the zero.  In my case im looking to transmit from the zero via gnu radio python scripts and receive on a pi4.  (my gnu radio build from source has been compiling [...]

Gotcha, Good luck! 🫡

thomasbruninx, 2025-02-09:
I'm having the exact same issue both on Debian 11 (Bullseye) and 12 (Bookworm). Transmission works fine, but trying to receive gives me the exact same "SMI data synchronization failed" error. As far as I can understand it seems to be an issue with the SMI driver. But I haven't figured out what exactly yet.

mrtnkhl, 2025-02-12:
Seeing the same issue on a Pi Zero 2W, running 64bit Bookworm (2024-11-19 Raspberry image). Fresh compile, only applied 

curl -L https://patch-diff.githubusercontent.com/raw/cariboulabs/cariboulite/pull/210.patch | git apply -v
and manually
curl -L https://github.com/cariboulabs/cariboulite/pull/176.patch | git apply -v
following
https://github.com/cariboulabs/cariboulite/pull/176#issuecomment-2596019689)

see also
https://www.jeffgeerling.com/blog/2025/cariboulite-sdr-hat-sdr-on-raspberry-pi

'''
Linux pisdr 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
'''

`Feb 12 01:53:13 pisdr SoapySDRServer[1694]: 02-12 01:53:13.627  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:13 pisdr SoapySDRServer[1694]: 02-12 01:53:13.881  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:14 pisdr SoapySDRServer[1694]: 02-12 01:53:14.137  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:14 pisdr SoapySDRServer[1694]: 02-12 01:53:14.389  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:14 pisdr SoapySDRServer[1694]: 02-12 01:53:14.640  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:14 pisdr SoapySDRServer[1694]: 02-12 01:53:14.895  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:15 pisdr SoapySDRServer[1694]: 02-12 01:53:15.150  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:15 pisdr SoapySDRServer[1694]: 02-12 01:53:15.401  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:15 pisdr SoapySDRServer[1694]: 02-12 01:53:15.655  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:15 pisdr SoapySDRServer[1694]: 02-12 01:53:15.909  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:16 pisdr SoapySDRServer[1694]: 02-12 01:53:16.161  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:16 pisdr SoapySDRServer[1694]: 02-12 01:53:16.416  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:16 pisdr SoapySDRServer[1694]: 02-12 01:53:16.670  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:16 pisdr SoapySDRServer[1694]: 02-12 01:53:16.924  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 02-12 01:53:17.177  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 02-12 01:53:17.430  1694  1713 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: ......
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
Feb 12 01:53:17 pisdr SoapySDRServer[1694]: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
`

unixpunk, 2025-02-18:
Have a look at step 7 (and then reboot) here: https://github.com/cariboulabs/cariboulite/issues/200 - Might solve your SMI issues.  There have been complaints about pi zero, specifically, but I think its just the bladerf issue and affects anyone, really.  That said, I have a zero 2 WH on the way so I plan to specifically test that, also.  (I don't think a Zero 1 model is fast enough to really be useful with the Cariboulite, but I suppose I could test that also since I do have one WH)