cariboulabs/cariboulite #225: Can we get SD image files for Pi3 Pi4 and PiZ2...

aiaaaa, 2024-12-29:
I am coming back to this project after attempting to get the system up and running for a third separate time.

I appreciate that each time I come back to attempt, the git hub has been updated and things seem to be moving forward, but the standard instructions here do not get to a full install, something always seems to fail along the way. I will post separately a total issue note, and where it is failing, but this issue thread is for a simple request:

**Can we get SD card images with up and running software.**

This unit should work (according to the docs here) with a rPi 3, 4, and Z2. Can we please get 3 images, where the drivers, kernel, and software are installed with the soapySDR. Something slightly out of the box where we just need to change the host name and password.

Looking through the help docs, it seems like people with the z2 cannot get past the soapySDR install... (install.sh fails on Raspberry Pi Zero 2 W, Pi OS Legacy 64bit, 2024-03-12 in SoapySDR v0.8.1-gbb33b2d2 install stage #198)

great9, 2025-01-19:
it won't happpen. there is literally 0 response from the dev(s)

unixpunk, 2025-01-19:
@aiaaaa closest you'll get for now is DragonOS: https://sourceforge.net/projects/dragonos-pi64/  There is a text file on the desktop and in there it has the exact steps required to get the hw working, its almost copy/paste.  

I've only tested cariboulite on pi3b+ and pi4.  No pi5 support.  I've not tested DragonOS on pi zero's, you may have success there though.  

As for those particular softwares, its been a while since I've tried any of them w/cariboulite.  CubicSDR was the easiest to get working because of how well it enumerates Soapy devices.  I think I also had gqrx working but there's an issue w/pulseaudio not running as root but gqrx/CubicSDR needs to run as root unless you add your user to the proper group to have rw access to the device.  (which is part of the text file instructions I mentioned)  Hope this helps.

alphafox02, 2025-01-25:
I just put up a 24.04 based version of DragonOS Pi64 so I need to see if the steps still work/compile with a 6.8 kernel.

aiaaaa, 2025-01-25:
@unixpunk, @alphafox02, I could use DragonOS no problems, but that requires a pi4. My problem is I refuse to try anything other than the PiZero2. The form factor drew me to this device and I am going to soldier on.  I will try the CubicSDR and add that to my running notes list.

To add to this, If anyone has a SDR app that they know that works with the PiZero2 or at least knows works low memory let me know. I am prepping a large note to post to the issues page on my current progress with the piZero2. I just saw JeffG on YouTube just showcased the device TODAY (albeit with a pi4), so maybe we will get some eyes here. He is in his comments claiming the caribou works with z2 and openwebrx....sigh. He did suggest a SDR app I did not try on the z2 - GQRX, so I will try that this week.

@great9, I am thinking to fork this project, and build an SD card image for people to follow along, maybe a docker after it progresses. to your comment - dev(s) are moving slowly, as I mentioned every 6 months it seems I come back to some improvement here. It's interesting to see the radio berry take off, the libreSDR, and clones of the Pluto and other SDRs drop in price on aliexpress. The caribou is still the coolest and sleekest device paired with a pi zero form factor. 6 months ago, the install script would not integrate with soapy - now it will. This is progress in my eyes. also in troubleshooting the driver code, I familiarized myself with the AT86RF215 and its ASK FSK and QPSK functions, the RFFC5072 mixer (acting kinda like an LNB), and the SPI communication train that ties the processes together. This is my first radio and I am happy I didn't just immediately set up a RTL dongle and get board with with... this has been a fun learning process thus far.

CubicSDR and gqrx, time to try these, and if I get one to work I will create a SD card image to share with the community.

alphafox02, 2025-01-25:
I have a pi zero 2 coming this weekend and so I will see if my new 24.04 based image will run on it now.

great9, 2025-01-25:
> [@great9](https://github.com/great9), I am thinking to fork this project, and build an SD card image for people to follow along, maybe a docker after it progresses. to your comment - dev(s) are [...]

one of the latest commits introduced a bug which led to compilter spitting out many errors and not being able to install the drivers. I quickly fixed it (2-3 lines had to be changed) and [pushed a PR](https://github.com/cariboulabs/cariboulite/pull/217). They completely ignored it and [published a PR](https://github.com/cariboulabs/cariboulite/pull/220) which fixes the same thing 3 days later, merging it a week later.

This project would be much better off forked and actually worked on.

my setup is customized so i won't be using an SD card image as I'm running other things on that Pi.

Tek465, 2025-01-27:
Have you had a look at installation.md instead of the README.md on the Cariboulite Git? It looks like Bookworm added a few problems.

" Note: Major issues with the newly published distro bookworm - the required kernel modules cannot compile agains the latest 6.1.0 'bookworm' distribution. We noticed that the kernel-headers in the latest rapsberry-pi image distributions do not agree with the mainline linux kernel modules compilation process. Same issues exist with other kernel modules compilations in other projects. This issue needs to be further inspected. Therefore we suggest using the latest bullseye distribution bullseye 6.1.21-*. See details bellow. "

unixpunk, 2025-03-10:
@aiaaaa @Tek465 @great9 HTH: https://github.com/cariboulabs/cariboulite/discussions/234

jadware, 2025-04-07:
the installation is quite simple now with a z2 and works perfectly on a fresh flash of buster and bullseye, have not tested on bookworm