cariboulabs/cariboulite #230: Gqrx don't see interfaces

MaciejStawiar, 2025-02-18:
Hi,
I followed the installation guide from installation.md file and everything went good. All the tests mentioned in the guide are passed.
But Gqrx installed on the same RPi don't see the interfaces. I tried with "other.." and then I put all the parameters from the `SoapySDRUtil --find` :

> $ SoapySDRUtil --find
> ######################################################
> ##     Soapy SDR -- the SDR abstraction library     ##
> ######################################################
> 
> [INFO] SoapyCaribouliteSession, sessionCount: 0
> 02-18 19:30:10.513  2206  2206 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:210 FPGA already operational - not programming (use 'force_prog=true' to force update)
> Printing 'findCariboulite' Request:
> Found device 0
>   channel = S1G
>   device_id = 0
>   driver = Cariboulite
>   label = CaribouLite S1G[5e3122]
>   name = CaribouLite RPI Hat
>   serial = 5e3122
>   uuid = a6331dd1-0beb-49ba-adf7-4e598f1fc3e5
>   vendor = CaribouLabs LTD
>   version = 0x0001
> 
> Found device 1
>   channel = HiF
>   device_id = 1
>   driver = Cariboulite
>   label = CaribouLite HiF[5e3123]
>   name = CaribouLite RPI Hat
>   serial = 5e3123
>   uuid = a6331dd1-0beb-49ba-adf7-4e598f1fc3e5
>   vendor = CaribouLabs LTD
>   version = 0x0001
> 

Gqrx opens and when I press "Play" it crashes with info:

> gqrx exited with an exception:
> In hierarchical block source_impl, output 1 is not connected internally
> More information can be found by launching it from the console.

Running from console gives such info:

> $ gqrx 
> gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
> built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
> gr::log :WARN: file_source0 - file size is not a multiple of item size
> Resampling audio 96000 -> 48000
> BandPlanFile is /home/maciek/.config/gqrx/bandplan.csv
> BookmarksFile is /home/maciek/.config/gqrx/bookmarks.csv
> [INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
> gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
> built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
> [INFO] Using format CF32.
> gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
> built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
> [INFO] Using format CF32.
> Qt has caught an exception thrown from an event handler. Throwing
> exceptions from an event handler is not supported in Qt.
> You must not let any exception whatsoever propagate through Qt code.
> If that is not possible, in Qt 5 you must at least reimplement
> QCoreApplication::notify() and catch all exceptions there.

What I noticed is after reboot, both LEDs LD1 and LD2 are on, but when I use SoapySDRUtil they turns off and remain off.

MaciejStawiar, 2025-02-18:
Installed CubicSDR and it also don't see the CaribouLite.

unixpunk, 2025-02-18:
@MaciejStawiar In case you hadn't seen this issue yet: https://github.com/cariboulabs/cariboulite/issues/200 I believe you can resolve this with the commands in step 7, but the rest is maybe worth a review as well.  There are slight differences between OSes, though.  Hope that helps!