cariboulabs/cariboulite #231: Incomplete documentation/instructions

lohriialo, 2025-02-22:
It's been a very tough experience owning a Cariboulite. Got the 6ghz unit in 2025, haven't been able to get it working so far on a Pi 4B

Successfully followed through the installation instructions at https://github.com/cariboulabs/cariboulite/blob/main/installation.md

@meexmachina please add an example(video/doc) on how to configure with SDR++ or GQRX locally on the pi or remotely accessing it on another machine.

lohriialo, 2025-02-26:
@Oisins could you please share how you did it using SoapyRemote?

MaciejStawiar, 2025-02-26:
This is the issue I described right below Your issue. This is why I opened a return request, I will send it back and buy HackRF instead...

Oisins, 2025-02-26:
@lohriialo I got the Cariboulite working on a Pi4 - 64bit - Debian Bookworm and gqrx. I will post how I did it soon, when I've cleaned it all up. For SoapyRemote I just started the version installed by the cariboulite install script (SoapySDRServer --bind) and then connected with `driver=remote,remote=tcp://127.0.0.1:55312,soapy=0,serial=<Your Serial Number>,remote:driver=Cariboulite`.

Habraken, 2025-05-02:
I have a carbioulite working on a Pi0 and a Pi4 both running the latest version of Raspberry Pi OS Bookworm. I have the recepi on all of the installation steps in my fork of cariboulite on github: https://github.com/Habraken/cariboulite/blob/main/ADDITIONAL-README.md

MaciejStawiar, 2025-05-05:
> I have a carbioulite working on a Pi0 and a Pi4 both running the latest version of Raspberry Pi OS Bookworm. I have the recepi on all of the installation steps in my fork of cariboulite on github: https://github.com/Habraken/cariboulite/blob/main/ADDITIONAL-README.md I use it mostly remote with SDR++ and on the pi I run SDR++ Server. However the standard SDR++ package does not have SoapySDR Source complied in as a standard, so you will have to build it from source. 

@Habraken , thanks for your work! I followed your additional manual and it works for a second (yuppi!) 

Then it crashed with the "SMI data synchronization" failed info. HW is Pi0 2W

> [05/05/2025 16:43:56.000] [INFO] SoapyModule 'Soapy Source': Menu Select!
> [05/05/2025 16:43:56.000] [INFO] Ready, listening on 0.0.0.0:5259
> [05/05/2025 16:44:13.000] [INFO] Connection from TODO:TODO
> [05/05/2025 16:44:49.000] [INFO] Connection from TODO:TODO
> [05/05/2025 16:44:59.000] [INFO] SoapyModule 'Soapy Source': Tune: 100898537.335938!
> Printing 'findCariboulite' Request:
>     {channel: HiF}
>     {device_id: 1}
>     {driver: Cariboulite}
>     {label: CaribouLite HiF[5e3123]}
>     {name: CaribouLite RPI Hat}
>     {serial: 5e3123}
>     {uuid: a6331dd1-0beb-49ba-adf7-4e598f1fc3e5}
>     {vendor: CaribouLabs LTD}
>     {version: 0x0001}
> [INFO] Initializing DeviceID: 1, Label: CaribouLite HiF[5e3123], ChannelType: HiF
> [INFO] Creating SampleQueue MTU: 131072 I/Q samples (524288 bytes)
> [WARNING] setSampleRate: using rounded rate 1333000 is deprecated; use 4e6/3 or 1333333.3.
> [INFO] setFrequency dir: 1, channel: 0, freq: 100898537.95
> [INFO] setupStream: dir= RX, format= CF32
> [05/05/2025 16:44:59.000] [INFO] SoapyModule 'Soapy Source': Start!
> [INFO] setFrequency dir: 1, channel: 0, freq: 100898537.95
> [05/05/2025 16:44:59.000] [INFO] SoapyModule 'Soapy Source': Tune: 100898537.335938!
> FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................ 
> 05-05 16:44:59.430   596   628 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
> FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................ 
> 05-05 16:44:59.431   596   628 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:1276 SMI data synchronization failed
> FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................