cariboulabs/cariboulite #236: On a Pi 4 with the latest Bookworm (Pi OS 12):

Habraken, 2025-04-21:
On a Pi 4 with the latest Bookworm (Pi OS 12):

```
$ uname -a
Linux pi4-sdr 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

When compiling the driver I get an error relating to the same function in this PR:

```
/home/pi/Downloads/cariboulite/driver/build/smi_stream_dev.c: In function ‘smi_stream_dev_probe’:
/usr/src/linux-headers-6.6.51+rpt-common-rpi/include/linux/export.h:29:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   29 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
/home/pi/Downloads/cariboulite/driver/build/smi_stream_dev.c:1053:37: note: in expansion of macro ‘THIS_MODULE’
 1053 |     smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
      |                                     ^~~~~~~~~~~
In file included from /usr/src/linux-headers-6.6.51+rpt-common-rpi/include/linux/device.h:31,
                 from /usr/src/linux-headers-6.6.51+rpt-common-rpi/include/linux/platform_device.h:13,
                 from /home/pi/Downloads/cariboulite/driver/build/smi_stream_dev.c:51:
/usr/src/linux-headers-6.6.51+rpt-common-rpi/include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  230 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/home/pi/Downloads/cariboulite/driver/build/smi_stream_dev.c:1053:24: error: too many arguments to function ‘class_create’
 1053 |     smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
      |                        ^~~~~~~~~~~~
...
make[5]: *** [/usr/src/linux-headers-6.6.51+rpt-common-rpi/scripts/Makefile.build:248: /home/pi/Downloads/cariboulite/driver/build/smi_stream_dev.o] Error 1
make[4]: *** [/usr/src/linux-headers-6.6.51+rpt-common-rpi/Makefile:1946: /home/pi/Downloads/cariboulite/driver/build] Error 2
make[3]: *** [/usr/src/linux-headers-6.6.51+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[2]: *** [CMakeFiles/smi_stream_dev.dir/build.make:79: smi_stream_dev] Error 2
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/smi_stream_dev.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
binary file doesn't exist: /home/pi/Downloads/cariboulite/driver/build/smi_stream_dev.ko
```

Applying the patch with `curl -L https://github.com/cariboulabs/cariboulite/pull/176.patch | git apply -v` I get:

```
Checking patch driver/smi_stream_dev.c...
error: while searching for:
	}

	// Create sysfs entries with "smi-stream-dev"
	smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
	ptr_err = smi_stream_class;
	if (IS_ERR(ptr_err))
	{

error: patch failed: driver/smi_stream_dev.c:1316
error: driver/smi_stream_dev.c: patch does not apply
```

So I applied the change manually, and here is my diff:

```patch
pi@pi4-sdr:~/Downloads/cariboulite $ git diff driver/smi_stream_dev.c
diff --git a/driver/smi_stream_dev.c b/driver/smi_stream_dev.c
index e0d6618..c7f3fba 100644
--- a/driver/smi_stream_dev.c
+++ b/driver/smi_stream_dev.c
@@ -1050,7 +1050,7 @@ static int smi_stream_dev_probe(struct platform_device *pdev)
     }
 
     // Create sysfs entries with "smi-stream-dev"
-    smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
+    smi_stream_class = class_create(DEVICE_NAME);
     ptr_err = smi_stream_class;
     if (IS_ERR(ptr_err))
     {
```

After re-running `install.sh`, it compiles now, but when it's installing the driver I see some errors like:

```
01-16 09:23:53.090  7491  7491 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:539 couldn't open smi driver file '/dev/smi' (Permission denied)
01-16 09:23:53.090  7491  7491 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule
01-16 09:23:53.090  7491  7491 E IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:487 the device is already empty
01-16 09:23:53.091  7491  7491 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
01-16 09:23:53.091  7491  7491 E FPGA caribou_fpga_close@caribou_fpga.c:247 caribou_fpga_close: dev not initialized
[ERROR] cariboulite_init_driver() failed
01-16 09:23:53.095  7491  7491 E FPGA caribou_fpga_close@caribou_fpga.c:247 caribou_fpga_close: dev not initialized
01-16 09:23:53.095  7491  7491 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:465 FPGA communication release failed (-1)
01-16 09:23:53.095  7491  7491 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:340 closing uninitialized device
```

After a reboot, I launched SoapySDRServer (`sudo systemctl start SoapySDRServer.service`), and was able to connect to the H1F and S1G bands from GQRX on my Mac remotely.

_Originally posted by @geerlingguy in https://github.com/cariboulabs/cariboulite/issues/176#issuecomment-2596019689_

Habraken, 2025-04-21:
After applying the patch manually, it got things working for me on DragonOS Beta37.

For Bookworm there a few additional edits needed:

@@ -60,10 +60,11 @@
 #include <linux/mutex.h>
 #include <linux/kfifo.h>
 #include <linux/wait.h>
 #include <linux/poll.h>
 #include <linux/init.h>
+#include <linux/vmalloc.h>
 
 #include "smi_stream_dev.h"
 
 
 // MODULE SPECIFIC PARAMETERS
@@ -1048,11 +1049,11 @@ static int smi_stream_dev_probe(struct platform_device *pdev)
         dev_err(dev, "could not load smi_stream_dev");
         return err;
     }
 
     // Create sysfs entries with "smi-stream-dev"
-    smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
+    smi_stream_class = class_create(DEVICE_NAME);
     ptr_err = smi_stream_class;
     if (IS_ERR(ptr_err))
     {
         cdev_del(&smi_stream_cdev);
         unregister_chrdev_region(smi_stream_devid, 1);
@@ -1100,22 +1101,21 @@ static int smi_stream_dev_probe(struct platform_device *pdev)
 *
 *   smi_stream_remove - called when the driver is unloaded.
 *
 ***************************************************************************/
 
-static int smi_stream_dev_remove(struct platform_device *pdev)
+static void smi_stream_dev_remove(struct platform_device *pdev)
 {
     //if (inst->reader_thread != NULL) kthread_stop(inst->reader_thread);
     //inst->reader_thread = NULL;	
     
     device_destroy(smi_stream_class, smi_stream_devid);
     class_destroy(smi_stream_class);
     cdev_del(&smi_stream_cdev);
     unregister_chrdev_region(smi_stream_devid, 1);
 
     dev_info(inst->dev, DRIVER_NAME": smi-stream dev removed");
-    return 0;
 }
 
 /****************************************************************************
 *
 *   Register the driver with device tree

Habraken, 2025-04-21:
Applying the patch manually, got things working for me on DragonOS Beta37 on a Pi4.

For Bookworm, a few additional changes to `smi_stream_dev.c` are required:

The following include statement needs be added:
```#include <linux/vmalloc.c>```

And the ```smi_stream_dev_remove``` function needs to be modified like so:

```
static void smi_stream_dev_remove(struct platform_device *pdev)
{
    //if (inst->reader_thread != NULL) kthread_stop(inst->reader_thread);
    //inst->reader_thread = NULL;

    device_destroy(smi_stream_class, smi_stream_devid);
    class_destroy(smi_stream_class);
    cdev_del(&smi_stream_cdev);
    unregister_chrdev_region(smi_stream_devid, 1);

    dev_info(inst->dev, DRIVER_NAME": smi-stream dev removed”);
}

```

nowls, 2025-04-22:
See https://github.com/cariboulabs/cariboulite/pull/237 for compiling with the 1 or 2 parameter `class_create` based on kernel version.