cariboulabs/cariboulite #240: Failed to build on RPi-Zero

kcman100, 2025-04-27:
Cannot get the smi interface to work on RPi Zero W rev 1.1

Just did a fresh/clean OS install with the Raspberry Pi Imager.

At bottom of config.txt:
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
dtparam=spi=off
[all]
# For the Caribou Lite SDR
dtparam=i2c_vc=on
dtoverlay=spi1-3cs
arm_64bit=0

ken@RPiZero2W:~/projects/cariboulite $ uname -a
Linux RPiZero2W 6.12.20+rpt-rpi-v6 #1 Raspbian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) armv6l GNU/Linux

ken@RPiZero2W:~/projects/cariboulite $ pinout
Description        : Raspberry Pi Zero W rev 1.1
Revision           : 9000c1
SoC                : BCM2835
RAM                : 512MB
Storage            : MicroSD
USB ports          : 1 (of which 0 USB3)
Ethernet ports     : 0 (0Mbps max. speed)
Wi-fi              : True
Bluetooth          : True
Camera ports (CSI) : 1
Display ports (DSI): 0

ken@RPiZero2W:~/projects/cariboulite/examples/c_api/build $ sudo ./test_app
Detection: 1, HWVer: 1, HWName: CaribouLite 6G, UUID: e367d55f-7a21-4137-becb-7e1463ce0016
Version: 01.02.00
04-26 23:18:49.804  1629  1629 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:539 couldn't open smi driver file '/dev/smi' (No such file or directory)
04-26 23:18:49.807  1629  1629 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 Error setting up smi submodule
04-26 23:18:49.811  1629  1629 E FPGA caribou_fpga_close@caribou_fpga.c:247 caribou_fpga_close: dev not initialized
CaribouLite driver init failed, terminating...Serial Number: 278DF7B0
Channel: 0, Name: CaribouLite S1G, Num. Freq. Ranges: 2
   Range 0:  [377000000.00, 530000000.00]
   Range 1:  [779000000.00, 1020000000.00]
Channel: 1, Name: CaribouLite 6GHz, Num. Freq. Ranges: 1
   Range 0:  [1000000.00, 6000000000.00]

ken@RPiZero2W:~/projects/cariboulite $ ./install.sh

[  1  ] CaribouLite Git Repo
Already up to date.

[  2  ] Updating system and installing dependencies...
Hit:1 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Hit:2 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-raspi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
module-assistant is already the newest version (0.11.11).
pkg-config is already the newest version (1.8.1-1).
libncurses5-dev is already the newest version (6.4-4).
cmake is already the newest version (3.25.1-1).
git is already the newest version (1:2.39.5-0+deb12u2).
[... 156 lines cut ...]
/usr/src/linux-headers-6.12.20+rpt-common-rpi/include/linux/device/class.h:228:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  228 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:1053:24: error: too many arguments to function ‘class_create’
 1053 |     smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
      |                        ^~~~~~~~~~~~
/usr/src/linux-headers-6.12.20+rpt-common-rpi/include/linux/device/class.h:228:29: note: declared here
  228 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c: At top level:
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:1134:15: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Werror=incompatible-pointer-types]
 1134 |     .remove = smi_stream_dev_remove,
      |               ^~~~~~~~~~~~~~~~~~~~~
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:1134:15: note: (near initialization for ‘smi_stream_dev_driver.<anonymous>.remove’)
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:643:13: warning: ‘stream_smi_write_dma_callback’ defined but not used [-Wunused-function]
  643 | static void stream_smi_write_dma_callback(void *param)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: /home/ken/projects/cariboulite/driver/build/smi_stream_dev.o] Error 1
make[4]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /home/ken/projects/cariboulite/driver/build] Error 2
make[3]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make[2]: *** [CMakeFiles/smi_stream_dev.dir/build.make:79: smi_stream_dev] Error 2
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/smi_stream_dev.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
binary file doesn't exist: /home/ken/projects/cariboulite/driver/build/smi_stream_dev.ko

kcman100, 2025-04-29:
Seems like in addition to the kernel "/usr/src/linux-headers-6.12.20+rpt-rpi-v6" also need "linux-headers-6.12.20+rpt-common-rpi"
Or is that already in the path?
How to add this?

Habraken, 2025-05-01:
please see my pull request and issue https://github.com/cariboulabs/cariboulite/pull/241 where I show that in ```…/cariboulite/driver/smi_stream_dev.c``` the following ```#include <linux/vmalloc.h>``` needs to be added.

Habraken, 2025-05-01:
I install kernel headers with: ```sudo apt install linux-headers-rpi-v8``` although if you have install the Bookworm Lite version I get a message from apt that they are already the latest.

Habraken, 2025-05-01:
I have two CaribouLite's running on Pi-Zerro 2W’s running the latest official Raspberry Pi OS Bookwork Lite (64bit). However the max. sample rate should not exceed 2MS/s.

kcman100, 2025-05-04:
I added the header file linux/vmalloc.h to smi_stream_dev.c
That helped, but still not quite working.
I am loading up fresh the current "lite" bookworm OS with the official raspberry pi imager.

As you pointed out, it says it has the latest version of the linux headers:
ken@RPiZero2Wb:~/projects/cariboulite $ sudo apt install linux-headers-rpi-v8
linux-headers-rpi-v8 is already the newest version (1:6.12.25-1+rpt1).

Still in the output from running ./install.sh:
E: Unable to locate package linux-headers-raspi

Also see:
depmod: ERROR: failed to load symbols from /lib/modules/6.12.25+rpt-rpi-v8/kernel/drivers/char/broadcom/smi_stream_dev.ko.xz: Invalid argument

And then the error that causes the build to fail:
compiling module smi_stream_dev.ko...
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:178:6: warning: no previous prototype for ‘print_smil_registers_ext’ [-Wmissing-prototypes]
  178 | void print_smil_registers_ext(const char* b)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/linux-headers-6.12.25+rpt-common-rpi/arch/arm64/include/asm/alternative.h:9,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/arch/arm64/include/asm/lse.h:14,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/arch/arm64/include/asm/cmpxchg.h:14,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/arch/arm64/include/asm/atomic.h:16,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/include/linux/atomic.h:7,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/include/asm-generic/bitops/atomic.h:5,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/arch/arm64/include/asm/bitops.h:25,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/include/linux/bitops.h:68,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/include/linux/kernel.h:23,
                 from /home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:48:
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c: In function ‘smi_stream_dev_probe’:
/usr/src/linux-headers-6.12.25+rpt-common-rpi/include/linux/init.h:180:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  180 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:1054:37: note: in expansion of macro ‘THIS_MODULE’
 1054 |     smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
      |                                     ^~~~~~~~~~~
In file included from /usr/src/linux-headers-6.12.25+rpt-common-rpi/include/linux/device.h:31,
                 from /usr/src/linux-headers-6.12.25+rpt-common-rpi/include/linux/platform_device.h:13,
                 from /home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:51:
/usr/src/linux-headers-6.12.25+rpt-common-rpi/include/linux/device/class.h:228:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  228 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:1054:24: error: too many arguments to function ‘class_create’
 1054 |     smi_stream_class = class_create(THIS_MODULE, DEVICE_NAME);
      |                        ^~~~~~~~~~~~
/usr/src/linux-headers-6.12.25+rpt-common-rpi/include/linux/device/class.h:228:29: note: declared here
  228 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c: At top level:
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:1135:15: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Werror=incompatible-pointer-types]
 1135 |     .remove = smi_stream_dev_remove,
      |               ^~~~~~~~~~~~~~~~~~~~~
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:1135:15: note: (near initialization for ‘smi_stream_dev_driver.<anonymous>.remove’)
/home/ken/projects/cariboulite/driver/build/smi_stream_dev.c:644:13: warning: ‘stream_smi_write_dma_callback’ defined but not used [-Wunused-function]
  644 | static void stream_smi_write_dma_callback(void *param)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [/usr/src/linux-headers-6.12.25+rpt-common-rpi/scripts/Makefile.build:249: /home/ken/projects/cariboulite/driver/build/smi_stream_dev.o] Error 1
make[4]: *** [/usr/src/linux-headers-6.12.25+rpt-common-rpi/Makefile:1974: /home/ken/projects/cariboulite/driver/build] Error 2
make[3]: *** [/usr/src/linux-headers-6.12.25+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make[2]: *** [CMakeFiles/smi_stream_dev.dir/build.make:79: smi_stream_dev] Error 2
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/smi_stream_dev.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
binary file doesn't exist: /home/ken/projects/cariboulite/driver/build/smi_stream_dev.ko

Could that (or some of that), but the why the new version of gcc defaults signed vs unsigned char?
I think there is a compiler flag to make that explicit?  And match the way previous versions defaulted?

Habraken, 2025-05-16:
For it to work there are in fact 2 install scripts that need to be run. In cariboulite/driver there is also an install script. This one needs to be run first. Once this has completed, you need to reboot first, before you run the install script in cariboulite.

Habraken, 2025-11-19:
@NespaLa, I have a couple of questions to better understand your situation.
Did you run both install scripts?
Were there any compile warnings or errors?